pschatzmann/arduino-audio-tools #1469: Sound glitches when using maximilian on esp32-s3

rupraph, 2024-03-27:
### Problem Description

I have sound glitches when trying out Maximilian exemples with my setup.
The glithes (crackling) show on a oscilloscope as short audio signal interuptions. Scope also showed this is due to sporadic interuptions on the I2S data stream to the DAC. (this was not the case when trying out basic generator from audiotools itself). 

### Device Description

I am running on an esp32-s3

### Sketch

```cpp
#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/MaximilianDSP.h"

I2SStream out; 
Maximilian maximilian(out);

maxiOsc mySine,myOtherSine;//Two oscillators

void setup(){
  Serial.begin(115200);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  // Custom I2S output pins
  config.pin_bck = 6;
  config.pin_ws = 4;
  config.pin_data = 5;
  out.begin(config);
  maximilian.begin(config);
}

void loop(){
  maximilian.copy();
}

void play(float *output) {
    output[0]=mySine.sinewave(440+(myOtherSine.sinewave(1)*100));
    output[1]=output[0];

}
```

### What is your development environment

PlatformIO. 
[env:adafruit_feather_esp32s3_nopsram]
platform = espressif32
board = adafruit_feather_esp32s3_nopsram
framework = arduino
lib_deps = 
    https://github.com/pschatzmann/arduino-audio-tools.git#v0.9.7
    https://github.com/pschatzmann/Maximilian.git#A2.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-28:
I guess your cpu is just too slow: why don't you try to reduce the sampling rate ?

rupraph, 2024-03-28:
Right, thanks ! Indeed, reducing the sample rate got rid of the cracks... Sorry for the false alarm :) 
Since I had tried increasing the buffer size and it did not help, I though the problem was somewhere else. 

Then I am curious then to have your opinion on the dsp library choice and their resources consumption. When Using audiotools generator I could have a generator together with a delay effect without any issues, but this simple synth in maximilian already crackles.
How do they compare in term of ressources consumption ? Is Maximilian more ressources intensive ? Is STK lighter ?  
(I undestood mozzi will be the lightest, but while it can be upscaled to 16bits, actual processing remains in 8 bits so a bit low Q...)

pschatzmann, 2024-03-28:
STK and Maximilian use both floating point operations and I need to convert them to integers to output. 
Floating point operations are usually the weak spot on microcontrollers unless you have one with a fast FPU.

I never compared a esp32-s3 with a regular esp32: but I assume that it is slower. You could try to run the [FFT test](https://github.com/pschatzmann/arduino-audio-tools/wiki/FFT) on your microcontroller to get a better feeling

rupraph, 2024-03-28:
Great. thanks for your support. 

Seems like regular ESP32 and ESP32-S3 are both very comparable in cpu perf (which make sense based on the datasheet).
 I just performed the FFT tests you proposed so that we can put them next to your previous tests, and here are the results :) (I pasted the ESP32 results from your own test as a reference)
 The reported values are the average over 200 serialprints since the reported values are integers
 
| ffttype|ESP32| ESP32-S3|	
|---|---|---|
AudioRealFFT |3.3|2.6| 	
AudioESP32FFT|1.1|1.25| 	
AudioEspressifFFT|3.5|3.2|

pschatzmann, 2024-03-28:
Thanks, I updated the table. 

But it is strange: I just repeated the test with your sketch and on my ESP32 this is working perfectly.
Does it make a difference if you reduce the volume ( maximilian.setVolume(0.5);) ?
)

rupraph, 2024-03-28:
Indeed this is strange. 
I retried the initial sketch and reducing the volume, and the glitches are back.
 Not sure what could cause this different behaviour btw the 2 boards. Could ram be an issue here ? The current platformio config I use runs without external ram.

pschatzmann, 2024-03-28:
Maybe you double check with Arduino, just to confirm that the issue is not coming from PlatformIO

closed 2024-04-04 by pschatzmann