pschatzmann/arduino-audio-tools #1470: >: -c: line 1: unexpected EOF while looking for matching `''

brainstorm, 2024-03-28:
### Problem Description

Arduino: 1.8.19 (Linux), Board: "ESP32C6 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled, Disabled"

arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/rvalls/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/rvalls/.arduino15/packages -libraries /home/rvalls/Arduino/libraries -fqbn=esp32:esp32:esp32c6:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none,EraseFlash=none,ZigbeeMode=default -ide-version=10819 -build-path /tmp/arduino_build_712472 -warnings=none -build-cache /tmp/arduino_cache_294747 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-12.2.0_20230208.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-12.2.0_20230208 -prefs=runtime.tools.esptool_py.path=/home/rvalls/.arduino15/packages/esp32/tools/esptool_py/4.6 -prefs=runtime.tools.esptool_py-4.6.path=/home/rvalls/.arduino15/packages/esp32/tools/esptool_py/4.6 -prefs=runtime.tools.esp-rv32.path=/home/rvalls/.arduino15/packages/esp32/tools/esp-rv32/2302 -prefs=runtime.tools.esp-rv32-2302.path=/home/rvalls/.arduino15/packages/esp32/tools/esp-rv32/2302 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-12.2.0_20230208.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-12.2.0_20230208 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=/home/rvalls/.arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/12.1_20221002 -prefs=runtime.tools.riscv32-esp-elf-gdb-12.1_20221002.path=/home/rvalls/.arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/12.1_20221002 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-12.2.0_20230208.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208 -prefs=runtime.tools.openocd-esp32.path=/home/rvalls/.arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20230921 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230921.path=/home/rvalls/.arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20230921 -prefs=runtime.tools.mklittlefs.path=/home/rvalls/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/home/rvalls/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/12.1_20221002 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20221002.path=/home/rvalls/.arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/12.1_20221002 -prefs=runtime.tools.mkspiffs.path=/home/rvalls/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/rvalls/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.dfu-util.path=/home/rvalls/.arduino15/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=/home/rvalls/.arduino15/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.esp32-arduino-libs.path=/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-3662303f31.path=/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31 -verbose /home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial/streams-i2s-serial.ino

Using board 'esp32c6' from platform in folder: /home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3
Using core 'esp32' from platform in folder: /home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3
WARNING: Category 'Sound' in library ESP_I2S is not valid. Setting to 'Uncategorized'
WARNING: Category 'Sound' in library ESP_SR is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library TFLite Micro is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
/usr/bin/env bash -c "[ ! -f "/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial"/partitions.csv ] || cp -f "/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial"/partitions.csv "/tmp/arduino_build_712472"/partitions.csv"
/usr/bin/env bash -c "[ -f "/tmp/arduino_build_712472"/partitions.csv ] || [ ! -f "/home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/variants/esp32c6"/partitions.csv ] || cp "/home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/variants/esp32c6"/partitions.csv "/tmp/arduino_build_712472"/partitions.csv"
/usr/bin/env bash -c "[ -f "/tmp/arduino_build_712472"/partitions.csv ] || cp "/home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3"/tools/partitions/default.csv "/tmp/arduino_build_712472"/partitions.csv"
/usr/bin/env bash -c "[ -f "/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial"/bootloader.bin ] && cp -f "/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial"/bootloader.bin "/tmp/arduino_build_712472"/streams-i2s-serial.ino.bootloader.bin || ( [ -f "/home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/variants/esp32c6"/bootloader.bin ] && cp "/home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/variants/esp32c6"/bootloader.bin "/tmp/arduino_build_712472"/streams-i2s-serial.ino.bootloader.bin || python3 "/home/rvalls/.arduino15/packages/esp32/tools/esptool_py/4.6"/esptool.py --chip esp32c6 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "/tmp/arduino_build_712472"/streams-i2s-serial.ino.bootloader.bin "/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6"/bin/bootloader_qio_80m.elf )"
esptool.py v4.6
Creating esp32c6 image...
Merged 2 ELF sections
Successfully created esp32c6 image.
/usr/bin/env bash -c "[ ! -f "/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial"/build_opt.h ] || cp -f "/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial"/build_opt.h "/tmp/arduino_build_712472"/build_opt.h"
/usr/bin/env bash -c "[ -f "/tmp/arduino_build_712472"/build_opt.h ] || : > "/tmp/arduino_build_712472"/build_opt.h"
/usr/bin/env bash -c ": > '/tmp/arduino_build_712472/file_opts'"
Detecting libraries used...
"/home/rvalls/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/riscv32-esp-elf-g++" -MMD -c "@/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/flags/cpp_flags" -w -Os -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10819 -DARDUINO_ESP32C6_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32C6_DEV" -DARDUINO_VARIANT="esp32c6" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="linux" -DARDUINO_FQBN="esp32:esp32:esp32c6:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none,EraseFlash=none,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0    -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0  "@/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/flags/defines" "-I/home/rvalls/Arduino/libraries/arduino-audio-tools-0.9.7/examples/examples-stream/streams-i2s-serial" -iprefix "/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/include/" "@/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/flags/includes" "-I/home/rvalls/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31 [... cut ...]
Using library WiFi at version 2.0.0 in folder: /home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/WiFi 
exit status 2
Error compiling for board ESP32C6 Dev Module.

### Device Description

ESP32C6 dev board

### Sketch

```cpp
Any sketch
```

### What is your development environment

Arduino desktop 1.8.19 on Ubuntu 23.10 (Mantic)

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-28:
What is your ESP32 core version ? 
What Board do you select ?
I am getting no boards found for ESP32C6...

pschatzmann, 2024-03-28:
I could not reproduce your exact issue. You can try to deactivate TDM and PDM in AudioConfig.h for your processor or install the latest version where the compile errors for TDM and PDM have been corrected.

brainstorm, 2024-04-01:
> What is your ESP32 core version ? What Board do you select ? I am getting no boards found for ESP32C6...

I think it says in the dump itself?:

Using board 'esp32c6' from platform in folder: /home/rvalls/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3

So that's 3.0.0-alpha3 as "core version" and I have an ESP32C6 board. Sorry for the wall of text :-S

brainstorm, 2024-04-01:
> I could not reproduce your exact issue. You can try to deactivate TDM and PDM in AudioConfig.h for your processor or install the latest version where the compile errors for TDM and PDM have been corrected.

I downloaded/installed this repo's zip the same day I filed this issue, I'll definitely check if there are changes since then and apply the changes you suggest... except that I'd need TDM/PDM for my INP441 microphone otherwise recording audio data will not work? :-S