pschatzmann/arduino-audio-tools #1473: Compilation error of example "base-i2s-a2dp"

acca84, 2024-03-29:
### Problem Description

Hello, i try to compile your example "[base-i2s-a2dp](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-i2s-a2dp)", i copied this example in main.cpp in Vscode
I tried using Vscode and arduino IDE after importing libs in arduino and vscode
Both methods give the same result
On arduino IDE i have 2.0.14 esp32 lib version.

I have many erros like
 `In file included from src/main.cpp:11:
.pio/libdeps/esp32dev/audio-tools/src/AudioLibs/A2DPStream.h:24:8: error: 'BufferRTOS' does not name a type; did you mean 'NBuffer'?
 static BufferRTOS<uint8_t>a2dp_buffer{A2DP_BUFFER_SIZE * A2DP_BUFFER_COUNT, A2DP_BUFFER_SIZE, portMAX_DELAY, portMAX_DELAY};
        ^~~~~~~~~~
        NBuffer`

### Device Description

esp32 wroom

### Sketch

```cpp
My platfomio.ini is :
[platformio]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
lib_deps = https://github.com/pschatzmann/arduino-audio-tools, https://github.com/pschatzmann/arduino-libhelix, https://github.com/greiman/SdFat, https://github.com/pschatzmann/ESP32-A2DP.git
build_flags = -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

### What is your development environment

Tried both Arduino IDE and Vscode

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-29:
I just committed the correction

closed 2024-03-29 by pschatzmann

acca84, 2024-03-29:
Thanks !
Now i have another problem, just opened a new issue.
Thanks for your work !