pschatzmann/arduino-audio-tools #1474: base-i2s-a2dp example cause Guru meditation error

acca84, 2024-03-29:
### Problem Description

Hi,
I'm trying tu use the "base-i2s-a2dp" example, it compiles and uploads to my esp wroom 32 board
But it crashes at startup

The error is : 
`

Backtrace: 0x400dd30b:0x3ffcf9d0 0x400d4221:0x3ffcfa10 0x400d42d9:0x3ffcfa80 0x400d638d:0x3ffcfaa0 0x400d6acb:0x3ffcfb10 0x400d7568:0x3ffcfb40 0x400d36b1:0x3ffcfb80 0x400d6d12:0x3ffcfba0 0x400d3425:0x3ffcfbc0 0x400d346d:0x3ffcfbe0 0x400d611b:0x3ffcfc00 0x400d624b:0x3ffcfc40 0x400d6289:0x3ffcfc60 0x400d86f6:0x3ffcfc80 0x400d8e26:0x3ffcfd00 0x400db436:0x3ffcfde0

ELF file SHA256: 730bf8e2bc503a90
`

And the Exception decoder gives : 
`
PC: 0x400dd30e: i2s_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver\i2s.c:2129
EXCVADDR: 0x00000018

Decoding stack results
0x400dd30b: i2s_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver\i2s.c:2129
0x400d4221:  is in audio_tools::I2SDriverESP32::writeBytes(void const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:89).
0x400d42d9:  is in audio_tools::I2SStream::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioI2S/I2SStream.h:94).
0x400d638d:  is in audio_tools::ResampleStream::write<short>(Print*, unsigned char const*, unsigned int, unsigned int&) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h:349).
0x400d6acb:  is in audio_tools::ResampleStream::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h:297).
0x400d7568:  is in audio_tools::NumberFormatConverterStreamT<int, short>::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:317).
0x400d36b1:  is in audio_tools::NumberFormatConverterStream::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:445).
0x400d6d12:  is in audio_tools::ChannelFormatConverterStreamT<int>::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:37).
0x400d3425:  is in audio_tools::ChannelFormatConverterStream::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:149).
0x400d346d:  is in audio_tools::FormatConverterStream::write(unsigned char const*, unsigned int) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:682).
0x400d611b: audio_tools::TransformationReader<audio_tools::ReformatBaseStream>::getByteFactor() at c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools\ResampleStream.h:118
0x400d624b:  is in audio_tools::TransformationReader<audio_tools::ReformatBaseStream>::begin(audio_tools::ReformatBaseStream*, Stream*) (c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h:35).
0x400d6289: audio_tools::ReformatBaseStream::setupReader() at c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools\ResampleStream.h:137
0x400d86f6: audio_tools::FormatConverterStream::begin(audio_tools::AudioInfo) at c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools\AudioStreamsConverter.h:671
0x400d8e26: setup() at c:\Users\Francois\Dropbox\ARDUINO\Mes sketchs\libraries\arduino-audio-tools-main\src/AudioTools\AudioStreamsConverter.h:636
0x400db436: loopTask(void*) at C:\Users\Francois\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\main.cpp:42
`

Is that problem known please ?

### Device Description

esp wroom 32  or esp32 wrover

### Sketch

```cpp
base-i2s-a2dp
```

### What is your development environment

Arduino IDE 2.3.2

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-30:
Uhh, that was hard to find: I committed some corrections and [tested the FormatConverter with this sketch](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/tests/conversion/format-converter-in/format-converter-in.ino)

acca84, 2024-03-30:
Thanks, it works fine now !

closed 2024-04-04 by pschatzmann