pschatzmann/arduino-audio-tools #1479: streams-url_mp3_helix-i2s example cause Guru meditation error

billism1, 2024-04-04:
### Problem Description

When running the streams-url_mp3_helix-i2s example in latest main branch (SHA 8915c0c) on my ESP32 S3 (esp32-s3-devkitc-1-n16r8v) the program crashes at start-up. Of note, the exact same sketch had worked a few days ago, so I tested with the version of main from March 31 (SHA e655f0a) and the error does not occur. So the issue was introduced after that.

This looks similar to issue #1474 but must be different since that has been addressed already.

The error details can be seen in the following log:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420c2bda
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   247][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[I] AudioTypes.h : 124 - sample_rate: 44100
[I] AudioTypes.h : 125 - channels: 2
[I] AudioTypes.h : 126 - bits_per_sample: 16
[I] I2SConfigESP32.h : 80 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Master
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 16
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: true
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 103 - pin_bck: 10
[I] I2SConfigESP32.h : 105 - pin_ws: 9
[I] I2SConfigESP32.h : 107 - pin_data: 11
[I] URLStream.h : 81 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): http://stream.srg-ssr.ch/m/rsj/mp3_128
[I] Url.h : 58 - Url::parse
[I] Url.h : 96 - url->http://stream.srg-ssr.ch/m/rsj/mp3_128
[I] Url.h : 97 - host->stream.srg-ssr.ch
[I] Url.h : 98 - protocol->http
[I] Url.h : 99 - path->/m/rsj/mp3_128
[I] Url.h : 100 - port->80
[I] URLStream.h : 399 - bool audio_tools::URLStream::login()
[   279][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   314][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   316][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   315][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.[   392][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: The Multiverse, BSSID: 70:3a:cb:09:b6:12, Channel: 1, Auth: WPA2_PSK       
[   394][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   431][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.86.40
[   432][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   435][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.86.40, MASK: 255.255.255.0, GW: 192.168.86.1

[I] URLStream.h : 373 - WiFiClient
[I] HttpRequest.h : 222 - process connecting to host stream.srg-ssr.ch port 80
[I] HttpRequest.h : 337 - connected 1 timeout 60000
[I] HttpRequest.h : 233 - Free heap: 261492
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420c1790  PS      : 0x00060530  A0      : 0x82008ee4  A1      : 0x3fcebb30
A2      : 0x3fcf1d60  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x3fcf1d60  A7      : 0x00000014  A8      : 0x420c1210  A9      : 0x3fcebb10
A10     : 0x3fcebbbc  A11     : 0x3d80b35c  A12     : 0x00000000  A13     : 0x60038000
A14     : 0x00000000  A15     : 0x3c0da384  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x420c178d:0x3fcebb30 0x42008ee1:0x3fcebb50 0x42008fcf:0x3fcebc40 0x42009d1e:0x3fcebca0 0x42004a65:0x3fcebcd0 0x42009558:0x3fcebd00 0x420096bc:0x3fcebd40 0x42009801:0x3fcebd60 0x42015f22:0x3fcebe20

  #0  0x420c178d:0x3fcebb30 in audio_tools::StrExt::operator=(audio_tools::StrExt&) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioBasic/StrExt.h:59
  #1  0x42008ee1:0x3fcebb50 in audio_tools::HttpHeaderLine::operator=(audio_tools::HttpHeaderLine&) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/HttpHeader.h:45
      (inlined by) audio_tools::Vector<audio_tools::HttpHeaderLine>::push_back(audio_tools::HttpHeaderLine) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioBasic/Collections/Vector.h:162
      (inlined by) audio_tools::HttpHeader::headerLine(char const*) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/HttpHeader.h:355
  #2  0x42008fcf:0x3fcebc40 in audio_tools::HttpHeader::put(char const*, char const*) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/HttpHeader.h:106 (discriminator 3)
  #3  0x42009d1e:0x3fcebca0 in audio_tools::HttpRequest::processBegin(MethodID, audio_tools::Url&, char const*, int) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/HttpRequest.h:243
  #4  0x42004a65:0x3fcebcd0 in audio_tools::HttpRequest::process(MethodID, audio_tools::Url&, char const*, char const*, int) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/HttpRequest.h:187
  #5  0x42009558:0x3fcebd00 in int audio_tools::URLStream::process<char const*>(MethodID, audio_tools::Url&, char const*, char const*, int) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/URLStream.h:336
  #6  0x420096bc:0x3fcebd40 in audio_tools::URLStream::begin(char const*, char const*, MethodID, char const*, char const*) at .pio/libdeps/url-mp3-stream-to-i2s-amp/audio-tools/src/AudioHttp/URLStream.h:86
  #7  0x42009801:0x3fcebd60 in setup() at src/url-mp3-stream-to-i2s-amp/main.cpp:35
  #8  0x42015f22:0x3fcebe20 in loopTask(void*) at C:/Users/Bill Krahmer/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42      

ELF file SHA256: a6f79c89da5f9fbc

Rebooting...
```

### Device Description

ESP32 S3: esp32-s3-devkitc-1-n16r8v

### Sketch

```cpp
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-url_mp3_helix-i2s/streams-url_mp3_helix-i2s.ino
```

### What is your development environment

Platform.io in VS Code 1.87.2
framework-arduinoespressif32 3.20014.231204

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-04:
This is strange: Can you double check if it helps if you change the value #define USE_ALLOCATOR  to false in AudioConfg.h line 310.

billism1, 2024-04-04:
That does indeed prevent the error. I can also confirm that your latest commits ( SHA f2f967c ) work with this sketch. Thank you. You are very much appreciated!

closed 2024-04-04 by billism1