pschatzmann/arduino-audio-tools #1487: rp2040 pico_w

nictoglobe, 2024-04-06:
### Problem Description

does not work with rp2040

### Device Description

pico_w

### Sketch

```cpp
all examples fail with improper includes
It seems this does not work with other microcontrollers , only for ESP . Is that correct?
If so , will it be possible to add pico (rp2040) support?
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-06:
Which core are you usin ? It should work with the one from Eearl Phil Hower!

If you want to use the mbed implementation from Arduino you need to install the [I2S library](https://github.com/pschatzmann/rp2040-i2s)!

closed 2024-04-06 by pschatzmann

nictoglobe, 2024-04-07:
Hi 

i am develooing with arduino sdk v2
Using pico_w on piromini vgboard
I triex to compile using supplied shell script
All examples returned rc=1 , ie failed

It seems rp2040 is nit supported or I do not know howto

Please advise, should I give up or do I need to make an adjustment somewhere?

kind regards

ajaco

pschatzmann, 2024-04-07:
Install the open source [RP2040 Arduino Core](https://arduino-pico.readthedocs.io/en/latest/) 
In Arduino
- Select the Target Rasperry PI Pico W
- Select a sketch from the examples: File -> examples -> audio-tools: e.g. example-streams/streams-generator-i2s.ino
- Click on the Verify button
- Click on the Upload Button

nictoglobe, 2024-04-07:
I tried to use several sketches one of them as an not working example:

**base-i2s-adp.ino** from audio-tools/examples/examples-basic-api/

**/Users/Andreas/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPCommon.h:86:4: error: #error Unsupported ESP32 Version: Upgrade the ESP32 version in the Board Manager**

Apparently somewhere a condition exists that ONLY esp microcontollers will work (I am using [RP2040 Arduino Core] now btw)

So still stuck 

Any advise?

ajaco

pschatzmann, 2024-04-07:
You are funny: The A2DP library only works with the ESP32!