pschatzmann/arduino-audio-tools #1517: Example "streams-audiokit-webserver_mp3" doesn't build anymore

dl9sec, 2024-04-20:
### Problem Description

The example "streams-audiokit-webserver_mp3" doesn't build anymore and ends up with the error message:
```
c:/users/thors/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/thors/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 600 bytes
```

### Device Description

ESP32 AudioKit 2.2 A402

### Sketch

```cpp
examples/examples-audiokit/streams-audiokit-webserver_mp3/streams-audiokit-webserver_mp3.ino
```

### Other Steps to Reproduce

auduino-audio-tools, arduino-audio-driver and arduino-liblame pulled yesterday evening from the repos.

### What is your development environment

Arduino 1.8.19 with latest updates for LIBs and BSPs

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-20:
Did you read the README of the liblame and try to set USE_MEMORY_HACK to 0 ?

dl9sec, 2024-04-20:
HI Phil,

sure, you set this to 0 by default.
Do you mean an ESP32 core?

pschatzmann, 2024-04-20:
Yes, the newer, the less memory you have available...

dl9sec, 2024-04-20:
Ok, thanks. Tried it with 2.0.14 and it builds fine again.
It's a pitty, that that newer versions of the ESP32 core a  more and more hungry for memory and there is no possibility to take influence on that in the board parameters...

pschatzmann, 2024-04-20:
Anyhow, I would not recommend the combination of MP3 encoding and WIFI!
This is somehow too close to the limits...

dl9sec, 2024-04-20:
Yes, I am aware of the situation. Maybe I will go back from MP3 to WAV..

dl9sec, 2024-04-20:
Oh wow, great stuff!
Thank you so much for your effort. Builds perfect with ESP32 core 2.0.15 and RAM usage decreases a lot :-)