pschatzmann/arduino-audio-tools #1521: variable-sized object may not be initialized

jaapnoordzij, 2024-04-22:
### Problem Description

I am trying to "cmake" the generator desktop example. 
It fails with an error:
```
/Users/jaapnoordzij/Documents/Arduino/libraries/arduino-audio-tools/src/AudioBasic/StrExt.h:129:17: error: variable-sized object may not be initialized
    char result[new_size + 1] = {0};
```
Is this an error in the code or can I specify some flag to prevent the error ?

### Device Description

Desktop build on MacOS Sonoma 14.4.1
I have done a 'git pull' just before compiling

### Sketch

```cpp
examples_desktop/generator (unmodified)
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-22:
Strange: I can't reproduce your problem. 
What compiler version are you using ?

jaapnoordzij, 2024-04-22:
Apple standard distribution

Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: x86_64-apple-darwin23.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

pschatzmann, 2024-04-22:
I committed [a correction](https://github.com/pschatzmann/arduino-audio-tools/commit/f2d732f72d169b7a9f214d6b973295e7b9de827c): it's compiling now with clang as well...

jaapnoordzij, 2024-04-22:
Perfect, Thank you !

jaapnoordzij, 2024-04-22:
The generation example compiles and runs fine now.
But what I actually wanted to do is play around with STK on my laptop.
So next I want to cmake 
`~/Documents/Arduino/libraries/arduino-audio-tools/examples/examples-dsp/examples-stk/streams-stk-desktop`
without any modifications just from the example.

In the Mac terminal I navigate to this directory, `mkdir build`, `cd build` and run `cmake -DCMAKE_BUILD_TYPE=Debug ..`

Apparently the cmake system sets the variable for CMAKE_SOURCE_DIR wrong because I get an error:
```
CMake Error at CMakeLists.txt:20 (add_subdirectory):
  The source directory

    /Users/jaapnoordzij/Documents/Arduino/libraries/arduino-audio-tools/examples

  does not contain a CMakeLists.txt file.
```

I am an absolute beginner with cmake and the documentation only shows me the the source dir should be set to the directory .. that contains the CMakeLists.txt file (it is there !). Why does cmake select a dir that is some levels up ?

What am I doing wrong ?

pschatzmann, 2024-04-22:
Just comment it out: this is only needed if you compile it for an AudioKit

pschatzmann, 2024-04-22:
From above:
but you need #include "AudioLibs/PortAudioStream.h"

With the last commit it is working for me...

closed 2024-04-22 by pschatzmann

jaapnoordzij, 2024-04-23:
Yes of course, that was a dumb reaction of me, I realized that only later on.
Thank you, works perfectly now.