pschatzmann/arduino-audio-tools #1522: DPS example does not compile

Aggebitter, 2024-04-22:
### Problem Description

Arduino IDE 2.30
example: streams_stk_sine_audiokit.ino
"...\arduino-audio-tools-0.9.7\src/AudioLibs/AudioSTK.h:8:10: fatal error: StkAll.h: No such file or directory"

Seems that there is missing files for Maximillian examples to:
 in MaximillianDSP.h missing #include "maximilian.h" #include "libs/maxiClock.h"

am I missing something 

// Agge

### Device Description

ESP wrower TTGO T-Audio T9 ver 1.5

### Sketch

```cpp
Example sketches
```

### Other Steps to Reproduce

Just did a compile without changes to reproduce the errors

### What is your development environment

Arduino 2.30

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-22:
Sees that you just forgot to [install the library](https://github.com/pschatzmann/arduino-audio-tools/wiki/Optional-Libraries)

closed 2024-04-22 by pschatzmann

Aggebitter, 2024-04-22:
Jupp noticed that and there is no comment on it in the wiki for arduino audiotools or in the example files ;-)