pschatzmann/arduino-audio-tools #1523: Vector.h errors

codegrue, 2024-04-22:
### Problem Description

Building with the latest code today I get these errors:

/audio-tools/src/AudioBasic/Collections/Vector.h:353:18: error: 'void* memmove(void*, const void*, size_t)' writing to an object of type 'class audio_tools::int24_4bytes_t' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
           memmove(p_data, oldData, len * sizeof(T));

/audio-tools/src/AudioBasic/Collections/Vector.h:355:17: error: 'void* memset(void*, int, size_t)' clearing an object of type 'class audio_tools::StrExt' with no trivial copy-assignment; use assignment or value-initialization instead [-Werror=class-memaccess]
           memset(oldData, 0, len * sizeof(T));

### Device Description

ESP32-S3

### Sketch

```cpp
No Code involved, just compiling the library produces these errors.
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-22:
I cant reproduce your problem. Please double check the version information:
```
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.6.0+sha.c9c9bbe) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 2.0.0
|-- audio-tools @ 0.9.6+sha.cd7a041
|-- audiokit @ 0.7.0+sha.f8c4139
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Building .pio/build/esp32dev/bootloader.bin
Generating partitions .pio/build/esp32dev/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/esp32dev/lib43f/Wire/Wire.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiAP.cpp.o
Archiving .pio/build/esp32dev/lib43f/libWire.a
Indexing .pio/build/esp32dev/lib43f/libWire.a
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiClient.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiMulti.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiScan.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiServer.cpp.o
Compiling .pio/build/esp32dev/lib8bc/WiFi/WiFiUdp.cpp.o
Compiling .pio/build/esp32dev/libf2b/WiFiClientSecure/WiFiClientSecure.cpp.o
Archiving .pio/build/esp32dev/lib8bc/libWiFi.a
Compiling .pio/build/esp32dev/libf2b/WiFiClientSecure/esp_crt_bundle.c.o
Indexing .pio/build/esp32dev/lib8bc/libWiFi.a
Compiling .pio/build/esp32dev/libf2b/WiFiClientSecure/ssl_client.cpp.o
Compiling .pio/build/esp32dev/lib07b/SPI/SPI.cpp.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/ac101/ac101.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es7148/es7148.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es7210/es7210.c.o
Archiving .pio/build/esp32dev/lib07b/libSPI.a
Indexing .pio/build/esp32dev/lib07b/libSPI.a
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es7243/es7243.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es7243e/es7243e.c.o
Archiving .pio/build/esp32dev/libf2b/libWiFiClientSecure.a
Indexing .pio/build/esp32dev/libf2b/libWiFiClientSecure.a
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es8156/es8156.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es8311/es8311.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es8374/es8374.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/es8388/es8388.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_driver/tas5805m/tas5805m.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/audio_gpio.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/audio_hal.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/audio_mem.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/audio_mutex.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/audio_volume.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/audiokit_logger.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/driver_pins.cpp.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/i2c_bus.c.o
Compiling .pio/build/esp32dev/liba53/audiokit/audio_hal/i2c_wire.cpp.o
[... 50 lines cut ...]
      ^~~~~~~~~~~
Compiling .pio/build/esp32dev/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32dev/libFrameworkArduino.a
Indexing .pio/build/esp32dev/libFrameworkArduino.a
Linking .pio/build/esp32dev/firmware.elf
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.6% (used 31552 bytes from 327680 bytes)
Flash: [=====     ]  46.1% (used 604449 bytes from 1310720 bytes)
Building .pio/build/esp32dev/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
===================================== [SUCCESS] Took 70.90 seconds =====================================
 *  Terminal will be reused by tasks, press any key to close it. 
 * 
```

closed 2024-04-22 by pschatzmann

codegrue, 2024-04-22:
Processing WT32SC01 (board: WT32SC01; framework: arduino, espidf; platform: espressif32)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/WT32SC01.html
PLATFORM: Espressif 32 (6.5.0) > PanLee SC01 Plus ESP32-35inch (16MB QD, 2MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - framework-espidf @ 3.40406.240122 (4.4.6)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LovyanGFX @ 1.1.12
|-- lvgl @ 8.3.11
|-- ArduinoJson @ 6.21.5
|-- optional-lite @ 3.4.0
|-- audio-tools @ 0.9.6+sha.cd7a041
|-- Wire @ 2.0.0
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- HTTPUpdate @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- Preferences @ 2.0.0
Building in release mode

codegrue, 2024-04-22:
I think in arduino this would be a warning but in espidf it's an error

codegrue, 2024-04-22:
Well your example has

```
framework = arduino
```

But I am using sdkconfig file for use with ESP32 so need both frameworks:

```
framework = arduino, espidf
```

Adding espidf turns on more strict error checking. I have been able to disable the warnings with `-Wno` but this is the first time an error broke my ability to build.

I tried this to no avail:

```
build_flags = 
	-Wno-error
```

codegrue, 2024-04-22:
Yes I did try that and it didn't work. Having both frameworks allows the use of Arduino libraries _and_ the use of ESPIDF libraries including customizing configuration options. Arduino alone hard codes a bunch of config settings that cannot be changed at compile time. I believe I had to use that to fine tune some PSRAM storage options not related to audio-tools.

That said, I think I may not need that any longer so I will play with just Arduino again and see if it kills my heap size. 

BTW, googling the error message, it seems like it involves having a constructor for an object that is implicitly created by declaring it.

pschatzmann, 2024-04-22:
I can try to undo my correction which might cause the problems: It was supposed to address some compile warnings with clang

pschatzmann, 2024-04-22:
Can you double check with the latest version.
Can you provide your platformio.ini file ? 
I don't manage to get your combination to compile

codegrue, 2024-04-22:
I'm converting a demo program to this config to show the issue. One sec....

codegrue, 2024-04-22:
Okay this project shows the error when you build it:

https://github.com/codegrue/WS32SC01-LVGL-Sound

pschatzmann, 2024-04-22:
I think you can ignore or better try to suppress this somehow...

codegrue, 2024-04-22:
Okay working to remove espidf from the project. Thanks.