pschatzmann/arduino-audio-tools #1528: send I2S Data in Slave Mode doesnt work

mutedh, 2024-04-26:
### Problem Description

Hello!

My Hardware is Esp32 devkit V4 and Adau1701 from wondom . I tested the example " streams-generator-i2s.ino" in master mode and its work, also in Serial monitor shows that the packets are copied. But when i do it in Slave Mode, also is_master= false, doesnt show nothing. I connected the ESp32 to the  DSP Adau1701 from Wondoom and it in Master Mode, but still doesnt work. What happend? i thought the ESp32 can work in slave mode and Tx mode. You know which errors i do?

### Device Description

ESp32 Devkit V4, ADAU1701 Wondom (APM2) and APM3

### Sketch

```cpp
`/**
 * @file streams-generator-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-i2s/README.md 
 * @copyright GPLv3
 */
 
#include "AudioTools.h"

AudioInfo info(44100, 2, 16);
SineWaveGenerator<int16_t> sineWave(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound(sineWave);             // Stream generated from sine wave
I2SStream out; 
StreamCopy copier(out, sound);                             // copies sound into i2s

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  while(!Serial);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);

  config.copyFrom(info); 
  config.is_master=false;
  out.begin(config);

  // Setup sine wave
  sineWave.begin(info, N_B4);
  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}
`
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-26:
You will need to check your signal in an oscilloscope to double check if you send a valid 16 bit bit clock and word select signal

closed 2024-04-26 by pschatzmann

mutedh, 2024-05-03:
hello! thankyou for answer.

i changed from 
AudioInfo info(44100, 2, 16);
 to 
 AudioInfo info(**48000**, 2, **24**);
and now  i hear but only noise.

i see in bck pin  that i have 72KHz and in WS Pin 48 KHz.

I tested in my ADAU1701 to do a loop ( the input signal of Adau1701 are the output of a signal generator from adau1701) and it works. but with the esp32 not. what could be happend? i dont receive  a good data signal

thank you

pschatzmann, 2024-05-03:
I hope you changed the sine generator as well from 16 to 24 bits!

mutedh, 2024-05-03:
yes, that is my code:

/**
 * @file streams-generator-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-i2s/README.md 
 * @copyright GPLv3
 */
 
#include "AudioTools.h"

AudioInfo info(48000, 2, 24);
SineWaveGenerator<int24_t> sineWave(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int24_t> sound(sineWave);             // Stream generated from sine wave
I2SStream out; 
StreamCopy copier(out, sound);                             // copies sound into i2s

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  while(!Serial);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);

  config.copyFrom(info); 
  config.is_master=false;

  config.pin_bck=26;
  config.pin_ws=25;
  config.pin_data=33;
  out.begin(config);
 
  // Setup sine wave
  sineWave.begin(info, N_B4);
  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}

pschatzmann, 2024-05-03:
Did you read the Technical Sheet for Adau1701 to figure out what you need to provide ?
By the way it is crazy to leave the logger at info with these settings!

mutedh, 2024-05-03:
i comment the logger and its works !

but one question, the left side i dont hear  too much in compare of the right side, could be it a problem of the adau1701?

in DSP i have only a volume control of both channel at same time

pschatzmann, 2024-05-03:
I suggest you read the Wiki about Volume Control and Panning

mutedh, 2024-05-15:
i have one question, i use your Tool FormatConverterStream, and when i change the bit_rate from 16 to 24, one channel doesnt work, also is only a noise. and when i put from 24 to 24, it works boths channels. here are the Code:

`#include "AudioTools.h"

AudioInfo from(48000,2,16);
AudioInfo to(48000,2,24);
SineWaveGenerator<int16_t> sineWave(32000);                
GeneratedSoundStream<int16_t> sound(sineWave); // Stream generated from sine wave
//CsvStream<int24_t> out(Serial); 
I2SStream out;   // or any other e.g. AudioBoardStream, CsvOutput<int16_t> out(Serial); 
//FormatConverterStream converter(out);  // or use converter(out)
NumberFormatConverterStream nfc(out);
StreamCopy copier(nfc, sound);       //        copier(converter, sound);     

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  //AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // Setup Input
  sineWave.begin(from, N_B4);
  sound.begin(from);

  // Define Converter
  //converter.begin(from, to);
  nfc.begin(16,24);
  // Start Output
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.copyFrom(to);
    
   config.is_master=false;
   config.pin_data = 33;
   config.pin_bck=26;
   config.pin_ws=25;
  out.begin(config);

  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}`

I read the documentation,search in Internet and view the code. But i dont find a solution. Could you help me?

Thank you!

pschatzmann, 2024-05-15:
I can't reproduce your issue: I was checking both the CSV output and the output via I2S and both generate a singal on both channels.

mutedh, 2024-05-15:
Thank you for the fast answer!

with the CSV Output for me are good too, but with the i2s become the noise.

I can not testing with 32 bits, because my dsp are only for 24 bits

mutedh, 2024-05-15:
i can only try with esp32 only as slave.

pschatzmann, 2024-05-15:
I tried to google the topic ESP32 and Adau1701 and came to the conclusion that this is a bad combination!
With the ESP32 as slave, you can't provide any master clock, so the ESP32 needs to be master...

mutedh, 2024-05-15:
but what i dont understand is, works when i use this code:
``/**
 * @file streams-generator-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-i2s/README.md 
 * @copyright GPLv3
 */
 
#include "AudioTools.h"

AudioInfo info(48000, 2, 24);
SineWaveGenerator<int24_t> sineWave(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int24_t> sound(sineWave);             // Stream generated from sine wave
I2SStream out; 
StreamCopy copier(out, sound);                             // copies sound into i2s

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  while(!Serial);
 // AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);

  config.copyFrom(info); 
  config.is_master=false;
  out.begin(config);

  // Setup sine wave
  sineWave.begin(info, N_B4);
  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}
``

and works, when i dont convert the bits, also with nfc.begin(24,24); works both channels too

pschatzmann, 2024-05-15:
With sineWave(32000); you limit the generated data to 16 bits !
I guess if you change it to SineWaveGenerator<int24_t> sineWave;  which will take the full 24 bit range, it will not work...

mutedh, 2024-05-15:
Also with SineWaveGenerator<int24_t> sineWave(32000); works If i Put : AudioInfo from(48000,2,24); or nfc.begin(24,24);