pschatzmann/arduino-audio-tools #1530: I2S Sample-rate not updating (stuck at 44.1k from setup)

z-l-p, 2024-04-27:
### Problem Description

Since updating the lib to v0.9.8 (April 23 2024) I have a problem with streams whose sample-rates are not 44.1k. The I2S sample-rate is not adapting to the stream rate, so the pitch is wrong. I'm using url streams and the libHelix MP3 decoder. In the past I used setNotifyAudioChange() to alert I2S when the decoder changed. With the new updates I thought it was automatic, but it hasn't worked for me. So I switched to addNotifyAudioChange() but the problem persists.

### Device Description

ESP32 Dev Board

### Sketch

```cpp
I can reproduce the problem using the example here: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-communication/http-client/player-url-i2s/player-url-i2s.ino
You can try these URLs as examples. The 44.1 stream should sound correct. The others should play at the wrong speed, and/or run out the buffer. (If it isn't obvious, start the same stream in another player like VLC)
48k stream http://stream.srg-ssr.ch/m/rsj/mp3_128
44.1 stream http://stream0.wfmu.org/freeform-128k
32k stream http://fm939.wnyc.org/wnycfm
```

### What is your development environment

Arduino IDE 2.x

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-27:
Update the [libhelix](https://www.pschatzmann.ch/home/2024/03/27/libhelix-behind-the-scene/) version !

z-l-p, 2024-04-27:
Thanks. I had already updated libhelix. I just deleted it, re-downloaded and re-compiled and the problem persists. I also manually checked to make sure that the commit you mentioned was included in my local version. Could there be something else going on?

pschatzmann, 2024-04-27:
Can you double check with a [simpler sketch](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-communication/http-client/streams-url_mp3_helix-i2s/streams-url_mp3_helix-i2s.ino) ?

z-l-p, 2024-04-27:
That sketch shows the same problem. I'm on core 2.0.9
Here is some log output if it's helpful.
```
[I] AudioTypes.h : 127 - MP3DecoderHelix sample_rate: 32000 / channels: 2 / bits_per_sample: 16
[I] I2SStream.h : 80 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioInfo)
[I] AudioTypes.h : 127 - in: sample_rate: 32000 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 127 - out: sample_rate: 32000 / channels: 2 / bits_per_sample: 16
[I] StreamCopy.h : 147 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
```

pschatzmann, 2024-04-27:
I can reproduce the issue: it must be somewhere in the i2s update logic

pschatzmann, 2024-04-27:
Found it: I [committed a correction](https://github.com/pschatzmann/arduino-audio-tools/commit/f59e0217c96eb869de2dbdbd0c86a99e1e803d26)

z-l-p, 2024-04-27:
Confirmed that it fixes my case. You are amazing!

closed 2024-04-28 by pschatzmann