pschatzmann/arduino-audio-tools #1538: ADC-I2S  example fails to compile on a ESP32-S3

ballistRC, 2024-04-30:
### Problem Description

From the log
board_build.mcu = esp32s
tony@ASUS-WORK:~/repo/arduino-audio-tools/examples/examples-stream/streams-adc-i2s$ pio debug
******************************************************************************************************************************
Obsolete PIO Core v6.1.15a1 is used (previous was 6.1.15)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
******************************************************************************************************************************
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; board: esp32-s3-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing adafruit/Adafruit BusIO @ ^1.14.1
Unpacking  [####################################]  100%
Library Manager: Adafruit BusIO@1.16.0 has been installed!
Library Manager: Installing git+https://github.com/pschatzmann/arduino-audio-tools.git
git version 2.34.1
Cloning into '/home/tony/.platformio/.cache/tmp/pkg-installing-upi_8b0p'...
remote: Enumerating objects: 1298, done.
remote: Counting objects: 100% (1298/1298), done.
remote: Compressing objects: 100% (1008/1008), done.
remote: Total 1298 (delta 242), reused 757 (delta 172), pack-reused 0
Receiving objects: 100% (1298/1298), 8.52 MiB | 11.45 MiB/s, done.
Resolving deltas: 100% (242/242), done.
Library Manager: audio-tools@0.9.6+sha.90ec739 has been installed!
Library Manager: Resolving dependencies...
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.6.0+sha.c9c9bbe) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing I2C
Library Manager: Warning! More than one package has been found by I2C requirements:
Library Manager:  - mbed-hemmer-matthias/I2C@0.0.0+sha.266ede55f588
Library Manager:  - wh1terabbithu/I2C@1.0.3
Library Manager:  - clefa/I2C@1.0.1
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts
Unpacking 0% 10% 20% 30% 40%
Library Manager: I2C@0.0.0+sha.266ede55f588 has been installed!
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- Adafruit BusIO @ 1.16.0
|-- audio-tools @ 0.9.6+sha.90ec739
Building in debug mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Building .pio/build/esp32dev/bootloader.bin
Generating partitions .pio/build/esp32dev/partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio/build/esp32dev/libafa/Wire/Wire.cpp.o
Compiling .pio/build/esp32dev/lib3ce/SPI/SPI.cpp.o
Compiling .pio/build/esp32dev/libafa/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/esp32dev/libafa/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Compiling .pio/build/esp32dev/libafa/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFi.cpp.o
Archiving .pio/build/esp32dev/libafa/libWire.a
Indexing .pio/build/esp32dev/libafa/libWire.a
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFiAP.cpp.o
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFiClient.cpp.o
Archiving .pio/build/esp32dev/lib3ce/libSPI.a
Indexing .pio/build/esp32dev/lib3ce/libSPI.a
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFiMulti.cpp.o
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/esp32dev/lib10d/WiFi/WiFiScan.cpp.o
src/main.cpp:13:1: error: 'AnalogAudioStream' does not name a type
 AnalogAudioStream in;
 ^~~~~~~~~~~~~~~~~
src/main.cpp:15:24: error: 'in' was not declared in this scope
 StreamCopy copier(out, in); // copy in to out
                        ^~
src/main.cpp:15:24: note: suggested alternative: 'yn'
 StreamCopy copier(out, in); // copy in to out
   

### Device Description

esp32-s3, compiles fine on an esp32.

### Sketch

```cpp
[platformio]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
;board = esp32dev
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
framework = arduino
lib_deps =
         Wire
         SPI
         I2C
         adafruit/Adafruit BusIO@^1.14.1
         https://github.com/pschatzmann/arduino-audio-tools.git
;        https://www.arduino.cc/reference/en/libraries/bounce2
;        https://github.com/thaaraak/firduino
;        https://github.com/thaaraak/LiquidCrystal_I2C
;        https://github.com/thaaraak/es8388
;        https://github.com/thaaraak/Si5351Arduino

build_flags = -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

### Other Steps to Reproduce

platformio run -t cleanall
platformio debug

### What is your development environment

Platformio/emacs

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-30:
The old I2S Analog API is only supported for the ESP32!
You will need to use the new API that will only be available starting from Arduino 3.0.0.

closed 2024-04-30 by pschatzmann

ballistRC, 2024-05-01:
Ok, thanks for the info.  Not important at this stage I guess, as I am still sorting out the PCM1808 issues and was just trying to reproduce it on a different host in case.