pschatzmann/arduino-audio-tools #1539: Setting volume for MemoryStream doesn't work

SalahuddinAsh, 2024-05-01:
### Problem Description

I'm trying to control the volume of an mp3 music saved as a memorystream.

Setting volume to 1.0 works and the audio get played. but setting it to other values between 0 and 1 makes the audio not get played. I'm not sure how to control volume here.

I checked this issue # 17 but don't understand how to use it in my case. https://github.com/pschatzmann/arduino-audio-tools/issues/17 

### Device Description

esp32-s3

### Sketch

```cpp
MemoryStream mp3_notify(ding_mp3, ding_mp3_len);
I2SStream i2s;
MP3DecoderHelix helix;
EncodedAudioStream out(&i2s, &helix); // output to decoder
StreamCopy copier_notify(out, mp3_notify); // copy in to i2s
ConverterScaler<int16_t> volume(1.0, 0, 32767);

void initAudio()
{
  auto config = i2s.defaultConfig(TX_MODE);
  config.sample_rate = 48000;
  config.channels = 2;
  config.bits_per_sample = 16;
  config.i2s_format = I2S_STD_FORMAT;
  config.pin_ws = AUDIO_WS_LRC; // LRC
  config.pin_bck = AUDIO_BCK;
  config.pin_data = AUDIO_DATA;
  config.use_apll = true;
  i2s.begin(config);
  out.begin();
}

void playSound()
{
    // volume.setFactor(0.8); 
    mp3_notify.begin();
    while (mp3_notify.available())
    {
      copier_notify.copy(volume);
    }
    mp3_notify.end();
}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-01:
Why do you think any result should change if you multiply the input with 1.0  ?

If you want to adjust the volume, I suggest that you read the [corresponding Wiki](https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control)

closed 2024-05-01 by pschatzmann

SalahuddinAsh, 2024-05-01:
Thanks for your quick response.
I have no idea about the audio concept and I only use an example code from your great library.

Therefore I would be grateful if you give me a link for an example or document to explain what you mean with VolumeStream and the newly supported pipelines.

Also, does changing the input change will allow me to change the volume?

Thanks

SalahuddinAsh, 2024-05-01:
I'll try going through your blog to learn more about the library.