pschatzmann/arduino-audio-tools #1541: Unable to use I2SCodecStream or AudioBoardStream Lyra-t Mini driver

neilshevlin, 2024-05-02:
### Problem Description

I am experiencing an issue when using either AudioBoardStream or I2SCodecStream class with the ESP32 lyrat-mini board. The problem arises specifically when calling the kit.begin() or i2s.begin() for either class.  

When I include the kit.begin() method in the setup, the debugging logs show something like the following output:
```
[D] I2SCodecStream.h : 88 - virtual bool audio_tools::I2SCodecStream::begin(audio_tools::I2SCodecConfig)
[D] I2SCodecStream.h : 257 - input: 2
[D] I2SCodecStream.h : 258 - output: 3
[D] I2SCodecStream.h : 273 - BIT_LENGTH_16BITS
[D] I2SCodecStream
```
Logging will nearly always cut out like this. With a different audio stream driver logs should look something like this. 

```
15:20:46.169 > [D] I2SCodecStream.h : 264 - input: 1
15:20:46.174 > [D] I2SCodecStream.h : 265 - output: 3
15:20:46.174 > [D] I2SCodecStream.h : 280 - BIT_LENGTH_16BITS
15:20:46.180 > [D] I2SCodecStream.h : 314 - RATE_44K
15:20:46.185 > [D] I2SCodecStream.h : 329 - I2S_NORMAL
15:20:46.185 > Warning: Pin '13' not set up because of conflict
15:20:46.191 > [    78][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=18 scl=23 freq=100000
15:20:46.216 > [D] I2SStream.h : 65 - bool audio_tools::I2SStream::begin(audio_tools::I2SConfig)
(... continues as usual...)
```
Nothing in the sketch will execute after this point. removal of the kit.begin() from setup() will allow execution of the following lines. 

The issue is most likely to do with the I2SCodecStream since it is used by the AudioBoardStream class. 

### Device Description

I am using an ESP32 LyraT-Mini-V1.2. The board has the same 8311 and 7243 codec chips as stated in the lyrat mini driver class. 

### Sketch

```cpp
// This is just pulled from your examples library. Applies to all sketches I have had where I am using AudioBoardStream

#include "AudioTools.h"
#include "AudioLibs/AudioBoardStream.h"

AudioBoardStream kit(LyratMini);
StreamCopy copier(kit, kit);

void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Debug);
    
    auto cfg = kit.defaultConfig(RXTX_MODE);
    cfg.sd_active = false;
    cfg.input_device = ADC_INPUT_LINE2;
    kit.begin(cfg);// exclude this line and you won't get this crash
}

void loop() {
    copier.copy();
}
```
```

### Other Steps to Reproduce

The same code above works as expected on `AudioKitEs8388V1` and `LyraTV43` boards. However fails to work on lyra t mini. 

### What is your development environment

Using both platformio and arduino IDE; issue replicates on both. Flashing from a Mac OS ARM

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-02:
Since I do not have any such board you will need to do the debugging yourself.
- Activate the [logging on the AudioDriver](https://github.com/pschatzmann/arduino-audio-driver/wiki/Logging)
- I would suggest to do some separate tests for TX_MODE and RX_MODE in order to identify where the issue is

pschatzmann, 2024-11-02:
To address this issue, I ordered a Lyrat Mini.
I committed some corrections, but I have some mixed results:
- The Output is working properly now
- I extended the logic so that the pins can be handled as well.

On the negative side:
- My SD card does not work: I would be greatful if you can reproduce this issue on another board
- The audio input provides some kind of signal, but it does not make any sense to me

Here is the link to the [demo test cases](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-custom-boards/lyrat-mini).

I am a bit puzzeled: the regular LyraT was quite easy to implement, quite unlike this night mare!

closed 2024-11-02 by pschatzmann

pschatzmann, 2024-11-03:
I finally manage to have the SD drive working via SDMMC!