pschatzmann/arduino-audio-tools #1548: Compilation errors when using Arduino ESP32 3.0.0-rc1

thegoodhen, 2024-05-05:
### Problem Description

I am not sure if this is an issue, since the 3.0.0-rc1 is still technically unstable, but just in case the problem persists till stable release - the compilation doesn't work, but it can be fixed by adding 

#include <WiFiClient.h>
#include <WiFi.h>

below #if defined(ESP32) inside URLStream.h

I can submit a pull request if you want.

### Device Description

NodeMCU32S

### Sketch

```cpp
(tested with streams-generator-pwm.ino example, but likely will be the same for any code including the library)
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-05:
I can't reproduce any issue. I guess you are still using an old version of the library!

closed 2024-05-05 by pschatzmann

thegoodhen, 2024-05-05:
Sorry, yes, confirming. I just realised that, too.