pschatzmann/arduino-audio-tools #1549: No pwm output  on ESP32 v. 3.0.0-rc1

thegoodhen, 2024-05-05:
### Problem Description

The streams-generator-pwm.ino example works just fine on version 2.x, but on 3.x it produces no output with the same code, IDE and hardware configuration. I will investigate further.

### Device Description

Nodemcu 32S

### Sketch

```cpp
#include "AudioTools.h"

AudioInfo info(8000, 1, 16);
SineWaveGenerator<int16_t> sineWave(32000); // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound(sineWave);  // Stream generated from sine wave
PWMAudioOutput pwm;                  
StreamCopy copier(pwm, sound);    // copy in to out

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);  

  // setup sine wave
  sineWave.begin(info, N_B4);

  // setup PWM output
  auto config = pwm.defaultConfig();
  config.copyFrom(info);
  //config.resolution = 8;  // must be between 8 and 11 -> drives pwm frequency (8 is default)
  // alternative 1
  //config.start_pin = 3;
  // alternative 2
  //int pins[] = {3};
  // alternative 3
  //Pins pins = {3};
  //config.setPins(pins); 
  pwm.begin(config);
}

void loop(){
  copier.copy();
}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-05:
Are you sure that you are not hunting any problems which have been resolved already ?
Just make sure you use the latest main branch

thegoodhen, 2024-05-05:
> Are you sure that you are not hunting any problems which have been resolved already ? Just make sure you use the latest main branch

Yeah, I just double-triple checked. Sorry about the false alarm on the previous issue. But this time I am actually sure. I just updated to the main branch and still no audio output... I can test with a different HW later if you'd like.

thegoodhen, 2024-05-05:
I read the code and did some tests and so far it seems like the ledc controller is getting attached properly. 

============ Before Setup End ============
[   706][V][esp32-hal-uart.c:396] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   715][V][esp32-hal-uart.c:482] uartBegin(): UART0 not installed. Starting installation
[   725][V][esp32-hal-uart.c:525] uartBegin(): UART0 initialization done.
[   741][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type LEDC (12) successfully set to 0x400dac44
[   752][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 12 successfully set to type LEDC (12) with bus 0x3ffb8b98
[   763][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 12 (channel 0, resolution 8)
=========== After Setup Start ============

I am not getting the debug prints (LOGI, etc.) for some reason, not yet sure how to turn them on, but I'll figure that out or just temporarily replace them with printf statements. Setting the core debug level to verbose did not help...

When I tried putting print statements into the timer callback, they crash the ESP (not entirely surprising, since it's an ISR), but that seems to suggest the timer is firing... I will need to do a minimum ledc test to see what's wrong and debug it further. 

I was suspecting that the default output pin is different for idf 4 and 5 for some reason, but that doesn't seem to be the case either. I hav work tomorrow, but I'll try figuring it out. It's pretty odd.

pschatzmann, 2024-05-06:
I think I have found it: the [ledWrite needed the channel in the prior version, but now it requires the pin](https://github.com/pschatzmann/arduino-audio-tools/commit/5e6e472df990029e599e8839aa0ce51c1b123b93).

I did not have the chance to test the correction yet...

thegoodhen, 2024-05-06:
Okay, thank you! That works! There is some crackling in the tone, but I have yet to determine if this is a hardware or a software issue. Thank you very much.

closed 2024-06-18 by pschatzmann