pschatzmann/arduino-audio-tools #1550: URL Stream not working

duevil, 2024-05-05:
### Problem Description

Since the last update (v0.9.8 or v.0.8.3 for Mp3Helix) I cannot play certain URLs using the URLStream class which did work flawlessly in version 0.9.7. The log shows a successfull connection, however no data is received.

I have this problem when using the new Pipeline example sketch as well as using an older approach.

The log when trying to play the URL that's not working:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[    30][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   255][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[I] URLStream.h : 82 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): https://streams.radio21.de/aurich/mp3-128
[I] Url.h : 58 - Url::parse
[I] Url.h : 96 - url->https://streams.radio21.de/aurich/mp3-128
[I] Url.h : 97 - host->streams.radio21.de
[I] Url.h : 98 - protocol->https
[I] Url.h : 99 - path->/aurich/mp3-128
[I] Url.h : 100 - port->443
[I] URLStream.h : 400 - bool audio_tools::URLStream::login()
[   334][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   405][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
.[   466][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   485][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   485][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.178.32, MASK: 255.255.255.0, GW: 192.168.178.1

[I] URLStream.h : 367 - WiFiClientSecure
[I] HttpRequest.h : 222 - process connecting to host streams.radio21.de port 443
[  2009][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[I] HttpRequest.h : 337 - connected 1 timeout 60000
[I] HttpRequest.h : 233 - Free heap: 180476
[I] HttpHeader.h : 276 - HttpHeader::write
[I] HttpHeader.h : 421 - -> GET /aurich/mp3-128 HTTP/1.1
[I] HttpHeader.h : 210 -  -> Host: streams.radio21.de 
[I] HttpHeader.h : 210 -  -> Connection: keep-alive 
[I] HttpHeader.h : 210 -  -> Accept-Encoding: identity 
[I] HttpHeader.h : 210 -  -> Accept: audio/mp3 
[I] HttpHeader.h : 342 -  -> <CR LF> 
[I] HttpRequest.h : 288 - Request written ... waiting for reply
[I] HttpHeader.h : 253 - Data available: 479
[W] URLStream.h : 342 - Redirected to: https://radio21.streamabc.net/radio21-aurich-mp3-128-9097805?sABC=6637s98p%230%23sqs1p198n899651q81933462qq3652q8%23&aw_0_1st.playerid=&amsparams=playerid:;skey:1714944396
[I] Url.h : 58 - Url::parse
[I] Url.h : 96 - url->https://radio21.streamabc.net/radio21-aurich-mp3-128-9097805?sABC=6637s98p%230%23sqs1p198n899651q81933462qq3652q8%23&aw_0_1st.playerid=&amsparams=playerid:;skey:1714944396
[I] Url.h : 97 - host->radio21.streamabc.net
[I] Url.h : 98 - protocol->https
[I] Url.h : 99 - path->/radio21-aurich-mp3-128-9097805?sABC=6637s98p%230%23sqs1p198n899651q81933462qq3652q8%23&aw_0_1st.playerid=&amsparams=playerid:;skey:1714944396
[I] Url.h : 100 - port->443
[I] URLStream.h : 367 - WiFiClientSecure
[I] HttpRequest.h : 222 - process connecting to host radio21.streamabc.net port 443
[  2870][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[I] HttpRequest.h : 337 - connected 1 timeout 60000
[I] HttpRequest.h : 233 - Free heap: 182776
[I] HttpHeader.h : 276 - HttpHeader::write
[I] HttpHeader.h : 421 - -> GET /radio21-aurich-mp3-128-9097805?sABC=6637s98p%230%23sqs1p198n899651q81933462qq3652q8%23&aw_0_1st.playerid=&amsparams=playerid:;skey:1714944396 HTTP/1.1
[I] HttpHeader.h : 210 -  -> Host: radio21.streamabc.net 
[I] HttpHeader.h : 210 -  -> Connection: keep-alive 
[I] HttpHeader.h : 210 -  -> Accept-Encoding: identity 
[I] HttpHeader.h : 210 -  -> Accept: audio/mp3 
[I] HttpHeader.h : 342 -  -> <CR LF> 
[I] HttpRequest.h : 288 - Request written ... waiting for reply
[I] HttpHeader.h : 244 - Waiting for data...
[I] HttpHeader.h : 253 - Data available: 666
[I] HttpRequest.h : 164 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 90 - size: 0
[... 100 lines cut ...]
    pipeline.add(volume);
    pipeline.add(bits);
    pipeline.setOutput(i2s);
    // start all components with their default values
    pipeline.begin();
}

void loop(){
    copier.copy();
}
```

### What is your development environment

PlatformIO

my lib_deps:
```
    https://github.com/pschatzmann/arduino-audio-tools#v0.9.8
    https://github.com/pschatzmann/arduino-libhelix#v.0.8.3
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-06:
Please don't report any problems that have already been corrected.
I can't reproduce your issue with the latest code from the master branch

closed 2024-05-06 by pschatzmann

duevil, 2024-05-06:
> Please don't report any problems that have already been corrected! I can't reproduce your issue with the latest code from the master branch...
> 
> Is there any reason why you convert to 32 bits ?

How was the problem corrected? I am using the newest version of the library and the problem still occurs. It seems to be caused by using a https connection. When changing the url to use a plain http connection I can get the stream  to work. Is this intended? In the previous version I also did use https and had no problems.

And no, there is no reason for why I use 32 bits, I simply copied the Pipeline example for testing

pschatzmann, 2024-05-06:
As far as I can see the issue is caused by the 32 bit conversion which is not notifying the i2s. 
It is corrected in the main branch, so please use this!

duevil, 2024-05-06:
> As far as I can see the issue is caused by the 32 bit conversion which is not notifying the i2s. It is corrected in the main branch, so please use this!

As far as I know I am using the main branch. And the problem persists without using the 23 bit conversion, so that is not the cause. I have testet this with a new project with the same pipeline implemented in my main project. The log shows the problem: When connecting using http, everything is fine, however when trying to connect using https the error occurs. In the test I have called the stream begin inside a Ticker callback which causes the ESP to crash, however when called in the setup the problem remains, though no crash occurs but the stream won't provide any data.

### Sketch

```#include <Arduino.h>
#include <AudioTools.h>
#include <AudioLibs/MemoryManager.h>
#include <AudioCodecs/CodecMP3Helix.h>
#include <Ticker.h>

#define SSID ""
#define PASS ""

constexpr auto PIN_DATA = GPIO_NUM_12;
constexpr auto PIN_BCK = GPIO_NUM_27;
constexpr auto PIN_LRC = GPIO_NUM_33;

MemoryManager mem{};
I2SStream i2s{};
AdapterAudioStreamToAudioOutput i2sOut{i2s};
VolumeOutput volumeOut{};
MultiOutput i2sAndVolumeOut{i2sOut, volumeOut};
FadeStream fadeStream{};
VolumeStream volumeStream{};
EncodedAudioOutput decodedOut{new MP3DecoderHelix()};
Pipeline pipeline{};
StreamCopy copier;
ICYStream icy{};

Ticker ticker;

static void metaData(MetaDataType type, const char *data, int) { log_v("Metadata: [%s] %s", toStr(type), data); }

struct : ModifyingOutput {
    Print *out = nullptr;

    void setOutput(Print &output) override { out = &output; }

    size_t write(const uint8_t *buffer, size_t size) override {
        if (out) {
            return out->write(buffer, size);
        }
        return size;
    }

    void setAudioInfo(AudioInfo newInfo) override {
        ModifyingOutput::setAudioInfo(newInfo);
        log_i("Audio info: sample rate: %d, bits per sample: %d, channels: %d",
              newInfo.sample_rate, newInfo.bits_per_sample, newInfo.channels);
    }
} info;

void play(const char *url) {
    log_i("Playing URL: %s", url);
    icy.begin(url);
    copier.begin(pipeline, icy);
    copier.setActive(true);
    fadeStream.setFadeInActive(true);
}

void setup() {
    Serial.begin(SERIAL_BAUD_RATE);

    WiFi.begin(SSID, PASS);
    while (WiFi.status() != WL_CONNECTED) {
        delay(100);
        log_d("Connecting to WiFi...");
    }
[... 194 lines cut ...]
  #18 0x4012a82d:0x3ffbaf60 in ecdh_gen_public_restartable at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdh.c:89
      (inlined by) mbedtls_ecdh_gen_public at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdh.c:107
  #19 0x4012aa34:0x3ffbaf90 in ecdh_make_public_internal at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdh.c:558
      (inlined by) mbedtls_ecdh_make_public at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdh.c:587
  #20 0x40125484:0x3ffbafb0 in ssl_write_client_key_exchange at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_cli.c:3725
      (inlined by) mbedtls_ssl_handshake_client_step at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_cli.c:4347
  #21 0x40122eb5:0x3ffbb050 in mbedtls_ssl_handshake_step at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_tls.c:5770
  #22 0x40122eef:0x3ffbb070 in mbedtls_ssl_handshake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_tls.c:5808
  #23 0x400e65ed:0x3ffbb090 in start_ssl_client(sslclient_context*, IPAddress const&, unsigned int, char const*, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**) at C:/Users/Malte/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:271
  #24 0x400e5607:0x3ffbb320 in WiFiClientSecure::connect(IPAddress, unsigned short, char const*, char const*, char const*, char const*) at C:/Users/Malte/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:141
  #25 0x400e56ce:0x3ffbb370 in WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at C:/Users/Malte/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:136
  #26 0x400e57aa:0x3ffbb3b0 in WiFiClientSecure::connect(char const*, unsigned short) at C:/Users/Malte/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:117
      (inlined by) WiFiClientSecure::connect(char const*, unsigned short) at C:/Users/Malte/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:113
  #27 0x400d50e0:0x3ffbb3d0 in audio_tools::HttpRequest::connect(char const*, unsigned short, int) at .pio/libdeps/adafruit_feather_esp32_v2/audio-tools/src/AudioHttp/HttpRequest.h:336
  #28 0x400db565:0x3ffbb3f0 in audio_tools::HttpRequest::processBegin(MethodID, audio_tools::Url&, char const*, int) at .pio/libdeps/adafruit_feather_esp32_v2/audio-tools/src/AudioHttp/HttpRequest.h:223 (discriminator 3)
[... cut ...]
  #32 0x400dae0b:0x3ffbb4c0 in audio_tools::ICYStream::begin(char const*, char const*, MethodID, char const*, char const*) at .pio/libdeps/adafruit_feather_esp32_v2/audio-tools/src/AudioHttp/ICYStream.h:61 (discriminator 3)
  #33 0x400daeba:0x3ffbb500 in play(char const*) at src/main.cpp:54
  #34 0x400daf4b:0x3ffbb520 in setup()::{lambda()#1}::_FUN() at src/main.cpp:110
      (inlined by) _FUN at src/main.cpp:114
  #35 0x401158f7:0x3ffbb540 in timer_process_alarm at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_timer/src/esp_timer.c:396
      (inlined by) timer_task at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_timer/src/esp_timer.c:422
```

pschatzmann, 2024-05-06:
To use the main branch you will need to change the 
    https://github.com/pschatzmann/arduino-audio-tools#v0.9.8
    https://github.com/pschatzmann/arduino-libhelix#v.0.8.3
to
    https://github.com/pschatzmann/arduino-audio-tools
    https://github.com/pschatzmann/arduino-libhelix

duevil, 2024-05-06:
> To use the main branch you will need to change the https://github.com/pschatzmann/arduino-audio-tools#v0.9.8 https://github.com/pschatzmann/arduino-libhelix#v.0.8.3 to https://github.com/pschatzmann/arduino-audio-tools https://github.com/pschatzmann/arduino-libhelix
> 
> What ESP32 core version are you using ? The actual one is 2.0.16

The problem still persist when omitting the version number. 

As for the core version: The latest platfom version is 6.6.0, [which supports the core version 2.0.14](https://github.com/platformio/platform-espressif32/releases/tag/v6.6.0). As far as I know there is no other way to change the core version but for changing the platform version.

pschatzmann, 2024-05-06:
Did you try to delete the .pio folder to trigger a reload of the libraries ? I am not sure if just updating the link is good enough.
Also go back to the initial version of your sketch. I don't want to work on an issue where you constantly change the sketch and potentially introduce other bugs.