pschatzmann/arduino-audio-tools #1570: InputMerge stream not working on I2S (resolved this)

fswart, 2024-05-23:
### Problem Description

After upgrade (I think) to version 0.9.6 the InputMerge function for I2S did not work anymore; it forces the ESP processor to a panic reboot. Looks like a nullptr operation.

After thorough investigation I found in class InputMerge the following protected declarations:
Vector<Stream*> streams{10};
Vector<float> weights{10}; 

With these, the input- and output-streams are added as 11 and 12 while the available() method tries to read stream[0]. Which is a nullptr and cause (again, I think) of the panic.
I changed the declaration to the following and this made things work. Not sure if it is a proper fix or just a workaround for my application.
Vector<Stream*> streams{0};
Vector<float> weights{0}; 
 

### Device Description

Tried several e.g. ESP32DevkitC, Wemos D1 esp32

### Sketch

```cpp
The example provided to merge sinewaves into PCM was rewritten to use I2S, standard stuff:
InputMerge<sound_t> merger;
I2SStream out; 
StreamCopy copier(out, merger);  

And in Setup():
auto config = out.defaultConfig(TX_MODE);
  config.copyFrom(info_out); 
  // custom pinning
  config.pin_bck = 15;                              // default 14 - bit clock BCLK or BCK
  config.pin_ws = 17;                               // default 15 - word select WS or LRCK
  config.pin_data = 16;                             // default 22 - data DIN
  out.begin(config);
```

### Other Steps to Reproduce

Tried in PlatformIO 6.1.15 and Arduino IDE 2.3.2

### What is your development environment

Mainly PlatformIO 6.1.15

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-23:
Thanks for letting me know.
I changed the constructor for the Vector to define only the capacity and not the size.
This should optimize the number of necessary allocations to 1 if you are merging less then 10 streams

closed 2024-05-23 by pschatzmann

fswart, 2024-05-23:
That was REALLY quick!
Thanks fort he update.

Best regards,
Fred