pschatzmann/arduino-audio-tools #1598: setSilenceOnInactive() not working with audioplayer and output mixer

itman496, 2024-06-18:
### Problem Description

I am trying to play two WAV files simultaneously off of an SDMMC card source.

One is a narration track that plays a WAV file of someone speaking.  This file will change periodically and be triggered and then let to play and then go quiet and at some arbitrary point will be triggered again to play something else.

The other is the background music.  This will be looping over and over and over again, fading a little softer into the background when the narration player is active, then rising back up again when its just by itself.

I'm having a lot of issues, but a core simple one is when both players play, it sounds and mixes fine.  When the narration player stops, the playback over i2s suddenly gets choppy and jittery.

From what I understand, should player.setSilenceOnInactive make it output 0's and still contribute silence to the mixer buffer?

Or am I fundamentally misunderstanding how this all works?

Here is a stripped down example that plays perfectly until the first file stops playing, and then it starts freaking out.

### Device Description

ESP32-WROVER with 4MB PSRAM

### Sketch

```cpp
//testing for audio glitches

#include "AudioTools.h"
#include "AudioLibs/AudioSourceSDMMC.h"

unsigned long lastMillis;

const char* ext = "wav";

AudioSourceSDMMC source("/System", ext);
AudioSourceSDMMC bgmSource("/bgm", ext);

I2SStream i2s;

WAVDecoder decoder;
WAVDecoder decoder2;

OutputMixer<int16_t> mixer(i2s, 2);

AudioPlayer player(source, mixer, decoder);
AudioPlayer bgmPlayer(bgmSource, mixer, decoder2);

AudioInfo info(44100, 2, 16);
//SineWaveGenerator<int16_t> sineWave1(0);                // subclass of SoundGenerator with max amplitude of 32000
//GeneratedSoundStream<int16_t> sound1(sineWave1);             // Stream generated from sine wave
//StreamCopy copier1(mixer, sound1);                          // copies sound into mixer

void setup() {

  Serial.begin(115200);
  delay(1000);
  Serial.println("INITIALIZING");
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = i2s.defaultConfig(TX_MODE);
  cfg.pin_bck = 27;
  cfg.pin_ws = 26;
  cfg.pin_data = 25;
  //cfg.buffer_count = 8;
  //cfg.buffer_size = 1024;
  i2s.begin(cfg);

//  sineWave1.begin(info, N_B4);
  
  mixer.begin(4096);
  
  // setup player
  source.setFileFilter("*wp0*");

  player.setVolume(0.9);
  player.setAutoNext(false);
  player.setSilenceOnInactive(true);
  player.begin();

  bgmSource.setFileFilter("ps1*");

  bgmPlayer.setVolume(0.2);
  bgmPlayer.setSilenceOnInactive(true);
  bgmPlayer.begin();
  

  mixer.setWeight(0,1.0);
  mixer.setWeight(1,1.0);

}

void loop() {
  player.copy();
  bgmPlayer.copy();

  mixer.flushMixer();
 
}
```

### What is your development environment

Arduino IDE

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-06-18:
Did you try with [setTimeoutAutoNext](https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_audio_source.html#a1d03133fe86ead67820854ef63dadfbd) (int millisec) to 0 ?

itman496, 2024-06-18:
I have tried adding your change and it still glitches out.

Here is a little snippet of video/audio I snapped.

If there is a better way to do this, I would love to know..  I'm a little out of ideas haha, I very much appreciate the assistance.

Do you know a better way to achieve this effect of 1 looping audioplayer and 1 that changes its file filter and gets retriggered periodically on demand, and then mixing them into an output?

https://www.dropbox.com/scl/fi/uqsgmb9og3dhfd85aoa6d/2024-06-18-01.31.23.mp4?rlkey=z6ri3skgvgsg28j3ozx4m11jv&dl=0

pschatzmann, 2024-06-18:
If you just stick with 1 music file you can try using the FileLoop class

itman496, 2024-06-18:
So I'd use the audioplayer for the file that changes, and fileloop for the background music file, and send both their outputs to the outputmixer?

pschatzmann, 2024-06-18:
That would be worth a trial.

itman496, 2024-06-18:
Okay, I will give implementing that a try, stand by.  Again, very much appreciate your assistance!  

Out of curiosity, how hard would it be to achieve a thing where the audioplayer when hitting the end of the file, just returns zeros when the player.copy(); is called instead of nothing?

pschatzmann, 2024-06-18:
I think the best way would be to check the result of the copy command and if it is less then the expected bytes fill the difference with writeSilence().

itman496, 2024-06-18:
I feel I maybe don't understand quite what you mean, I tried this:

`void loop() {
  if(player.copy() < 1023) {
    writeSilence();
  }
  bgmPlayer.copy();

  mixer.flushMixer();
 
}`

But it just says writeSilence was not declared in this scope.  I took a look on the reference page but I couldn't find a function called writeSilence() ?

pschatzmann, 2024-06-18:
I have added the writeSilence() to the OutputMixer:

```
  size_t n1 = player.copy();
  mixer.writeSilence(0, 1024 - n1);
  
  size_t n2 = bgmPlayer.copy();
  mixer.writeSilence(1, 1024 - n2);
```

itman496, 2024-06-18:
Oh, that makes a lot more sense!  Is there something I need to modify elsewhere to allow this to work?  On compile I got an error:

`error: 'class audio_tools::OutputMixer<short int>' has no member named 'writeSilence'`

pschatzmann, 2024-06-18:
You need to update the AudioTools library

itman496, 2024-06-18:
Oh!  Thats what you meant when you said you added it.  Stand by! I will update.

itman496, 2024-06-18:
I appear to be getting another error after updating to the latest version:

```
452:10: error: 'size_t audio_tools::OutputMixer<T>::writeSilence(int) [with T = short int; size_t = unsigned int]' marked 'override', but does not override
   size_t writeSilence(int bytes) override {
```

pschatzmann, 2024-06-18:
Oh, I see: it inherits form Print!
The correction has been committed...

itman496, 2024-06-18:
That actually does something!  But it seems to be having some more issues.

It mixes the two perfectly at first as long as both are playing, when player finishes the file I hear a pop and then it goes silent, and my console gets filled with spam of:

`[W] AudioOutput.h : 387 - Available Buffer too small 480: requested: 512 -> increase the buffer size`

I've tried changing mixer.begin() to have 10000 instead of 4096.  Is there somewhere else I should be changing this?  I'm not sure which buffer it's referring to.

pschatzmann, 2024-06-18:
Did you remove the setSilenceOnInactive(true) ?

itman496, 2024-06-18:
Yes, this is currently the entire program, just to be sure:

```
//testing for audio glitches

#include "AudioTools.h"
#include "AudioLibs/AudioSourceSDMMC.h"

unsigned long lastMillis;

const char* ext = "wav";

AudioSourceSDMMC source("/System", ext);
AudioSourceSDMMC bgmSource("/bgm", ext);

I2SStream i2s;

WAVDecoder decoder;
WAVDecoder decoder2;

OutputMixer<int16_t> mixer(i2s, 2);

AudioPlayer player(source, mixer, decoder);
AudioPlayer bgmPlayer(bgmSource, mixer, decoder2);

AudioInfo info(44100, 2, 16);
//SineWaveGenerator<int16_t> sineWave1(0);                // subclass of SoundGenerator with max amplitude of 32000
//GeneratedSoundStream<int16_t> sound1(sineWave1);             // Stream generated from sine wave
//StreamCopy copier1(mixer, sound1);                          // copies sound into mixer

void setup() {

  Serial.begin(115200);
  delay(1000);
  Serial.println("INITIALIZING");
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = i2s.defaultConfig(TX_MODE);
  cfg.pin_bck = 27;
  cfg.pin_ws = 26;
  cfg.pin_data = 25;
  //cfg.buffer_count = 8;
  //cfg.buffer_size = 1024;
  i2s.begin(cfg);

//  sineWave1.begin(info, N_B4);
  
  mixer.begin(4096);
  
  // setup player
  source.setFileFilter("*wp0*");
  source.setTimeoutAutoNext(0);

  player.setVolume(0.9);
  player.setAutoNext(false);
  player.begin();

  bgmSource.setFileFilter("no gravity*");
  bgmSource.setTimeoutAutoNext(0);

  bgmPlayer.setVolume(0.2);
  bgmPlayer.begin();
  

  mixer.setWeight(0,1.0);
  mixer.setWeight(1,1.0);

}

void loop() {
  size_t n1 = player.copy();
  mixer.writeSilence(0, 1024 - n1);
  
  size_t n2 = bgmPlayer.copy();
  mixer.writeSilence(1, 1024 - n2);

  mixer.flushMixer();
 
}
```

itman496, 2024-06-18:
I've tried increasing both buffers to massive levels, (i2s buffer increased to 32 counts of 1024 size) and mixer.begin changed to 100,000.  No affect, when player.copy() returns less than 1024 and it switches to playing silence only on mixer channel zero, the whole system grinds to a halt with a spam of
```
11:15:21.662 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.662 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.662 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.694 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.694 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.694 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.727 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.727 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.760 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size
11:15:21.760 -> [W] AudioOutput.h : 387 - Available Buffer too small 88: requested: 512 -> increase the buffer size

```

pschatzmann, 2024-06-18:
I dont think that the AudioPlayer is a good source for the Output Mixer:
The only way I could make it work is by switching off the automatic index management with   mixer.setAutoIndex(false);
And then fill the mixer with something like this

```
void fillBuffer(AudioPlayer& p, int idx, int size) {
  LOGI("%d", idx);
  mixer.setIndex(idx);

  size_t to_write = mixer.availableForWrite(idx);
  bool written = false;
  int open = size;
  if (to_write >= size) {
    size_t bytes1 = p.copy(size);
    open = size - bytes1;
    //LOGW("idx %d  %d -> %d", idx, to_write, bytes1);
  }

  if (open > 0) {
    mixer.writeSilence(idx, open);
    LOGW("silence %d %d",idx, open);
  }
}

void loop() {
  fillBuffer(bgmPlayer, 0, 1024);
  fillBuffer(player, 1, mixer.available(0) - mixer.available(1));
  mixer.flushMixer();
}
```
Any change of a file will take quite some time during which there will be no output at all.
I think an InputMixer together with a FileLoop and/or CatStream will give a better result.

itman496, 2024-06-18:
How would you implement that to achieve a similar effect to the example I had above?  I will admit I tried to do an input mixer in the past and it was a little confusing..  But I will try again!

pschatzmann, 2024-06-19:
Did you try with SD_MMC.begin("/sdcard", true) ? 

I suggest that you use the SDMMC test examples provided by Arduino ESP32 first

closed 2024-06-21 by pschatzmann