pschatzmann/arduino-audio-tools #1606: #include "AudioAnalog/AnalogAudioBase.h"

keko30388, 2024-06-24:
### Problem Description

i building snapcast client and i got this error while building code

### Device Description

ESP32

### Sketch

```cpp
In file included from C:\Users\keko3\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogAudioStream.h:8,
                 from C:\Users\keko3\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools.h:78,
                 from S:\New folder\SnapClient\api\SnapOutput.h:7,
                 from S:\New folder\SnapClient\SnapClient.ino:22:
C:\Users\keko3\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogDriverESP32.h:5:10: fatal error: AudioAnalog/AnalogAudioBase.h: No such file or directory
 #include "AudioAnalog/AnalogAudioBase.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

### What is your development environment

Arduino IDE

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-06-24:
Take it back... I am getting different errors now

keko30388, 2024-06-24:
> I could not reproduce your issue with the ESP32 Arduino 3.0.1 Core and the
> latest AudioTools.
> What core are you using ?
> Try to reinstall the latest corrections of the AudioTools

im using esp32 by espressif 2.0.17 bcs on 3.0.1 IR dont work

pschatzmann, 2024-06-24:
I committed a correction to the AudioTools to correct the 2.0.17.
What is IR ?

keko30388, 2024-06-24:
infrared sensor (PIR sensor)

Does that mean it will work now?

pschatzmann, 2024-06-24:
Yes, I also committed a correction to the SnapClient to address the changed API for MDNS in 3.0.1

keko30388, 2024-06-24:
thank you very much also for fast response <3