pschatzmann/arduino-audio-tools #1624: FLAC encoder unavailable

meteor-LZX, 2024-07-22:
### Problem Description

Hello, Dear pschatzmann :

Recently, I used customized esp32s3-n16r8 board that connected a small MEMS digital microphone which owned i2s interface.
I would like to use it to record audio and then send it into a webserver through WiFi.

After searching for relevant information on the Internet, I found that data with high sampling rate is easy to lose data packets when transmitted by WiFi, but part of the information will be lost if lossy compression algorithm is used. Therefore, I decided to use lossless compression method (FLAC) to process the data and then send it to the WebServer over WiFi. (I'm not sure if this is the right idea.)

I referred to the example that you showed on **examples/tests/codecs/test-codec-flac/test-codec-flac.ino** and **examples/examples-communication/http-server/streams-i2s-webserver_wav/streams-i2s-webserver_wav.ino**, But when I use the flac encoder, the terminal will output the following error :
**[E] CodecFLAC.h : 421 - write_callback 4 -> 0
[E] CodecFLAC.h : 337 - ERROR: initializing decoder: FLAC__STREAM_ENCODER_INIT_STATUS_ENCODER_ERROR
[E] CodecFLAC.h : 339 -  -> FLAC__STREAM_ENCODER_CLIENT_ERROR
[E] AudioServer.h : 366 - encoder begin failed**

So I go to **stream_encoder.h** to find **FLAC__STREAM_ENCODER_INIT_STATUS_ENCODER_ERROR**, and call **FLAC__stream_encoder_get_state()** as instructed. Finally I got the **FLAC__STREAM_ENCODER_UNINITIALIZED** error. It seems to mean that the encoder is not initialized. But I found that your **CodecFLAC.h** already calls the **FLAC__stream_encoder_init_stream()** function to initialize the encoder in line 334.

I'm so confused...Do you have any suggestions or solutions?

Thank you!

### Device Description

My custom board, on board module is ESP32S3-N16R8.

### Sketch

```cpp
#include <Arduino.h>

#include "AudioTools.h"
#include "AudioCodecs/CodecFLAC.h"

// Microphone
static const uint8_t mic_bck = GPIO_NUM_21;
static const uint8_t mic_ws = GPIO_NUM_47;
static const uint8_t mic_din = GPIO_NUM_48;
static const uint8_t mic_lr = GPIO_NUM_45;  // Channel the data into --- 0:left, 1:right
// WiFi
static const char *ssid = "meteorrrrrr";
static const char *psw = "5xR0156mm";

// Audio settings
FLACEncoder encoder;
AudioEncoderServer server(&encoder, ssid, psw);
I2SStream in_stream;
ConverterFillLeftAndRight<int16_t> filler(RightIsEmpty); // fill both channels - or change to RightIsEmpty

void setup() {
    Serial0.begin(115200);
    AudioLogger::instance().begin(Serial0, AudioLogger::Info);

    // Setup microphone & i2s
    pinMode(mic_lr, OUTPUT);
    digitalWrite(mic_lr, LOW);    // Use left
    auto cfg = in_stream.defaultConfig(RX_MODE);
    cfg.i2s_format = I2S_STD_FORMAT;
    cfg.pin_bck = mic_bck;
    cfg.pin_ws = mic_ws;
    cfg.pin_data = mic_din;
    cfg.sample_rate = 44100;
    cfg.bits_per_sample = 16;
    cfg.channels = 2;
    in_stream.begin(cfg);
    Serial0.println("I2S started!");

    // start data sink
    server.begin(in_stream, cfg, &filler);
}

void loop() {
    server.copy();
}
```

### Other Steps to Reproduce

The entire output of terminal :

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   217][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] I2SConfigESP32.h : 80 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Master
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 16
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: false
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 103 - pin_bck: 21
[I] I2SConfigESP32.h : 105 - pin_ws: 47
[I] I2SConfigESP32.h : 107 - pin_data: 48
I2S started!
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[E] CodecFLAC.h : 421 - write_callback 4 -> 0
[E] CodecFLAC.h : 337 - ERROR: initializing decoder: FLAC__STREAM_ENCODER_INIT_STATUS_ENCODER_ERROR
[E] CodecFLAC.h : 339 -  -> FLAC__STREAM_ENCODER_CLIENT_ERROR
[E] AudioServer.h : 366 - encoder begin failed

**!!!**
**But when I changed FLACEncoder class to WAVEncoder class, everything was fine and the system worked correctly...**

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-07-22:
Did you try the FLAC test sketch to confirm that the codec is working properly ?
If this is working I would guess that you just don't have enough RAM for Wifi and this codec.

I would recommend to stick with the WAV and just reduce the sample rate and channels. There is no point of doing 44100 and you will not hear any difference to using just 8000 when using these cheap microphones.

If you want to use a codec, then I would recommend that you stick with something micro controller friendly....

meteor-LZX, 2024-07-23:
Yes, I tried FLAC test sketch on the same hardware, here is my sketch:

```
#include <Arduino.h>
#include "AudioTools.h"
#include "AudioCodecs/CodecFLAC.h"

// Microphone
static const uint8_t mic_bck = GPIO_NUM_21;
static const uint8_t mic_ws = GPIO_NUM_47;
static const uint8_t mic_din = GPIO_NUM_48;
static const uint8_t mic_lr = GPIO_NUM_45;  // Channel the data into --- 0:left, 1:right

I2SStream in_stream;
HexDumpOutput out(Serial0);
EncodedAudioStream encoder(&out, new FLACEncoder()); // encode and write
StreamCopy copier(encoder, in_stream);

void setup() {
    Serial0.begin(115200);
    AudioLogger::instance().begin(Serial0, AudioLogger::Info);

    // Setup microphone & i2s
    pinMode(mic_lr, OUTPUT);
    digitalWrite(mic_lr, LOW);
    auto cfg = in_stream.defaultConfig(RX_MODE);
    cfg.i2s_format = I2S_STD_FORMAT;
    cfg.pin_bck = mic_bck;
    cfg.pin_ws = mic_ws;
    cfg.pin_data = mic_din;
    cfg.sample_rate = 44100;
    cfg.bits_per_sample = 16;
    cfg.channels = 2;
    in_stream.begin(cfg);
    Serial0.println("I2S started!");

    // Begin encoder with cfg(subclass of AudioInfo)
    encoder.begin(cfg);
}

void loop() {
    copier.copy();
}
```

It looks like the system is working properly, here is the partial output of terminal:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   147][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] I2SConfigESP32.h : 80 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Master
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 16
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: false
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 103 - pin_bck: 21
[I] I2SConfigESP32.h : 105 - pin_ws: 47
[I] I2SConfigESP32.h : 107 - pin_data: 48
I2S started!
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
66 4C 61 43 0 0 0 22  - 2 0 2 0 0 0 0 0
0 0 A C4 42 F0 0 0  - 0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0  - 0 0 84 0 0 28 20 0
0 0 72 65 66 65 72 65  - 6E 63 65 20 6C 69 62 46
4C 41 43 20 31 2E 33 2E  - 34 20 32 30 32 32 30 32
32 30 0 0 0 0 [I] StreamCopy.h : 158 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 158 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
clipping rice_parameter (15 -> 14) @6
clipping rice_parameter (15 -> 14) @6
clipping rice_parameter (15 -> 14) @6
clipping rice_parameter (15 -> 14) @6
clipping rice_parameter (15 -> 14) @6
clipping rice_parameter (15 -> 14) @6
clipping rice_parameter (15 -> 14) @6
FF F8  - 99 18 0 E6 12 0 0 14
E2 8 81 D0 0 3E 1F 0  - 88 36 25 21 78 9 40 CC
10 82 F0 4 20 F C0 4  - 26 0 A1 60 9 22 0 0
0 1 64 C0 1 DF C0 B  - AA 0 FE 60 C A 0 AC
52 0 42 0 BC 3 50 23  - 4 A0 5A 4 A1 74 81 68
```

About the RAM issue you mentioned, my board comes with 8M of PSRAM, and the compilation of the program seems to have a lot of RAM left. Is it because WiFi takes up too much dynamic RAM when used with flac codecs? I don't quite understand the question.
(here is the compilation result):
[image]

Besides, for the sample rate, because I want to collect ultrasonic signals to do some research, and according to Nyquist sampling theorem, the sampling rate must be above 40kHz, so I can only choose the commonly used sample rate of 44.1kHz and above. If I use WAV format to transfer raw PCM data, and because the amount of data is too large may lead to lost packets. But if I use lossy codecs such as MP3 or SBC, I may lose high-frequency information, so I can only choose lossless codecs such as FLAC.

Do you have any good solutions? Thank you so much!

pschatzmann, 2024-07-24:
A sampling rate of 40 khz will give you only frequencies up to 20'000hz, so you hardly get any ultrasonc sounds with this.

You don't see the RAM that is dynamically allocated in the statistics above and you would really need to print out the statistics in your sketch to get the available memory.
 
I hope you did not forget to activate PSRAM in the tools menu! 

You can drastically reduce the amount of data by using 1 channel only with 8 bits (using the EncoderL8) instead of 16.

closed 2024-07-24 by pschatzmann

meteor-LZX, 2024-07-25:
okay, thank you so much!