pschatzmann/arduino-audio-tools #1649: Wiki names wrong pins for PCM5102A dac board

madskjeldgaard, 2024-08-04:
### Problem Description

The HiLetGo pcm5102 I2S board mentioned here needs has wrong connections for the Pico. Specifically gpio's mentioned for LCB and DIN should be flipped. The correct connections in my testing was the following:

| PCM5102 Pin | Pico PIN     | Notes                   |
| ----------- | ------------ | ----------------------- |
| SCK         | GND          |                         |
| DIN         | GPIO28       |                         |
| LCK         | GPIO27       |                         |
| BCK         | GPIO26       |                         |
| GND         | GND          |                         |
| VIN         | VBUS (5V)    | could also be 3V3 (OUT) |

https://github.com/pschatzmann/arduino-audio-tools/wiki/External-DAC#hiletgo-pcm5102-i2s-iis-lossless-digital-audio-dac-decoder

### Device Description

Raspberry Pi Pico

### Sketch

```cpp
#include "AudioTools.h"
#include "alice.h"
// #include "knghtsng.h"
#include <Arduino.h>

MemoryStream wavfileMemory(alice_wav, alice_wav_len);
// MemoryStream wav(knghtsng_wav, knghtsng_wav_len);

constexpr auto sample_rate = 8000;
constexpr auto channels = 2;
constexpr auto bits_per_sample = 16;
constexpr auto baudrate = 115200;

AudioInfo info(sample_rate, channels, bits_per_sample);
I2SStream i2sOutput;
EncodedAudioStream sig(&i2sOutput, new WAVDecoder()); // Decoder stream
StreamCopy copier(sig, wavfileMemory);                // copy in to out

void setup() {
  Serial.begin(baudrate);

  while (!Serial)
    ;
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  // setup I2S output
  auto config = i2sOutput.defaultConfig(TX_MODE);
  config.copyFrom(info);

  auto result = i2sOutput.begin(config);

  if (!result) {
    LOGE("Failed to start I2S output");
    while (1)
      ;
  }

  // Setup audio file
  // wavfileMemory.setLoop(true);
  wavfileMemory.begin();
  sig.begin();
}

void loop() {
  if (wavfileMemory) {
    copier.copy();
  } else {
    // after we are done we just print some info from the wav file
    auto info = sig.audioInfo();
    LOGI("The audio rate from the wav file is %d", info.sample_rate);
    LOGI("The channels from the wav file is %d", info.channels);
    Serial.println("Restarting...");

    // Reset file
    wavfileMemory.end();
    wavfileMemory.begin();
  }
}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-04:
Documentation has been updated.

closed 2024-08-04 by pschatzmann

madskjeldgaard, 2024-08-04:
Looks good, thanks