pschatzmann/arduino-audio-tools #1650: source.IsAutoNext() not navigating to the wrong song

ra0943-VK3ACH, 2024-08-05:
### Problem Description

with the AudioSourceSDMMC ` source.isAutoNext();` is incorrectly navigating to the wrong audio files. With the following folder structure if song3.mp3 is selected via calling SelectFile("/Folder 2/song3.mp3"); (this is a custom function) the expected next song to play is song4.mp3 however song2.mp3 is played instead

```
├───Folder 1
│       song1.mp3
│       song2.mp3
│
└───Folder 2
        song3.mp3
        song4.mp3
```

### Device Description

ESP32-S3 with custom pcb. i2s DAC and 4bit sd card

### Sketch

```cpp
AudioSourceSDMMC source(startFilePath, ext);
I2SStream i2s;
MP3DecoderHelix decoder;
AudioPlayer player(source, i2s, decoder);
void setup(){
    source.isAutoNext();
}
void SelectFile(String filePath){
    player.stop();
    player.begin();
    player.setPath(filePath.c_str());
}
```
```

### What is your development environment

Arduino platformio

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-06:
This works as designed: select the first file with an index!

ps your sketch can't work and will give compile errors

closed 2024-08-06 by pschatzmann

ra0943-VK3ACH, 2024-08-06:
[idx.txt](https://github.com/user-attachments/files/16516197/idx.txt)
With the following idx.txt file if any file apart from the first 2 file is selected and played via `player.setPath("/The Cat Empire/The Cat Empire/01 - How to Explain.mp3");` the next file to play is always `/Wintergatan/Marble Machine (Piano Version)/01 - Marble Machine (Piano Version).mp3` it appears to mistake other files that are played with setPath as having index 0 instead on their propper index in index.txt

pschatzmann, 2024-08-07:
As I mentioned above, setPath() is not changing the internal index position.

Use setIndex(pos) to change the actual position as mentioned above!