pschatzmann/arduino-audio-tools #1653: previously working code no longer working (v0.9.7 > v0.9.8)

bobisaperson1, 2024-08-08:
### Problem Description

code working before latest update no longer compiles with following error:

```
In file included from d:\Users\----\Documents\Arduino\libraries\audio-tools\src/AudioAnalog/AnalogAudioStream.h:7,
                 from d:\Users\----\Documents\Arduino\libraries\audio-tools\src/AudioTools.h:80,
                 from D:\Users\----\Documents\Arduino\analog_to_i2s_test\analog_to_i2s_test.ino:10:
d:\Users\----\Documents\Arduino\libraries\audio-tools\src/AudioAnalog/AnalogDriverESP32V1.h: In member function 'bool audio_tools::AnalogDriverESP32V1::setup_rx()':
d:\Users\----\Documents\Arduino\libraries\audio-tools\src/AudioAnalog/AnalogDriverESP32V1.h:525:20: error: 'struct adc_continuous_handle_cfg_t' has no member named 'flags'
  525 |         adc_config.flags.flush_pool  = true;
      |                    ^~~~~

exit status 1

Compilation error: exit status 1
```

### Device Description

ESP32-WROOM-32 (ESP32 DEVKITV1)

### Sketch

```cpp
/**
 * @file streams-analog-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-adc-i2s/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"

AudioInfo info(22050, 2, 16);
AnalogAudioStream in; 
I2SStream out;                        
StreamCopy copier(out, in); // copy in to out

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgRx.sample_rate = 22050;
  cfgRx.adc_bit_width = 12;
  cfgRx.adc_calibration_active = true;
  cfgRx.is_auto_center_read = false;
  cfgRx.adc_attenuation = ADC_ATTEN_DB_0;
  cfgRx.channels = 1;
  cfgRx.adc_channels[0] = ADC_CHANNEL_4;
  in.begin(cfgRx);
 
  // TX on I2S_NUM_1 
  auto cfgTx = out.defaultConfig(TX_MODE);
  cfgTx.port_no = 1;
  cfgTx.i2s_format = I2S_LSB_FORMAT;
  cfgTx.copyFrom(info);
  cfgTx.pin_bck = 27;
  cfgTx.pin_ws = 25;
  cfgTx.pin_data = 26;
  out.begin(cfgTx);
}

// Arduino loop - copy data 
void loop() {
  copier.copy();
}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

bobisaperson1, 2024-08-08:
code is a slightly modified example "examples/examples-stream/streams-adc-i2s"

pschatzmann, 2024-08-08:
But this [has already been corrected to my knowlege](https://github.com/pschatzmann/arduino-audio-tools/commit/016ded59a86df21f811b7d1cec7d4973e7fa638f)!

What Arduino IDF version are you using ?
Double check that you use the latest master branch...

closed 2024-08-08 by pschatzmann