pschatzmann/arduino-audio-tools #1654: sparatic data from adc

bobisaperson1, 2024-08-09:
### Problem Description

getting extremly quiet garbage form andthing using the adc, tested 2 different examples on 2 diferent ESP32s. using the adc serial example i can see that the is some data every now and then but not contant as expected, almost like its only using the loudest samples.

[image]
blue is what most of the data looks like with red being the intermitent data.

I'm using pin 32 which I only worked out through trial and error as all other pins with all other adc channels gave garbage all the time while 32 and channel 4 only give garbage when something is being inputed, I am testing using a phone pluged into a stripped 3.5mm jack with gnd to gnd and left channel to pin 32.

sorry if I have missed any infomation

### Device Description

ESP32-WROOM-32 (ESP32 DEVKITV1)

### Sketch

```cpp
/**
 * @file adc-serial.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/adc-serial/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 * #TODO retest is outstanding
 */
 
#include "Arduino.h"
#include "AudioTools.h"

/**
 * @brief We use a mcp6022 analog microphone on GPIO34 and write it to Serial
 */ 

AnalogAudioStream adc;
const int32_t max_buffer_len = 1024;
uint8_t buffer[max_buffer_len];
// The data has a center of around 26427, so we we need to shift it down to bring the center to 0
ConverterScaler<int16_t> scaler(1.0, -26427, 32700 );

// Arduino Setup
void setup(void) {

  delay(3000); // wait for serial to become available

  Serial.begin(115200);
  // Include logging to serial
  AudioLogger::instance().begin(Serial, AudioLogger::Info); //Warning, Info, Error, Debug
  Serial.println("starting ADC...");
  auto adcConfig = adc.defaultConfig(RX_MODE);

  // For ESP32 by Espressif Systems version 3.0.0 and later:
  // see examples/README_ESP32.md
  adcConfig.sample_rate = 44100;
  adcConfig.adc_bit_width = 12;
  adcConfig.adc_calibration_active = true;
  adcConfig.is_auto_center_read = false;
  adcConfig.adc_attenuation = ADC_ATTEN_DB_12; 
  adcConfig.channels = 1;
  adcConfig.adc_channels[0] = ADC_CHANNEL_4;

  adc.begin(adcConfig);
}

// Arduino loop - repeated processing 
void loop() {
  size_t len = adc.readBytes(buffer, max_buffer_len); 
  // move center to 0 and scale the values
  scaler.convert(buffer, len);

  int16_t *sample = (int16_t*) buffer; 
  int size = len / 4;
  for (int j=0; j<size; j++){
    Serial.print(*sample++);
    Serial.print(", ");
    Serial.println(*sample++);
  }
}
```

### What is your development environment

Using the latest audio-tools pre-release (also had the same issue with 0.9.7 and 0.9.8)

Arduino ESP32 version 3.0.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-09:
But this is just pure nonsense!

- Why are you printing 2 channels but you have only 1 ?
- Try to [look up the correct pin for the channel](https://github.com/espressif/esp-idf/blob/v4.2/components/soc/soc/esp32/include/soc/adc_channel.h) that you are using: Channel 4 is GPIO 32 on a ESP32
Remove the ConverterScaler: I think you just use some nonsense values: in the end it should oscillate around 0 and not -26000

Try to start to record a sine wave and take it up from there...

closed 2024-08-09 by pschatzmann

bobisaperson1, 2024-08-09:
> But this is just pure nonsense!
It's a copy and paste of your example......

pschatzmann, 2024-08-09:
Which one ?

bobisaperson1, 2024-08-09:
examples/examples-basic-api/base-adc-serial/base-adc-serial.ino
and I get the same results with
examples/examples-stream/streams-adc-serial/streams-adc-serial.ino
and all I change is removing the second channel

pschatzmann, 2024-08-09:
In the this example it is demonstrated in the comment how to change some default setttings.
The comments are not active: so it is using stereo on 2 channels (ADC_CHANNEL_0, ADC_CHANNEL_3)

The key here is that it will not work if you don't use the correct pins!

bobisaperson1, 2024-08-09:
so the example out the box the way it is on github will work with the pins that match ADC_CHANNEL_0 and ADC_CHANNEL_3 in stereo? does this work with < 3.0.0? I was under the understanding that the commented out code was for <3.0.0 and with it comented it wouldnt work properly

pschatzmann, 2024-08-09:
No the commented out code are example parameters in > 3.0
Before 3.0 has different parameters and pin handling....

bobisaperson1, 2024-08-09:
Sorry I got my arrows the wrong was around, I meant "> 3.0.0" not "< 3.0.0"

I just did a fresh copy, paste and upload (sketch is exactly the same as "examples/examples-basic-api/base-adc-serial/base-adc-serial.ino") swapped to using stereo on pins 36 and 39 (channel 0 and channel 3) and am getting the same results

pschatzmann, 2024-08-09:
Did you follow the advice that I gave you ? 
Share the display of the sine wave!

bobisaperson1, 2024-08-09:
my sketch now looks like this

'''
/**
 * @file adc-serial.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/adc-serial/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 * #TODO retest is outstanding
 */
 
#include "Arduino.h"
#include "AudioTools.h"

/**
 * @brief We use a mcp6022 analog microphone on GPIO34 and write it to Serial
 */ 

AnalogAudioStream adc;
const int32_t max_buffer_len = 1024;
uint8_t buffer[max_buffer_len];

// Arduino Setup
void setup(void) {

  delay(3000); // wait for serial to become available

  Serial.begin(115200);
  // Include logging to serial
  AudioLogger::instance().begin(Serial, AudioLogger::Info); //Warning, Info, Error, Debug
  Serial.println("starting ADC...");
  auto adcConfig = adc.defaultConfig(RX_MODE);

  // For ESP32 by Espressif Systems version 3.0.0 and later:
  // see examples/README_ESP32.md
  // adcConfig.sample_rate = 44100;
  // adcConfig.adc_bit_width = 12;
  // adcConfig.adc_calibration_active = true;
  // adcConfig.is_auto_center_read = false;
  // adcConfig.adc_attenuation = ADC_ATTEN_DB_12; 
  // adcConfig.channels = 2;
  // adcConfig.adc_channels[0] = ADC_CHANNEL_4; 
  // adcConfig.adc_channels[1] = ADC_CHANNEL_5;

  adc.begin(adcConfig);
}

// Arduino loop - repeated processing 
void loop() {
  size_t len = adc.readBytes(buffer, max_buffer_len); 

  int16_t *sample = (int16_t*) buffer; 
  int size = len / 4;
  for (int j=0; j<size; j++){
    Serial.print(*sample++);
    Serial.print(", ");
    Serial.println(*sample++);
  }
}
'''

and i am getting the same result just around 0

[image]

you can see parts of the sinewave (100hz) but its cutting in and out still

pschatzmann, 2024-08-09:
This does not look even close to a sine wave to me! But maybe this is due to the low frequency: Use something which you can see better: e.g.  2000 hz

Are you sure that you generate a clean sine wave  in the range of 0 ~ 2450 mV ?

bobisaperson1, 2024-08-09:
[image]
1200hz

looks better, doesnt cut in and out as much as 100hz or music did

im not sure if its between 0 ~ 2450 mV, I dont have any fancy equipment its just whats comming out of my IPhone Xr through a lightning to aux adapter at full volume

pschatzmann, 2024-08-09:
Not sure if this can be used this way, but this looks a bit better.

But it is still way off. Maybe this could be because of the high sampling rate that generates too much data to output it via Serial: Try to reduce it to 8000

ps. To increase the amplitude you can adjust the DB setting. But I would try this only after you really see a sine wave...

bobisaperson1, 2024-08-09:
would i do that by uncomenting the "  //  adcConfig.sample_rate = 44100;" line and changing that?

pschatzmann, 2024-08-09:
yes

bobisaperson1, 2024-08-09:
i am getting this over and over

```
E (24084) adc_continuous: adc_continuous_read(418): The driver isn't initialised
[E] AnalogDriverESP32V1.h : 419 - adc_continuous_read unsuccessful
```

from the following code

```
/**
 * @file adc-serial.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/adc-serial/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 * #TODO retest is outstanding
 */
 
#include "Arduino.h"
#include "AudioTools.h"

/**
 * @brief We use a mcp6022 analog microphone on GPIO34 and write it to Serial
 */ 

AnalogAudioStream adc;
const int32_t max_buffer_len = 1024;
uint8_t buffer[max_buffer_len];

// Arduino Setup
void setup(void) {

  delay(3000); // wait for serial to become available

  Serial.begin(115200);
  // Include logging to serial
  AudioLogger::instance().begin(Serial, AudioLogger::Info); //Warning, Info, Error, Debug
  Serial.println("starting ADC...");
  auto adcConfig = adc.defaultConfig(RX_MODE);

  // For ESP32 by Espressif Systems version 3.0.0 and later:
  // see examples/README_ESP32.md
  adcConfig.sample_rate = 8000;
  // adcConfig.adc_bit_width = 12;
  // adcConfig.adc_calibration_active = true;
  // adcConfig.is_auto_center_read = false;
  // adcConfig.adc_attenuation = ADC_ATTEN_DB_12; 
  // adcConfig.channels = 2;
  // adcConfig.adc_channels[0] = ADC_CHANNEL_4; 
  // adcConfig.adc_channels[1] = ADC_CHANNEL_5;

  adc.begin(adcConfig);
}

// Arduino loop - repeated processing 
void loop() {
  size_t len = adc.readBytes(buffer, max_buffer_len); 

  int16_t *sample = (int16_t*) buffer; 
  int size = len / 4;
  for (int j=0; j<size; j++){
    Serial.print(*sample++);
    Serial.print(", ");
    Serial.println(*sample++);
  }
}
```

pschatzmann, 2024-08-09:
Not sure what the limits are. Maybe you need to play around to find what's supported.

ps. To record audio from an external device, I am usually using an [AudioKit](https://github.com/pschatzmann/arduino-audio-tools/wiki/Audio-Boards#esp32-a1s-based-audio-boards-audiokit-lyrat): This has a proper Audio DAC

pschatzmann, 2024-08-09:
Just below you find a module that provides the same audio hardware...

pschatzmann, 2024-08-09:
It is expected that you don't see any values below 0 because that ADC will measure and provide only positive values. The thing that is worrying me, is that quite some information seems to be missing between the displayed peaks.

If you activate the autocenter the whole thing should be shifted down so that you get negative values as well: but this will be way off if you don't get any clean sine input in the first place

bobisaperson1, 2024-08-09:
plugging in a different phone i got this coming up

[image]

bobisaperson1, 2024-08-09:
[image]

same thing out the laptop

bobisaperson1, 2024-08-09:
> ps. did you enventually forgot to connect the audio ground from the connector to the ESP32 ?

definatly conected, just double checked

pschatzmann, 2024-08-09:
This looks lightly better but I am running out of ideas:
- Try another GND (I have an ESP32 where one GND pin is not working properly)
- check if calibration makes a difference
- check with different DB values
- finally check if autocenter moves the wave down with the center as 0.

bobisaperson1, 2024-08-09:
- other gnd pin: nope
- db 12: nope
- db 6: maybe? marginally if anything
- cant find the page that says what other db values are compatible
- autocentre: as you expected just moves the whole thing down a bit (about -75)
- calibration: just moves the whole thing up 142

im out of ideas and am going to have a proper look into the hardware solutions you mentioned earlier

pschatzmann, 2024-08-09:
https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/adc.html#adc-attenuation

bobisaperson1, 2024-08-09:
[image]
[image]
[image]

bobisaperson1, 2024-08-09:
> I am usually using an [AudioKit]

where did you get yours? I cant seem to find anywhere that isnt a roll the dice and see if it actually arives china store

pschatzmann, 2024-08-09:
I got mine from Aliexpress.

Just googled the topic Audio Aux and Oszillosope and found the following image :
https://www.edn.com/wp-content/uploads/media-1188863-f5.jpg

So the sound wave seems to oscillate between a negative and positive voltage!
The ESP32 ADC can handle positive values, so that would explain your picture.

pschatzmann, 2024-08-09:
You could also [just use some I2S ADC](https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC#i2s-adc-audio-i2s-capture-card-module)

bobisaperson1, 2024-08-09:
> You could also [just use some I2S ADC](https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC#i2s-adc-audio-i2s-capture-card-module)

I tried to find one of these boards but they don’t seem to very easily available

bobisaperson1, 2024-08-09:
How am I the first person to run into this issue? Am I the only person who has tried to use your library with the adc on an ESP32????

pschatzmann, 2024-08-09:
Usually it is used with a analog microphone module...

bobisaperson1, 2024-08-09:
Ah yes makes sense, over the weekend I’m going to look at making a circuit to shift the voltage up 1.5v to try and get it all above 0v

bobisaperson1, 2024-08-21:
gave up on that idea and just bought an audiokit, works fantastic right out the box. now my main issue is just having enough gpio which is easy to solve