pschatzmann/arduino-audio-tools #1662: Incorrect size of core dump image on YourCee AI Thinker ESP32-A1S Audio Kit V2.2 A417

peterburk, 2024-08-14:
### Problem Description

After flashing the example send-receive sketch in Arduino Audio Tools, the serial monitor shows: 

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
E (398) esp_cor⸮յ⸮}⸮⸮⸮͡⸮ Incorrect size of core dump image: -4
[W] I2SESP32V1.h : 184 - Invalid channel_format: 3

```

I'm using the ESP32 Dev Module board definition in Arduino. 

### Device Description

YourCee AI Thinker ESP32-A1S Audio Kit V2.2 A417

### Sketch

```cpp
send-receive example
```

### Other Steps to Reproduce

Tried squeezelite-esp32 and esp-idf successfully

### What is your development environment

Arduino IDE 2.3.2, macOS Sonoma

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-14:
If you are using an AudioKit, did you replace the I2SStream ?

peterburk, 2024-08-14:
Yes, I replaced the I2SStream. Now I see:
`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
E (421) esp_cor⸮Eյ⸮}⸮⸮⸮͡⸮ Incorrect size of core dump image: -4
E (54) i2s_common: i2s_new_channel(812): no available channel found
[E] I2SESP32V1.h : 433 - i2s_channel
`

pschatzmann, 2024-08-14:
I was using AudioBoardStream out(AudioKitEs8388V1);  and can't reproduce your issue.
Start with a simple AudioKit Sine Generator sketch and continue only if this is working!

[See here](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)

closed 2024-08-14 by pschatzmann

peterburk, 2024-08-14:
The same "Incorrect size of core dump image: -4" error shows when installing other sketches from other libraries, such as MinimalDiagnostics.ino. 

What is the correct board and driver to use for the ESP32-A1S?

pschatzmann, 2024-08-14:
I am using "ESP32 Dev Module"

peterburk, 2024-08-14:
I got it working! The "Incorrect size of core dump" message still shows, but a sine wave is generated and comes out of the DAC. Here's the sketch:

```
#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/I2SCodecStream.h"
#include <Wire.h>

// I2C
#define SDAPIN 33        // I2C Data,  Adafruit ESP32 S3 3, Sparkfun Thing Plus C 23
#define SCLPIN 32        // I2C Clock, Adafruit ESP32 S3 4, Sparkfun Thing Plus C 22
#define I2CSPEED 100000 // Clock Rate

#define ES8388_ADDR 0x10

#define ACK_CHECK_EN 0x1

/* ES8388 register */
#define ES8388_CONTROL1 0x00
#define ES8388_CONTROL2 0x01
#define ES8388_CHIPPOWER 0x02
#define ES8388_ADCPOWER 0x03
#define ES8388_DACPOWER 0x04
#define ES8388_CHIPLOPOW1 0x05
#define ES8388_CHIPLOPOW2 0x06
#define ES8388_ANAVOLMANAG 0x07
#define ES8388_MASTERMODE 0x08

/* ADC */
#define ES8388_ADCCONTROL1 0x09
#define ES8388_ADCCONTROL2 0x0a
#define ES8388_ADCCONTROL3 0x0b
#define ES8388_ADCCONTROL4 0x0c
#define ES8388_ADCCONTROL5 0x0d
#define ES8388_ADCCONTROL6 0x0e
#define ES8388_ADCCONTROL7 0x0f
#define ES8388_ADCCONTROL8 0x10
#define ES8388_ADCCONTROL9 0x11
#define ES8388_ADCCONTROL10 0x12
#define ES8388_ADCCONTROL11 0x13
#define ES8388_ADCCONTROL12 0x14
#define ES8388_ADCCONTROL13 0x15
#define ES8388_ADCCONTROL14 0x16

/* DAC */
#define ES8388_DACCONTROL1 0x17
#define ES8388_DACCONTROL2 0x18
#define ES8388_DACCONTROL3 0x19
#define ES8388_DACCONTROL4 0x1a
#define ES8388_DACCONTROL5 0x1b
#define ES8388_DACCONTROL6 0x1c
#define ES8388_DACCONTROL7 0x1d
#define ES8388_DACCONTROL8 0x1e
#define ES8388_DACCONTROL9 0x1f
#define ES8388_DACCONTROL10 0x20
#define ES8388_DACCONTROL11 0x21
#define ES8388_DACCONTROL12 0x22
#define ES8388_DACCONTROL13 0x23
#define ES8388_DACCONTROL14 0x24
#define ES8388_DACCONTROL15 0x25
#define ES8388_DACCONTROL16 0x26
#define ES8388_DACCONTROL17 0x27
#define ES8388_DACCONTROL18 0x28
#define ES8388_DACCONTROL19 0x29
#define ES8388_DACCONTROL20 0x2a
#define ES8388_DACCONTROL21 0x2b
#define ES8388_DACCONTROL22 0x2c
#define ES8388_DACCONTROL23 0x2d
#define ES8388_DACCONTROL24 0x2e
#define ES8388_DACCONTROL25 0x2f
[... 137 lines cut ...]
and the serial console:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
E (420) esp_cor⸮Eյ⸮}⸮⸮⸮͡⸮ Incorrect size of core dump image: -4
Setup starting...
I2C pin ...
I2S pin ...
Pins begin ...
Board begin ...
I2S begin ...
Sine wave begin...
Setup completed ...

```

pschatzmann, 2024-08-14:
Why so complicated ?
This [short sketch should work as well](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-generator-audiokit/streams-generator-audiokit.ino)

peterburk, 2024-08-15:
The streams-generator-audiokit sketch did work as well! And so did the streams-synth-audiokit :)

Now I'm trying to get effects to work, with a goal of being able to add reverb to audio from the line in port. 

I've based the following on streams-audiokit-effects-audiokit, but the output just sounds like the original input. 

```
/**
 * @file streams-i2s-i2s.ino
 * @brief Copy audio from I2S to I2S  - I2S uses 1 i2s port
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "es8388.h"
#include "Wire.h"

//uint16_t sample_rate=40000;
//uint16_t channels = 2;
// Audio Format
const int sample_rate = 44100;
const int channels = 1;

uint16_t bits_per_sample = 16; // or try with 24 or 32
I2SStream i2s;

FilteredStream<int16_t, float> filtered(i2s, channels);  // Defiles the filter as BaseConverter
AudioEffectStream effects(filtered); // input from kit
StreamCopy copier(filtered, i2s); // copies sound into i2s (both from kit to filtered or filered to kit are supported)

// StreamCopy copier(i2s, i2s); // copies sound into i2s

// Effects control input
float volumeControl = 0.1;
int16_t clipThreashold = 6000;
float fuzzEffectValue = 32200;
int16_t fuzzMaxValue = 32200;
int16_t tremoloDuration = 2000;
float tremoloDepth = 100;

// Effects
Boost boost(volumeControl);
Distortion distortion(clipThreashold);
Fuzz fuzz(fuzzEffectValue);
Tremolo tremolo(tremoloDuration, tremoloDepth, sample_rate);

// Update values in effects
void updateValues(){
  // update values in controls
  boost.setVolume(volumeControl);
  boost.setActive(volumeControl>0);
  distortion.setClipThreashold(clipThreashold);
  distortion.setActive(clipThreashold>0);
  fuzz.setFuzzEffectValue(fuzzEffectValue);
  fuzz.setMaxOut(fuzzMaxValue);
  fuzz.setActive(fuzzEffectValue>0);
  tremolo.setDepth(tremoloDepth);
  tremolo.setDuration(tremoloDuration);
  tremolo.setActive(tremoloDuration>0);
}

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  // change to Warning to improve the quality
  AudioLogger::instance().begin(Serial, AudioLogger::Error); 

  // Input/Output Modes
  es_dac_output_t output = (es_dac_output_t) ( DAC_OUTPUT_LOUT1 | DAC_OUTPUT_LOUT2 | DAC_OUTPUT_ROUT1 | DAC_OUTPUT_ROUT2 );
  es_adc_input_t input = ADC_INPUT_LINPUT2_RINPUT2;
  //  es_adc_input_t input = ADC_INPUT_LINPUT1_RINPUT1;

  TwoWire wire(0);
  wire.setPins( 33, 32 );
  
  es8388 codec;
  codec.begin( &wire );
  codec.config( bits_per_sample, output, input, 90 );

  // start I2S in
  Serial.println("starting I2S...");
  auto config = i2s.defaultConfig(RXTX_MODE);
  config.sample_rate = sample_rate; 
  config.bits_per_sample = bits_per_sample; 
  config.channels = 2;
  config.i2s_format = I2S_STD_FORMAT;
  config.pin_ws = 25;
  config.pin_bck = 27;
  config.pin_data = 26;
  config.pin_data_rx = 35;
  config.pin_mck = 0;

  // setup effects
  effects.addEffect(boost);
  effects.addEffect(distortion);
  effects.addEffect(fuzz);
  effects.addEffect(tremolo);
  effects.begin(config);
  updateValues();

  i2s.begin(config);

  Serial.println("I2S started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}
```

When I tried adding freeverb, I got the following error:
```
/Users/peter/Documents/Arduino/libraries/es8388/examples/streams-i2s-i2s/streams-i2s-i2s.ino: In function 'void setup()':
/Users/peter/Documents/Arduino/libraries/es8388/examples/streams-i2s-i2s/streams-i2s-i2s.ino:103:20: error: no matching function for call to 'audio_tools::AudioEffectStreamT<short int>::addEffect(stk::FreeVerb&)'
  103 |   effects.addEffect(freeverb);
      |   ~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /Users/peter/Documents/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:92,
                 from /Users/peter/Documents/Arduino/libraries/es8388/examples/streams-i2s-i2s/streams-i2s-i2s.ino:8:
/Users/peter/Documents/Arduino/libraries/arduino-audio-tools/src/AudioEffects/AudioEffects.h:346:10: note: candidate: 'void audio_tools::AudioEffectStreamT<T>::addEffect(audio_tools::AudioEffect&) [with T = short int]'
  346 |     void addEffect(AudioEffect &effect){
      |          ^~~~~~~~~
/Users/peter/Documents/Arduino/libraries/arduino-audio-tools/src/AudioEffects/AudioEffects.h:346:33: note:   no known conversion for argument 1 from 'stk::FreeVerb' to 'audio_tools::AudioEffect&'
  346 |     void addEffect(AudioEffect &effect){
      |                    ~~~~~~~~~~~~~^~~~~~
/Users/peter/Documents/Arduino/libraries/arduino-audio-tools/src/AudioEffects/AudioEffects.h:352:10: note: candidate: 'void audio_tools::AudioEffectStreamT<T>::addEffect(audio_tools::AudioEffect*) [with T = short int]'
  352 |     void addEffect(AudioEffect *effect){
      |          ^~~~~~~~~
/Users/peter/Documents/Arduino/libraries/arduino-audio-tools/src/AudioEffects/AudioEffects.h:352:33: note:   no known conversion for argument 1 from 'stk::FreeVerb' to 'audio_tools::AudioEffect*'
  352 |     void addEffect(AudioEffect *effect){
      |                    ~~~~~~~~~~~~~^~~~~~

exit status 1

Compilation error: no matching function for call to 'audio_tools::AudioEffectStreamT<short int>::addEffect(stk::FreeVerb&)'
```

This might not matter though, because I read the [Limitations](https://github.com/pschatzmann/arduino-audio-tools/wiki/Audio-Effects):
"The Effects work only on one channel, so if you provide 2 channels, the are combined before applying the effects and then are replicated to give the correct channels as output"

I was hoping to create a stereo reverb effect, but it may only work with mono channels.

pschatzmann, 2024-08-15:
Look at the error message: you did not provide an [AudioEffect](https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_audio_effect.html) subclass!