pschatzmann/arduino-audio-tools #1663: streams-generator-analog on ESP32 gives conversion error

panseeatta, 2024-08-14:
### Problem Description

I'm trying to generate a simple tone using ESP32 (with built-in DAC, and therefore sending the audio stream to analog). I'm using the example sketch as provided. It compiles fine, but then the following error repeats, and no sound is generated:

```
22:31:30.106 -> [I] StreamCopy.h : 408 - try write  - 12 (open 1024 bytes) 
22:31:30.106 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.106 -> [I] StreamCopy.h : 408 - try write  - 13 (open 1024 bytes) 
22:31:30.106 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.140 -> [I] StreamCopy.h : 408 - try write  - 14 (open 1024 bytes) 
22:31:30.140 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.140 -> [I] StreamCopy.h : 408 - try write  - 15 (open 1024 bytes) 
22:31:30.180 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.180 -> [I] StreamCopy.h : 408 - try write  - 16 (open 1024 bytes) 
22:31:30.180 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.221 -> [I] StreamCopy.h : 408 - try write  - 17 (open 1024 bytes) 
22:31:30.221 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.221 -> [I] StreamCopy.h : 408 - try write  - 18 (open 1024 bytes) 
22:31:30.221 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.221 -> [I] StreamCopy.h : 408 - try write  - 19 (open 1024 bytes) 
22:31:30.258 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.258 -> [I] StreamCopy.h : 408 - try write  - 20 (open 1024 bytes) 
22:31:30.258 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.292 -> [I] StreamCopy.h : 408 - try write  - 21 (open 1024 bytes) 
22:31:30.292 -> [E] AudioStreamsConverter.h : 546 - bit combination not supported 16 -> 0
22:31:30.292 -> [E] StreamCopy.h : 401 - write  to target has failed after 22 retries! (1024 bytes)
```

I've tried changing the bit sampling value, to no avail. Any assistance would be appreciated.

### Device Description

ESP32-WROOM-32
Like so:
https://ecksteinimg.de/Datasheet/CP06072/Pinout.pdf

### Sketch

```cpp
/**
 * @file streams-generator-analog.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-analog/README.md 
 * @author Phil Schatzmann
 * @copyright GPLv3
 **/
 
#include "AudioTools.h"

AudioInfo info(8000, 1, 16);
SineWaveGenerator<int16_t> sineWave(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound(sineWave);             // Stream generated from sine wave
AnalogAudioStream out; 
StreamCopy copier(out, sound);                             // copies sound into i2s

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start the analog output
  auto config = out.defaultConfig(TX_MODE);
  config.copyFrom(info); 
  out.begin(config);

  // Setup sine wave
  sineWave.begin(info, N_B4);
  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}
```

### Other Steps to Reproduce

I've tried attaching either speakers or earphones, and neither receives sound.

### What is your development environment

Arduino IDE.

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-15:
In "productive code" you should check the result of the begin() method and stop when it is not successful.
If you check in your log you are getting:

```
04:11:00.353 -> [I] AnalogConfigESP32V1.h : 145 - I2S_MODE_DAC_BUILT_IN
04:11:00.385 -> [I] AnalogAudioStream.h : 66 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
04:11:00.385 -> E (604) DAC_DMA: s_dac_dma_periph_set_clock(96): Frequency is too small, the mclk division exceed the maximum value 255
04:11:00.385 -> E (606) DAC_DMA: dac_dma_periph_init(125): Failed to set clock of DMA peripheral
04:11:00.385 -> E (613) dac_continuous: dac_continuous_new_channels(254): Failed to initialize DAC DMA peripheral
04:11:00.417 -> [E] AnalogDriverESP32V1.h : 448 - new_channels
```
If you want to go that low, you need to activate the apll !

I committed a correction that activates the apll by default for TX_MODE...

closed 2024-08-15 by pschatzmann

panseeatta, 2024-08-15:
Ah yes, the issue was the sample rate! Increasing it fixed it. Thank you so much!!
(I was not getting that message in my log though, I just checked!)