pschatzmann/arduino-audio-tools #1689: can't Load SD samples with maximilian

rupraph, 2024-09-06:
### Problem Description

Description:
I cannot manage to read a sample from SD with maximilian. Acces to SD seems ok, but maximilian won't load it's content.
When printing the summury info, charactics are there, but content remains at zero. So of course it is not playing.

I do the following for loading:
sample1.load("/sdcard/pong.wav"); Serial.println(sample1.getSummary().c_str());

See serial output below:
Loading: ### /sdcard/pong.wav
Ch: 1, len: 0
Format: 1
Channels: 1
SampleRate: 11025
ByteRate: 11025
BlockAlign: 1
BitsPerSample: 8
Length: 0

I tried with different samples (mono). I tried short to very short samples, like 2kb to 60kb, which I would expect to work without any issue, not matter if I use PSRAM ? 

I checked a bit how the load function works but can't find what is going wrong. The data seems to be red, but the last manipulation which manipulate the chunks to the "amplitude" variable which seems to be the final data to be played is not filled.

### Device Description

Esp32-s3 N8R8 devkitC

### Sketch

```cpp
Below a minimal reproductible exemple, based on the example provided in the lib. Issue might be esp32 related ? or am I missing some constraints on the sample type/format ? :

`#include "AudioTools.h"
#include "AudioLibs/MaximilianDSP.h"
#include "maximilian.h"
#include <FS.h>
#include <SD_MMC.h>

I2SStream audio_output;
Maximilian maximilian(audio_output);

AudioInfo info = AudioInfo(44100, 2, 16);

maxiSample beats;

#define D0 40 //(D0) SPI_MISO
#define CLK 41 //(CLK) SPI_CLK
#define CMD 42
#define I2S_WS 21
#define I2S_DATA 47
#define I2S_BCK 48

void setup() {
Serial.begin(115200);
I2SConfig config = audio_output.defaultConfig(TX_MODE);
config.pin_bck = I2S_BCK;
config.pin_ws = I2S_WS;
config.pin_data = I2S_DATA;
audio_output.begin(config);
maximilian.begin(config);
SD_MMC.setPins(CLK, CMD, D0);
if (!SD_MMC.begin("/sdcard", true))
{
Serial.println("Card Mount Failed");
return;
}
beats.load("/sdcard/pong.wav");
Serial.println(beats.getSummary().c_str());

}

void play(float *output) {
output[0]=beats.playAtSpeed(0.68);
output[1]=output[0];
}

void loop() {
maximilian.copy();
}`
```

### What is your development environment

PlatformIO, but the example above was also tested to have the issue with arduino IDE

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-08:
Did you try with the provided example that is using the [wav in the resource directory ](https://github.com/pschatzmann/Maximilian/tree/master/doc/resources)?
I think your example files are wrong: a length of 0 does not give you much!

rupraph, 2024-09-08:
Thanks. 
I just downloaded these files to confirm werther this could be related to the files, and the result is unfortunately the same, the loaded sample still reports a len of 0  (just tried with snare.wav which you also mention in your reference exemple, and blip.wav)

Which is weird. I can list the content of the SD, but when loading the samples, something goes wrong. 

Would you have any lead on what could cause this ? I don't really know where to start here.

pschatzmann, 2024-09-10:
Did you try a test sketch that confirms that reading of the file is working ?

rupraph, 2024-09-10:
I only tested I can access the SD, by listing all files in the SD. 
Also, the correct sample rate and audio resolution is calculated by the sample summury function, which makes me think the file is correctly accessed ?

pschatzmann, 2024-09-10:
If I remember right the functionality uses the ESP32 virtual file system and the related c file API.

So I would not take any assumptions and test if you can open and read the file with fopen and fread.

rupraph, 2024-09-10:
Thanks for the directions !
So I can confirm the file can be accessed and red by the sd_mmc library. I could print the content with the reference exemple.
However the maximilian sample.load("/xxx.wav") function return 0

I noted maximilian is using :
```ifstream inFile( myPath.c_str(), ios::in | ios::binary);
result = inFile.is_open();```
While the sd_mmc file system uses:
```File file = fs.open(path); then file.read()```

I am not really familiar with filesystem stuff, but I guess I might not be in the right config to use ifstream ?

pschatzmann, 2024-09-11:
hmm: try to make a [read test using ifstream](https://cplusplus.com/reference/fstream/ifstream/ifstream/)

pschatzmann, 2024-09-11:
Just noticed that you forgot the prefix to specify the file system:
use 
beats.load("/sdcard/xxx.wav");

when the file is in the root directory of the drive

rupraph, 2024-09-11:
> Just noticed that you forgot the prefix to specify the file system: use beats.load("/sdcard/xxx.wav");
> 
> when the file is in the root directory of the drive that you named /sdcard

Actually this prefix was in my initial issue code. But yes, I'll test ifstream alone first and see.

pschatzmann, 2024-09-26:
I added some asserts to the load() method and found that the reported size of amplitudes was always 0!
I committed some correctons to maximilian and added some logging for the memory allocation that can be activated via the Arduino Tools menu.

closed 2024-09-26 by pschatzmann

rupraph, 2024-09-26:
So nice.
I could just find 5 minutes to test it, and it works like a charm ! 
Thank you so much !