pschatzmann/arduino-audio-tools #1696: WiFiClientSecure not found

dizcza, 2024-09-11:
### Problem Description

It would be nice if you provide the installation instructions.

I'm using an ESP32 board and building your project with platformio - is it correct? It doesn't seem that I can use it directly in Arduino IDE - there is no such lib as audio-tools.

```
Compiling .pio/build/ttgo-t1/libbec/WiFi/WiFi.cpp.o
Compiling .pio/build/ttgo-t1/libbec/WiFi/WiFiAP.cpp.o
Compiling .pio/build/ttgo-t1/libbec/WiFi/WiFiClient.cpp.o
In file included from .pio/libdeps/ttgo-t1/audio-tools/src/AudioHttp/AudioHttp.h:8,
                 from .pio/libdeps/ttgo-t1/audio-tools/src/AudioTools.h:96,
                 from src/main.cpp:2:
.pio/libdeps/ttgo-t1/audio-tools/src/AudioHttp/URLStream.h:9:10: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

 #include <WiFiClientSecure.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/ttgo-t1/src/main.cpp.o] Error 1
```

### Device Description

TTGO T8

### Sketch

```cpp
#include <Arduino.h>
#include "AudioTools.h"

AudioInfo info(44100, 1, 16);
I2SStream i2sStream; // Access I2S as stream
CsvOutput<int16_t> csvStream(Serial);
StreamCopy copier(csvStream, i2sStream); // copy i2sStream to csvStream

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  
  auto cfg = i2sStream.defaultConfig(RX_MODE);
  cfg.i2s_format = I2S_MSB_FORMAT; // or try with I2S_LSB_FORMAT
  cfg.bits_per_sample = 16;
  cfg.channels = 1; // optional because default setting
  cfg.sample_rate = 44100;
  cfg.is_master = false;
  cfg.use_apll = true;

  cfg.pin_ws = 22;
  cfg.pin_data = 19;
  cfg.pin_bck = 23;

  i2sStream.begin(cfg);

  // make sure that we have the correct channels set up
  csvStream.begin(info);
}

// Arduino loop - copy data
void loop() {
    copier.copy();
}
```

### What is your development environment

PLATFORM: Espressif 32 (6.6.0) > TTGO T1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-11:
Follow the instruction given in the Wiki !

closed 2024-09-11 by pschatzmann

dizcza, 2024-09-16:
Thanks, now it compiles but I need a "how to start" hint.

I'm using a newer version of I2S-ADC converter:

[image]

And I found 3 different config setups for older modules of yours that you linked on the blog and here.

First, 

```
    I2SStream i2sStream; // Access I2S as stream
    auto cfg = i2sStream.defaultConfig(RX_MODE);
    cfg.bits_per_sample = 32;
    cfg.channels = 2;
    cfg.sample_rate = 96000;
    cfg.is_master = false;
    cfg.i2s_format = I2S_MSB_FORMAT;
    cfg.use_apll = true;
    i2sStream.begin(cfg);
```

Then on Wiki

```
I2SStream i2sStream;
auto cfg = i2sStream.defaultConfig(RX_MODE);
cfg.i2s_format = I2S_STD_FORMAT; // or try with I2S_LSB_FORMAT
cfg.bits_per_sample = 32;
cfg.channels = 2; // optional because default setting
cfg.sample_rate = 44100;
cfg.is_master = true; // optional because default setting
// this module needs a master clock if the ESP32 is master
cfg.use_apll = false;  
cfg.pin_mck = 3; 
i2sStream.begin(cfg);
```

And finally master mode

```
    auto cfg = i2sStream.defaultConfig(RX_MODE);
    cfg.copyFrom(info);
    cfg.i2s_format = I2S_STD_FORMAT; // or try with I2S_LSB_FORMAT
    cfg.is_master = true;
     // this module nees a master clock if the ESP32 is master
    cfg.use_apll = false;  // try with yes
    cfg.pin_mck = 3; 
    i2sStream.begin(cfg);
```

If I set the jumper to MASTER (default) on the module, I think I need to set `is_master` to false then. But in either case, I'm getting no output from the `copier`

```cpp
#include <Arduino.h>
#include "AudioTools.h"

AudioInfo info(44100, 2, 32);
I2SStream i2sStream; // Access I2S as stream
CsvOutput<int16_t> csvStream(Serial);
StreamCopy copier(csvStream, i2sStream); // copy i2sStream to csvStream

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  
  auto cfg = i2sStream.defaultConfig(RX_MODE);

  if (1) {
    cfg.i2s_format = I2S_MSB_FORMAT; // or try with I2S_LSB_FORMAT
    cfg.bits_per_sample = 32;
    cfg.channels = 2; // optional because default setting
    cfg.sample_rate = 44100;
    cfg.is_master = false;
    cfg.use_apll = true;
    cfg.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT;
  } else {
    cfg.copyFrom(info);
    cfg.i2s_format = I2S_STD_FORMAT; // or try with I2S_LSB_FORMAT
    cfg.is_master = true;
     // this module nees a master clock if the ESP32 is master
    cfg.use_apll = false;  // try with yes
  }

  cfg.pin_ws = 22;
  cfg.pin_data = 19;
  cfg.pin_bck = 23;

  bool i2sStreamStatus = i2sStream.begin(cfg);
  log_i("i2sStreamStatus %d", i2sStreamStatus);

  // make sure that we have the correct channels set up
  bool csvStreamStatus = csvStream.begin(info);

  log_i("csvStreamStatus %d", csvStreamStatus);
}

// Arduino loop - copy data
void loop() {
  log_i("loop");
  copier.copy();
}
```

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (282) esp_core_dump_flash: Core dum"�х�check failed:
Calculated checksum='2d662ddc'
Image checksum='972e7f2a'
[    32][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    93][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   102][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[   113][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[D] I2SStream.h : 71 - bool audio_tools::I2SStream::begin(audio_tools::I2SConfig)
[D] BaseStream.h : 118 - virtual void audio_tools::AudioStream::setAudioInfo(audio_tools::AudioInfo)
[I] AudioTypes.h : 128 - in: sample_rate: 44100 / channels: 2 / bits_per_sample: 32
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 32
[D] I2SESP32.h : 63 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfigESP32)
[D] I2SESP32.h : 182 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfigESP32, int, int)
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 32
[I] I2SConfigESP32.h : 80 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Slave
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 32
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_MSB_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: false
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 103 - pin_bck: 23
[I] I2SConfigESP32.h : 105 - pin_ws: 22
[I] I2SConfigESP32.h : 107 - pin_data: 19
[D] I2SESP32.h : 214 - i2s_driver_install
[D] I2SESP32.h : 230 - i2s_set_pin
[D] I2SESP32.h : 241 - i2s_zero_dma_buffer
[D] I2SESP32.h : 245 - begin - started
[   247][I][main.cpp:37] setup(): i2sStreamStatus 1
[D] AudioOutput.h : 148 - bool audio_tools::CsvOutput<T>::begin(int) [with T = short int]
[   262][I][main.cpp:42] setup(): csvStreamStatus 1
[   274][I][main.cpp:47] loop(): loop
[D] StreamCopy.h : 98 - copy 1024 bytes 
[D] StreamCopy.h : 192 - available: 3072
```

The loop is printed only once. The copier locks the app because no data available.

But I do see signal on the DATA pin 
[image]

And the scope sees a 12MHz clock on the BLCK.

pschatzmann, 2024-09-16:
There is a chapter about ADCs in the Wiki!

dizcza, 2024-09-16:
Actually, I've read it. But it took some trial and error to understand it.

All right, I managed to poll this I2S mode in both master and slave modes. There are other issues and questions, however. Here is the updated code:

```cpp
#include <Arduino.h>
#include "AudioTools.h"

AudioInfo info(48000, 1, 32);
I2SStream i2sStream;
CsvOutput<int16_t> csvStream(Serial);
StreamCopy copier(csvStream, i2sStream);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  
  auto cfg = i2sStream.defaultConfig(RX_MODE);
  cfg.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;
  cfg.copyFrom(info);
  cfg.i2s_format = I2S_STD_FORMAT;  // MSB & STD work
  cfg.use_apll = false;

  cfg.is_master = true;
  if (cfg.is_master) {
    cfg.pin_mck = 3;
  }

  cfg.pin_ws = 22;
  cfg.pin_data = 19;
  cfg.pin_bck = 23;

  bool i2sStreamStatus = i2sStream.begin(cfg);
  log_i("i2sStreamStatus %d", i2sStreamStatus);

  // make sure that we have the correct channels set up
  bool csvStreamStatus = csvStream.begin(info);

  log_i("csvStreamStatus %d", csvStreamStatus);
}

void loop() {
  copier.copy();
}
```

Just to be clear, I'm reading a 3.3V VCC output pin from an ESP32 connected to the RIGHT jacket of this module. The module is powered with 5V.

The bitness jumper must be set to 16 bits (not 24 bits). Otherwise, it does not work in either mode.

1. It doesn't work when I set `CsvOutput<int32_t> csvStream(Serial)` (I see garbage). Only `int16_t` works with this module. The opposite is stated in the streams-i2s-serial example.
2. I see ~32k spikes when I enable the APLL. Without APLL, the output is clean. Could it be an issue on the ESP-IDF side?
3. I expect the output to be a nonzero constant value that scales 3.3V to 5.0V max: 3.3/5.0 * INT16_MAX. The values I get are always around zero. Are they AC coupled? How to turn this filtering off?
4. How to convert raw int16_t values back to voltage?
5. Are there ESP32 pins that could provide CLOCK OUT other than GPIO3?

Maybe you're not the right person to ask. But I imagine you're more experienced in this topic than I am anyway.

pschatzmann, 2024-09-16:
- AudioInfo info(48000, 1, 32); does not match with CsvOutput<int16_t> it must be CsvOutput<int32_t>!  and this does not match whith the settings on your picture! But I guess these settings might be only relevant for the esp32 as master 
- If I remember right, I use the APLL by default only in output mode and not in input mode.
- Audio is always oscillating around 0 with a value range defined by the bits and your question about converting it back to voltage does not make any sense w/o context.
- Google the supported GPIOs: if I remember right it is only 0 and 3.

dizcza, 2024-09-16:
> ps. What has this to do with WiFiClientSecure not found ?

Nothing. I had a compilation issue, it's solved. I could start a discussion but decided to proceed here.

> Audio is always oscillating around 0 with a value range defined by the bits and your question about converting it back to voltage does not make any sense w/o context.

This is correct about audio. But we are talking about an ADC by means of I2S. The only reason I bought this module is to read an analog sensor with I2S using ESP32. Previously I tried more advanced ADCs like ADS1256 connected to ESP32 via SPI. Analog values are absolute. And I need to convert values to voltage just like I'd do with any ADC raw output.

Is it possible to sample absolute voltage with such an I2S ADC module?

----

If I set `AudioInfo info(96000, 1, 32)` as in your blog example (the sample rate jumper is placed accordingly on the module), I get

```
E (95) I2S: i2s_calculate_common_clock(1181): sample rate is too large
E (96) I2S: i2s_calculate_clock(1230): Common clock calculate failed
```

dizcza, 2024-09-16:
> E (95) I2S: i2s_calculate_common_clock(1181): sample rate is too large
E (96) I2S: i2s_calculate_clock(1230): Common clock calculate failed

I don't get these errors with APLL enabled. Weird.