pschatzmann/arduino-audio-tools #1701: class ConverterSwitchLeftAndRight

fobi55, 2024-09-16:
### Problem Description

Typo error:
`/**
 * @brief Switches the left and right channel
 * @ingroup convert
 * @author Phil Schatzmann
 * @copyright GPLv3
 *
 * @tparam T
 */
template <typename T>
class ConverterSwitchLeftAndRight : public BaseConverter {
 public:
  ConverterSwitchLeftAndRight(int channels = 2) { this->channels = channels; }

  size_t convert(uint8_t *src, size_t byte_count) override  {
    if (channels == 2) {
      int size = byte_count / channels / sizeof(T);
      T *sample = (T *) src;
      for (size_t j = 0; j < size; j++) {
        T temp = *sample;
        *sample = *(sample + 1) ;
        *(sample + 1) = temp;
        *sample += 2;
      }
    }
    return byte_count;
  }`

### Device Description

NodeMCU-32S

### Sketch

```cpp
/**
 * @file streams-generator-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-i2s/README.md 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */
 
#include "AudioTools.h"
#include "coeffs60.h"

#define MAX_VOLUME 32767

typedef int16_t sound_t;                                   // sound will be represented as int16_t (with 2 bytes)
uint16_t sample_rate=44100;
uint8_t channels = 2;                                      // The stream will have 2 channels 

SineWaveGenerator<sound_t> sineWave(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<sound_t> sound(sineWave);             // Stream generated from sine wave
AnalogAudioStream out;                                     //internal DAC output: FOB
//I2SStream out; 
//StreamCopy copier(out, sound);                           // copies sound into i2s, but we need filter

volatile float volFactor = 0.4;
//ConverterScaler<int16_t> vol(volFactor, 0, MAX_VOLUME);   // sound is generated as <int16_t>
ConverterSwitchLeftAndRight<int16_t> swLR(2);             // must have 2 as argument!

// copy filtered values
FilteredStream<int16_t, float> filtered(sound, channels);  // Defiles the filter as BaseConverter
StreamCopy copier(out, filtered, 512);                     // copies filtered into i2s/DAC : FOB

// Timer0 Configuration Pointer (Handle)
hw_timer_t *Timer0_Cfg = NULL;

volatile int counter = 0;
// The Timer0 ISR Function (Executes Every Timer0 Interrupt Interval)
void IRAM_ATTR Timer0_ISR() {
  counter++;
}

//*******************************************************************************************************************
// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Error);

    // Configure Timer0 Interrupt 10000/second
  Timer0_Cfg = timerBegin(0, 10000, true);
  timerAttachInterrupt(Timer0_Cfg, &Timer0_ISR, true);
  timerAlarmWrite(Timer0_Cfg, 10, true);
  timerAlarmEnable(Timer0_Cfg);

  pinMode(26, OUTPUT);
  pinMode(27, OUTPUT);

  // 90 degree phase shift 60 taps (~4.5 KHz cutoff)
  filtered.setFilter(0, new FIR<float>(coeffs_60plus45));
  filtered.setFilter(1, new FIR<float>(coeffs_60minus45));

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);   // DAC has only TX_MODE!
  config.sample_rate = sample_rate;
  config.bits_per_sample = 16;
  /*
  config.i2s_format = I2S_STD_FORMAT;
  config.is_master = true;
  config.port_no = 0;
  config.pin_ws = 18;
  config.pin_bck = 5;
  config.pin_data = 19;
  config.pin_data_rx = 17;
  config.pin_mck = 0;
  config.use_apll = true;  
  */
  out.begin(config);

  // Setup sine wave
//  sineWave.begin(channels, sample_rate, N_F4);
  sineWave.begin(channels, sample_rate, 3000);
  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {

  if (counter > 1000) { // 1ms timer
    counter = 0;
    /*
    volFactor = volFactor + 0.1;
    if(volFactor > 1){volFactor = 0;}
    vol.setFactor(volFactor);
    Serial.println(vol.factor());
    */
  }

  if (counter < 500) {
    copier.copy(swLR);
  }

   if (counter > 500) {
     copier.copy();
  }

  //copier.copy(vol);

}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-16:
I don't understand what the issue is. Please provide a description and a reproducible example!
If you now how to correct it don't hesitate to submit a pull request

pschatzmann, 2024-09-16:
Sorry, I stll don't see it and this is impossible to compare! What line did you change ? Where is the missspelling ?

fobi55, 2024-09-16:
It was something like that:

*sample = *(sample + 1) *(sample + 1) = temp;
sample += 2;

I changed it to:

*sample = *(sample + 1) ;
*(sample + 1) = temp;
*sample += 2;

Again, sorry, I'm new to the git. And I am HW engineer ... :)

fobi55, 2024-09-16:
Here, I found the ZIP file:

/**
 * @brief Switches the left and right channel
 * @ingroup convert
 * @author Phil Schatzmann
 * @copyright GPLv3
 *
 * @tparam T
 */
template <typename T>
class ConverterSwitchLeftAndRight : public BaseConverter {
 public:
  ConverterSwitchLeftAndRight(int channels = 2) { this->channels = channels; }

  size_t convert(uint8_t *src, size_t byte_count) override {
    if (channels == 2) {
      int size = byte_count / channels / sizeof(T);
      T *sample = (T *)src;
      for (size_t j = 0; j < size; j++) {
        T temp = *sample;
        *sample = *(sample + 1) * (sample + 1) = temp;  <<<<<<<<<<<<<<<<<<<<<<<<,,
        sample += 2; <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
      }
    }
    return byte_count;
  }

 protected:
  int channels = 2;
};

fobi55, 2024-09-16:
Apologies, my mistake.
This should be as is:
sample += 2; 

`/**
 * @brief Switches the left and right channel
 * @ingroup convert
 * @author Phil Schatzmann
 * @copyright GPLv3
 *
 * @tparam T
 */
template <typename T>
class ConverterSwitchLeftAndRight : public BaseConverter {
 public:
  ConverterSwitchLeftAndRight(int channels = 2) { this->channels = channels; }

  size_t convert(uint8_t *src, size_t byte_count) override  {
    if (channels == 2) {
      int size = byte_count / channels / sizeof(T);
      T *sample = (T *) src;
      for (size_t j = 0; j < size; j++) {
        T temp = *sample;
        *sample = *(sample + 1) ;
        *(sample + 1) = temp;
        sample += 2;
      }
    }
    return byte_count;
  }

 protected:
  int channels = 2;
};`

pschatzmann, 2024-09-16:
I see, somwhow a semicolon got deleted along the way.
I committed the correction and added it back...

closed 2024-09-16 by pschatzmann