pschatzmann/arduino-audio-tools #1720: Volume up and volume down steps incorrect in AudioBoardStream.h in actionVolumeUp and actionVolumeDown

Jvsjvs, 2024-09-27:
### Problem Description

When the default actions are used the methods: actionVolumeUp and actionVolumeDown are used to control the volume in the "MP3 player" example. actionVolumeUp and actionVolumeDown try to increase and decrease the volume with a step of 2.0f using the incrementVolume method.

However, the volume range is between 0.0f and 1.0f, hence the volume up action sets the volume to max.

When I change this increment to +0.1f and -0.1f the volume buttons work as expected.

### Device Description

Any device, but I am using the AI Tinker audio board.

### Sketch

```cpp
void incrementVolume(float inc) {
    float current_volume = getVolume();
    float new_volume = current_volume + inc;
    LOGI("incrementVolume: %f -> %f", current_volume, new_volume);
    setVolume(new_volume);
  }

  /**
   * @brief Increase the volume
   *
   */
  static void actionVolumeUp(bool, int, void *) {
    TRACEI();
    selfAudioBoard->incrementVolume(+2);
  }

  /**
   * @brief Decrease the volume
   *
   */
  static void actionVolumeDown(bool, int, void *) {
    TRACEI();
    selfAudioBoard->incrementVolume(-2);
  }
```

### Other Steps to Reproduce

Run the "MP3 player" example and press the volume up and volume down buttons.

### What is your development environment

PlatformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-27:
You are right. I was converting this method from the old AudioKitStream and there the range was from 0 to 100.
I corrected it to 0.02!

closed 2024-09-27 by pschatzmann

Jvsjvs, 2024-09-28:
Thanks for fixing it this quickly.

Btw, great libraries you have built, very nice!

I personally changed the +0.02 and -0.02 to +0.05 and -0.05, this works better with the twitchy small buttons on the AI Thinker audio kit.