pschatzmann/arduino-audio-tools #1722: error: no match for 'operator[]' in buffers.h

mutedh, 2024-09-29:
### Problem Description

I try to use the example "basic-a2dp-eq-audiokit.ino" in AI Thinker Audiokit, but when i upload the example i recive this  error:

C:\Users\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Buffers.h: In instantiation of 'void audio_tools::NBuffer<T>::resize(int, int) [with T = unsigned char]':
C:\Users\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Buffers.h:701:16:   required from here
C:\Users\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Buffers.h:711:24: error: no match for 'operator[]' (operand types are 'audio_tools::QueueFromVector<audio_tools::BaseBuffer<unsigned char>*>' and 'int')
  711 |       available_buffers[j] = new SingleBuffer<T>(size);
      |       ~~~~~~~~~~~~~~~~~^
C:\Users\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Buffers.h:712:28: error: no match for 'operator[]' (operand types are 'audio_tools::QueueFromVector<audio_tools::BaseBuffer<unsigned char>*>' and 'int')
  712 |       if (available_buffers[j] == nullptr) {
  
  what can be wrong?

### Device Description

AI-Thinker AudioKit

### Sketch

```cpp
basic-a2dp-eq-audiokit.ino example
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-29:
I think you picked up the audiotools in some inconsistent state.
Try again with the latest corrections and let me know if this did not help

closed 2024-09-29 by pschatzmann

mutedh, 2024-09-29:
i did a pull now at works, thanks :)

mutedh, 2024-10-03:
Hello, i have one question, i use sigmadsp to control a dsp ADAU1701 with esp32-a1s but i think it doesnt work a2dp and i2c, . could be?

because when i use a2dp, the i2c doesnt send nothing.but if i dont use a2dp, the i2c send the information.

here is the code: 

`/**********************************************|
| SigmaDSP library                             |
| https://github.com/MCUdude/SigmaDSP          |
|                                              |
| 1_Volume1.ino                                |
| This example fades the volume up and down    |
| using a slew volume adjustment slider.       |
| See the SigmaStudio project file if you want |
| to learn more, tweak or do modifications.    | 
|**********************************************/

// Include Wire and SigmaDSP library
#include <Wire.h>
#include <SigmaDSP.h>

// Include generated parameter file
#include "SigmaDSP_parameters.h"

#include "AudioTools.h"
#include "AudioLibs/A2DPStream.h" // install https://github.com/pschatzmann/ESP32-A2DP
#include "AudioLibs/AudioBoardStream.h" // install https://github.com/pschatzmann/arduino-audio-driver
// The first parameter is the Wire object we'll be using when communicating wth the DSP
// The second parameter is the DSP i2c address, which is defined in the parameter file
// The third parameter is the sample rate
// An optional fourth parameter is the pin to physically reset the DSP
SigmaDSP dsp(Wire, DSP_I2C_ADDRESS, 48000.00f /*,12*/);
BluetoothA2DPSink a2dp_sink;
AudioBoardStream kit(AudioKitEs8388V1); 

// Only needed if an external i2c EEPROM is present + the DSP is in selfboot mode
// The first parameter is the Wire object we'll be using when communicating wth the EEPROM
// The second parameter is the EEPROM i2c address, which is defined in the parameter file
// The third parameter is the EEPROM size in kilobits (kb)
// An optional fourth parameter is the pin to toggle while writing content to EEPROM
//DSPEEPROM ee(Wire, EEPROM_I2C_ADDRESS, 256, LED_BUILTIN);
void read_data_stream(const uint8_t *data, uint32_t length) {
    kit.write(data, length);
}

void setup() 
{  
  Serial.begin(9600);
  Serial.println(F("SigmaDSP 1_Volume example\n"));

  Wire.begin();
  dsp.begin();
  //ee.begin();

  delay(2000);

  
  Serial.println(F("Pinging i2c lines...\n0 -> deveice is present\n2 -> device is not present"));
  Serial.print(F("DSP response: "));
  Serial.println(dsp.ping());
  //Serial.print(F("EEPROM ping: "));
  //Serial.println(ee.ping());
  
 
  // Use this step if no EEPROM is present
  Serial.print(F("\nLoading DSP program... "));
  loadProgram(dsp);
  Serial.println("Done!\n");

auto cfg = kit.defaultConfig(TX_MODE);
  cfg.sd_active = false;
  kit.begin(cfg);

  // register callback
  a2dp_sink.set_stream_reader(read_data_stream, false);
    a2dp_sink.start("AudioKit");
  // Comment out the three code lines above and use this step instead if EEPROM is present
  // The last parameter in writeFirmware is the FW version, and prevents the MCU from overwriting on every reboot
  //ee.writeFirmware(DSP_eeprom_firmware, sizeof(DSP_eeprom_firmware), 0);
  //dsp.reset();
  //delay(2000); // Wait for the FW to load from the EEPROM
}

void loop() 
{
  // Slowly adjust the volume down to -50dB
  Serial.println(F("Fading down"));
  for(int8_t i = 0; i > -51; i--)
  {
    dsp.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, i, 12); // (First volume slider address, dB value, optional slew value)
    delay(200);
  }
  
  // Slowly adjust the volume up to 0dB
  Serial.println(F("Fading up"));
  for(int8_t i = -50; i < 1; i++)
  {
    dsp.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, i); // Slew rate is default 12
    delay(200);
  }
}
`

pschatzmann, 2024-10-03:
Please create a new discussion