pschatzmann/arduino-audio-tools #1723: WAVDecoder memory leak?

y-nk, 2024-09-30:
### Problem Description

I'm trying to use the WAVDecoder to play a wav file and found this.

I've tried with Helix decoder and it plays without much trouble, but the sound comes in the speaker by chunks with blanks of approx 200ms, so I thought to try wav instead. Only swapping the decoder leads to a panic.

### Device Description

ESP32-D0WD-V3

### Sketch

```cpp
I've swapped the MP3Decoder for the WAVDecoder + use a dedicated SPI for the AudioSourceSD.

/**
 * @file player-sd-i2s.ino
 * @brief example using the SD library
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/A2DPStream.h"
#include "AudioLibs/AudioSourceSD.h"
#include "AudioCodecs/CodecWAV.h"

#define HSPI_SCLK 26
#define HSPI_MISO 33
#define HSPI_MOSI 25
#define HSPI_SS   27

const char *startFilePath="/";
const char *ext="wav";

SPIClass hspi(HSPI);
AudioSourceSD source(startFilePath, ext, HSPI_SS, hspi);
A2DPStream out;
WAVDecoder decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);

  // setup output
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "Bose Color II SoundLink";
  cfg.auto_reconnect = true;
  out.begin(cfg);

  // setup player
  player.begin();
}

void loop() {
  player.copy();
}
```

### Other Steps to Reproduce

I've ran the exception decoder and found this:

```
PC: 0x40083d80 EXCVADDR: 0x00000000  Decoding stack results
  0x400e71fa: pthread_key_create at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread_local_storage.c line 56
  0x401a2e0d: mbedtls_ecdsa_read_signature_restartable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecdsa.c line 762
  0x400df265: SPIClass::endTransaction() at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI/src/SPI.cpp line 196
  0x400dec09: VFSImpl::mkdir(char const*) at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src/vfs_api.cpp line 215
  0x400deb0d: VFSImpl::remove(char const*) at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src/vfs_api.cpp line 438
  0x40099171: prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c line 587
  0x40099bae: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162
```

...which involves the SPI class. I have been reporting this to esp32-idf but since swapping the decoders lead to that panic, I'm still unsure it is not related to this library.

### What is your development environment

Arduino IDE latest

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-30:
Did you try to test the wav decoder e.g. by just swapping out the output with a CSVOutput ?
You could also set the log level to Info or Debug to get a better feeling at what step this is crashing...

pschatzmann, 2024-09-30:
The WAV codec is quite lean and stable I do not see any risks of memory violations there.
Therefore I am closing this issue unless you can proove it otherwise.

Please double check that you are using the latest esp32 3.0.5 version and if you do try to doubel check with the old 2.0.17

closed 2024-09-30 by pschatzmann

y-nk, 2024-09-30:
Thank you for the suggestion of swapping outputs. I'll try that.