pschatzmann/arduino-audio-tools #1724: MP3 libMAD memory leak?

y-nk, 2024-09-30:
### Problem Description

I've been trying to swap Codec Helix to the MAD one. At build I had a bunch of warnings such as these ones:

```
In function 'sdctII',
    inlined from 'dctIV' at /Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1735:3,
    inlined from 'imdct36' at /Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1761:3,
    inlined from 'III_imdct_l' at /Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:2086:3:
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1697:3: warning: 'fastsdct' accessing 72 bytes in a region of size 68 [-Wstringop-overflow=]
 1697 |   fastsdct(tmp, &X[1]);
      |   ^~~~~~~~~~~~~~~~~~~~
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1697:3: note: referencing argument 2 of type 'mad_fixed_t[18]' {aka 'int[18]'}
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c: In function 'III_imdct_l':
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1597:6: note: in a call to function 'fastsdct'
 1597 | void fastsdct(mad_fixed_t const x[9], mad_fixed_t y[18])
      |      ^~~~~~~~
In function 'sdctII',
    inlined from 'dctIV' at /Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1735:3,
    inlined from 'imdct36' at /Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1761:3,
    inlined from 'III_imdct_l' at /Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:2086:3:
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1697:3: warning: 'fastsdct' accessing 72 bytes in a region of size 68 [-Wstringop-overflow=]
 1697 |   fastsdct(tmp, &X[1]);
      |   ^~~~~~~~~~~~~~~~~~~~
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1697:3: note: referencing argument 2 of type 'mad_fixed_t[18]' {aka 'int[18]'}
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c: In function 'III_imdct_l':
/Users/julien/Documents/Arduino/libraries/arduino-libmad/src/libmad/layer3.c:1597:6: note: in a call to function 'fastsdct'
 1597 | void fastsdct(mad_fixed_t const x[9], mad_fixed_t y[18])
```

But then when running, I have a panic with the following stacktrace:

```
Decoding stack results
0x40082669: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 466
0x40096705: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 84
0x4009c8be: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 81
0x400970ea: vQueueDelete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c line 2300
0x40099bae: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162
```

### Device Description

ESP32-D0WD-V3

### Sketch

```cpp
/**
 * @file player-sd-i2s.ino
 * @brief example using the SD library
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/A2DPStream.h"
#include "AudioLibs/AudioSourceSD.h"
#include "AudioCodecs/CodecMP3MAD.h"

#define HSPI_SCLK 26
#define HSPI_MISO 33
#define HSPI_MOSI 25
#define HSPI_SS   27

const char *startFilePath="/";
const char *ext="mp3";

SPIClass hspi(HSPI);
AudioSourceSD source(startFilePath, ext, HSPI_SS, hspi);
A2DPStream out;
MP3DecoderMAD decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);

  // setup output
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "Bose Color II SoundLink";
  cfg.auto_reconnect = true;
  out.begin(cfg);

  // setup player
  player.begin();
}

void loop() {
  player.copy();
}
```

### Other Steps to Reproduce

Swapping the Helix decoder back looks like working fine except for the hashed output on the speaker.

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-30:
What esp32 core version are you using ? 
Did you try with another version ?
Did you also try to optimize the A2DP configuration settings ?

I think CodecMP3MAD is using much more resources so I don't think that this is a good idea here...

y-nk, 2024-09-30:
1. I'm using 3.0.5 esp core from expressif.
2. I did NOT try an older version (I should know better but still think latest is best).
3. Until you mentioned it, I did not know there were more than `auto_reconnect` but `buffer_size` and `delay_ms` are things I'm gonna play with now.
4. I agree that MAD seems beefier (at least in usage), I was looking for a quick win but thought to report my findings on the way.
5. Aside from the panic I've decoded (using the Arduino v1 plugin), none.
6. Yes, I usually do so now, as you've taught me :) As far as I can understand the logs, there's nothing wrong there.

I've explored further how to achieve my project, trying to make the Helix example work flawlessly and found another panic when using Helix and a 64kbps mp3. I'm starting to wonder if it's not all the fault of the uSD reader i own + the custom VSPI pins i'm using.

pschatzmann, 2024-09-30:
I think you should wirte some [test cases to pin down the issue](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)

pschatzmann, 2024-10-02:
My conclusion is that you are running out of RAM. A2DP is not leaving much RAM available and MP3 decoding also needs considerable RAM.

Monitor the RAM usage and better activate and use PSRAM

closed 2024-10-02 by pschatzmann

y-nk, 2024-10-03:
Thank you, it seems a good diagnostic indeed.