pschatzmann/arduino-audio-tools #1725: Previously working code now constantly reboots.

mooballs, 2024-09-30:
### Problem Description

Code that worked on an older version (6 months or more) of the library now just reboots. Couldn't see anything obvious in the Volume Control or Equilizer wiki entries, apologies if its something silly.

### Device Description

ESP32-WROOM-32D

### Sketch

```cpp
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioConfigLocal.h"
#include "BluetoothA2DPSinkQueued.h"  //trying this for ipad issue

AudioInfo info(44100, 2, 16);  //configuration for out1
//BluetoothA2DPSink a2dp_sink;
BluetoothA2DPSinkQueued a2dp_sink;  //trying this for ipad issue
A2DPStream in;                      // had to copy a2dp in to out, then wrap out in volume and eq then use multioutput to get vol_info
MultiOutput out;
StreamCopy copier(out, in);  //copying in to out
VolumePrint out1;            //changed from vol.info
I2SStream out2;              //changed from i2s
//A2DPStream in;
VolumeStream volume(out);
LinearVolumeControl(lvc);  //this is for using the log pot

Equilizer3Bands eq(volume);  //must wrap the stream up correctly
//StreamCopy copier(eq, i2s); // copy in to equilizer
ConfigEquilizer3Bands cfg_eq;

#define I2S_BUFFER_SIZE 1024  //default is 512
#define I2S_BUFFER_COUNT 8    //default is 6

int VolLevel = 0;

void writeDataStream(const uint8_t *data, uint32_t length) {
  out1.write(data, length);
}

void read_data_stream(const uint8_t *data, uint32_t length) {
  eq.write(data, length);
}

// Arduino Setup
void setup() {

  Serial.begin(115200);

  out.add(out1);  //multi-outs
  out.add(out2);

  AudioLogger::instance().begin(Serial, AudioLogger::Warning);  //tried setting to Warning from Info to aid audio quality

  out1.begin(info);  //begin stream

  //i2s_out esp32 external DAC
  auto config = out2.defaultConfig(TX_MODE);
  config.pin_bck = 18;         
  config.pin_ws = 25;          
  config.pin_data = 19;        
  config.sample_rate = 44100;  //changed from a2dp_sink.sample_rate();
  config.channels = 2;
  config.bits_per_sample = 16;
  config.buffer_size = I2S_BUFFER_SIZE;
  config.buffer_count = I2S_BUFFER_COUNT;
  out2.begin(config);

  cfg_eq = eq.defaultConfig();
  cfg_eq.setAudioInfo(config);  // use channels, bits_per_sample and sample_rate from kit

  cfg_eq.gain_high = 0.6;
  cfg_eq.gain_medium = 0.5;
  cfg_eq.gain_low = 0.6;

  eq.begin(cfg_eq);

  // set initial volume
  volume.begin(config);  // we need to provide the bits_per_sample and channels
  volume.setVolume(1.0);
  volume.setVolumeControl(lvc);  //for using log pot
  // register callback

  a2dp_sink.set_stream_reader(writeDataStream, false);
  a2dp_sink.set_stream_reader(read_data_stream, false);

  a2dp_sink.set_i2s_ringbuffer_size(65536);  // try this for ipad (double the default)

  Serial.print("starting a2dp...");
  // Start Bluetooth Audio Receiver
  a2dp_sink.set_auto_reconnect(true);
  a2dp_sink.start("Radio");  //bluetooth name
}

// Arduino loop
void loop() {
  copier.copy();  
}
```

### Other Steps to Reproduce

Exception Decoder o/p with line from src in **bold** 

Decoding stack results
0x40083bb1: efuse_hal_flash_encryption_enabled at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/efuse_hal.c line 22
0x400965fd: taskYIELD_OTHER_CORE at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c line 653
0x4009c8e1: spi_flash_chip_winbond_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/spi_flash/spi_flash_chip_winbond.c line 60
0x40099e9e: wdt_hal_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/rwdt_ll.h line 163
0x400d2721: audio_tools::BufferRTOS ::readArray(unsigned char*, int) at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/Concurrency/BufferRTOS.h line 77    **int readArray(T data[], int len) {**

0x400d2736: audio_tools::int24_4bytes_t::set(int const&) at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioBasic/Int24_4bytes_t.h line 57            **value = INT24_MAX << 8;**

0x400d5213: audio_tools::StreamCopyT ::write(unsigned int, unsigned int&) at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h line 85                 **return log_level;**

0x400d6653: _GLOBAL__sub_I_info() at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h line 57                **printPrefix(file,line,current_level);**

0x400d67f8: _GLOBAL__sub_I_info() at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioTypes.h line 205      **class VolumeSupport {**

0x400d9a11: gpio_pulldown_en at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h line 578

### What is your development environment

Arduino IDE 2.3.2

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-09-30:
Your sketch is a bad mess: I suggest that you clean it up first and take it up from  there!
Remove all the unused functionality like  writeDataStream, A2DPStream, StreamCopy.

I would not see how you would be ending up using any 24 bit logic...

mooballs, 2024-09-30:
Thanks mate, I'm using all that functionality to output the volume level as a number so that I can light some LEDs using FASTLED. Have updated the code with a bare bones usage, the exception decoder output was the same except for this:

0x400d6657: _GLOBAL__sub_I_info() at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/A2DPStream.h line 82                 **assert(A2DPStream_self==nullptr);
            A2DPStream_self = this;**

pschatzmann, 2024-09-30:
No, you don't: The A2DPStream is not used. You use the BluetoothA2DPSinkQueued directly via the callback!
It's only one or the other!

mooballs, 2024-09-30:
Ok, I was just using it as you'd told me previously 

https://github.com/pschatzmann/arduino-audio-tools/discussions/677

Either way, it used to work and now it doesn't.

mooballs, 2024-09-30:
What would be the best way of getting the volume info using the callback? You mentioned examples a while ago but I couldn't see them.

pschatzmann, 2024-09-30:
Just write to the VolumeMeter as you do for the Equilizer

mooballs, 2024-09-30:
OK, many thanks, sorry you had to use so many !s. I'll mark it as closed and go away and try to implement it.

closed 2024-09-30 by mooballs

mooballs, 2024-10-01:
Ok, so I've tried this and it's just returning zeros. I want the volume level of the output post processing, after eq and volume control. 

`
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioConfigLocal.h"
#include "BluetoothA2DPSinkQueued.h"  //trying this for ipad issue

AudioInfo info(44100, 2, 16);  //configuration for out1
BluetoothA2DPSinkQueued a2dp_sink;  //trying this for ipad issue
I2SStream i2s;
VolumeStream volume(i2s);
LinearVolumeControl(lvc);  //this is for using the log pot

Equilizer3Bands eq(volume);  //must wrap the stream up correctly

VolumeMeter meter(eq);

#define I2S_BUFFER_SIZE 1024  //default is 512
#define I2S_BUFFER_COUNT 8    //default is 6

ConfigEquilizer3Bands cfg_eq;

int VolLevel = 0;

void read_data_stream(const uint8_t *data, uint32_t length) {
  eq.write(data, length);
}

// Arduino Setup
void setup() {

  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Warning);  //tried setting to Warning from Info to aid audio quality

  //i2s_out esp32 external DAC
  auto config = i2s.defaultConfig(TX_MODE);
  config.pin_bck = 18;         
  config.pin_ws = 25;          
  config.pin_data = 19;        
  config.sample_rate = 44100;  //changed from a2dp_sink.sample_rate();
  config.channels = 2;
  config.bits_per_sample = 16;
  config.buffer_size = I2S_BUFFER_SIZE;
  config.buffer_count = I2S_BUFFER_COUNT;
  i2s.begin(config);

  cfg_eq = eq.defaultConfig();
  cfg_eq.setAudioInfo(config);  // use channels, bits_per_sample and sample_rate from kit

  cfg_eq.gain_high = 0.6;
  cfg_eq.gain_medium = 0.5;
  cfg_eq.gain_low = 0.6;

  eq.begin(cfg_eq);

  // set initial volume
  volume.begin(config);  // we need to provide the bits_per_sample and channels
  volume.setVolume(1.0);
  volume.setVolumeControl(lvc);  //for using log pot
  // register callback

  meter.begin(info);

 
  a2dp_sink.set_stream_reader(read_data_stream, false);

  Serial.print("starting a2dp...");
  // Start Bluetooth Audio Receiver
  a2dp_sink.set_auto_reconnect(true);
  a2dp_sink.start("Radio");  //bluetooth name
}

// Arduino loop
void loop() {
int16_t volume = meter.volume();
		Serial.print("Volume: ");
		Serial.println(volume);
}`

pschatzmann, 2024-10-01:
You do not provide any data to the VolumeMeter, so it is normal that you get 0. Use:

```cpp

VolumeMeter meter;

void read_data_stream(const uint8_t *data, uint32_t length) {
eq.write(data, length);
meter.write(data, length);
}
```

mooballs, 2024-10-01:
Of course, many thanks.

mooballs, 2024-10-10:
Okay, so I got it working with the above code this way:
```
BluetoothA2DPSinkQueued a2dp_sink;  
I2SStream i2s;
VolumeStream volume(i2s);
LinearVolumeControl(lvc);  
Equilizer3Bands eq(volume);  

VolumeMeter meter;

```

And

```
void read_data_stream(const uint8_t *data, uint32_t length) {                           
  eq.write(data, length);
  meter.write(data, length);
}
```

But there was some glitching/drop outs when you played the first track via bluetooth.

So I removed this line 
```
a2dp_sink.set_stream_reader(read_data_stream, true);

```
and tried this:
```
I2SStream i2s;
VolumeStream volume(i2s);
LinearVolumeControl(lvc);  
Equilizer3Bands eq(volume);  
BluetoothA2DPSinkQueued a2dp_sink(eq);  

```
Which solved the drop outs but I’ve lost the VolumeMeter functionality.

I tried this
```
VolumeMeter meter;

void read_data_stream(const uint8_t *data, uint32_t length) {  
  eq.write(data, length);
  meter.write(data, length);
}
```

And this

```
VolumeMeter meter(eq);
void read_data_stream(const uint8_t *data, uint32_t length) {  
  
  meter.write(data, length);
}
```

But I’m not getting the volume info. Apologies for being so dumb, but where do I take the VolumeMeter feed from?

pschatzmann, 2024-10-10:
Just put the VolumeMeter in the chain where you want to measure: if you want to measure what you receive, put it before the Equilizer3Bands; if you want to measure what you hear: put it just before I2SStream.

And delete the LinearVolumeControl which is not used anywhere.

mooballs, 2024-10-11:
Thanks again, I did this:
```
I2SStream i2s;
VolumeMeter meter(i2s);    
VolumeStream volume(meter);
Equilizer3Bands eq(volume); 
BluetoothA2DPSinkQueued a2dp_sink(eq);
```
And this:
```
void read_data_stream(const uint8_t *data, uint32_t length) {  
  meter.write(data, length);
}
```

pschatzmann, 2024-10-11:
Remove the callback, you already provide the data to the VolumeMeter

mooballs, 2024-10-11:
Right, because I've put A2DPSink in a stream rather than using a callback, I don't need to write the eq, etc to a callback?

pschatzmann, 2024-10-11:
yes, it is part of your outout chain