pschatzmann/arduino-audio-tools #1760: https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-url-i2s

Ephesian, 2024-10-30:
### Problem Description

> Hi 

I have read the the wiki (https://github.com/pschatzmann/arduino-audio-tools/wiki/The-Audio-Player-Class) and it says there is an example for the Audio player, but when I click on the link I get a 404 error

> I would like to have an example to work through.

### Device Description

ESP32 AudioKit v2.2 A247

### Sketch

```cpp
No sketch available at page
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-10-30:
You did not write for which link you were getting: there are many examples!
The examples are logically named and stored by topic, so you should be finding your example also w/o link.

closed 2024-10-30 by pschatzmann

Ephesian, 2024-10-30:
Hello

I put the link in the title.  It is https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-url-i2s 

which gives 404 missing page.

pschatzmann, 2024-10-30:
Did you double check: all links have been corrected and should be working now