pschatzmann/arduino-audio-tools #1763: Compilation Error with XIAO esp32c3

ltsui, 2024-10-30:
### Problem Description

Tested stream_url_mp3_helix_i2s sketch, on esp32. It works well. However, I get compilation errors when same code is used on esp32c3 and s3 as well. 

### Device Description

XIAO ESP32C3

### Sketch

```cpp
/**
 * @file streams-url_mp3-i2s.ino
 * @author Phil Schatzmann
 * @brief decode MP3 stream from url and output it on I2S
 * @version 0.1
 * @date 2021-96-25
 * 
 * @copyright Copyright (c) 2021
 */

// install https://github.com/pschatzmann/arduino-libhelix.git

#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

URLStream url("ssid","password");
I2SStream i2s; // final output of decoded stream
EncodedAudioStream dec(&i2s, new MP3DecoderHelix()); // Decoding stream
StreamCopy copier(dec, url); // copy url to decoder

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);
  // you could define e.g your pins and change other settings
  //config.pin_ws = 10;
  //config.pin_bck = 11;
  //config.pin_data = 12;
  //config.mode = I2S_STD_FORMAT;
  i2s.begin(config);

  // setup I2S based on sampling rate provided by decoder
  dec.begin();

// mp3 radio
  url.begin("http://stream.srg-ssr.ch/m/rsj/mp3_128","audio/mp3");

}

void loop(){
  copier.copy();
}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

ltsui, 2024-10-30:
Thanks for your reply, Phil! 

If you are unable to reproduce the issues on your esp32c3 dev module, then it is probably something to do with the Arduino IDE. I pasted the error below and would appreciate any guidance you can give. Thanks!

```
In file included from /Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SStream.h:10,
                 from /Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools.h:81,
                 from /private/var/folders/q3/94nxpsdn7kl52mknmnn8s9x40000gn/T/.arduinoIDE-unsaved2024930-85819-veaiyz.rw0of/streams-url_mp3_helix-i2s/streams-url_mp3_helix-i2s.ino:13:
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h: In member function 'i2s_tdm_slot_config_t audio_tools::I2SDriverESP32V1::DriverTDM::getSlotConfig(audio_tools::I2SConfigESP32V1&)':
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:359:29: error: declaration of 'i2s_tdm_slot_config_t cfg' shadows a parameter
  359 |       i2s_tdm_slot_config_t cfg = I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG(
      |                             ^~~
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:353:59: note: 'audio_tools::I2SConfigESP32V1& cfg' previously declared here
  353 |     i2s_tdm_slot_config_t getSlotConfig(I2SConfigESP32V1 &cfg) {
      |                                         ~~~~~~~~~~~~~~~~~~^~~
In file included from /Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:11:
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:360:41: error: 'struct i2s_tdm_slot_config_t' has no member named 'bits_per_sample'
  360 |               (i2s_data_bit_width_t)cfg.bits_per_sample, I2S_SLOT_MODE_STEREO,
      |                                         ^~~~~~~~~~~~~~~
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:34:24: note: in definition of macro 'I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG'
   34 |     .data_bit_width = (bits_per_sample), \
      |                        ^~~~~~~~~~~~~~~
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:363:18: error: 'struct i2s_tdm_slot_config_t' has no member named 'i2s_format'
  363 |       switch(cfg.i2s_format){
      |                  ^~~~~~~~~~
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:369:41: error: 'struct i2s_tdm_slot_config_t' has no member named 'bits_per_sample'
  369 |               (i2s_data_bit_width_t)cfg.bits_per_sample, I2S_SLOT_MODE_STEREO,
      |                                         ^~~~~~~~~~~~~~~
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:34:24: note: in definition of macro 'I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG'
   34 |     .data_bit_width = (bits_per_sample), \
      |                        ^~~~~~~~~~~~~~~
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:46:1: error: no match for 'operator=' (operand types are 'i2s_tdm_slot_config_t' and '<brace-enclosed initializer list>')
   46 | }
      | ^
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:368:17: note: in expansion of macro 'I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG'
  368 |           cfg = I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:151:3: note: candidate: 'constexpr i2s_tdm_slot_config_t& i2s_tdm_slot_config_t::operator=(const i2s_tdm_slot_config_t&)'
  151 | } i2s_tdm_slot_config_t;
      |   ^~~~~~~~~~~~~~~~~~~~~
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:151:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const i2s_tdm_slot_config_t&'
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:151:3: note: candidate: 'constexpr i2s_tdm_slot_config_t& i2s_tdm_slot_config_t::operator=(i2s_tdm_slot_config_t&&)'
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:151:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'i2s_tdm_slot_config_t&&'
/Users/lt/Documents/Arduino/libraries/audio-tools/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:375:41: error: 'struct i2s_tdm_slot_config_t' has no member named 'bits_per_sample'
  375 |               (i2s_data_bit_width_t)cfg.bits_per_sample, I2S_SLOT_MODE_STEREO,
      |                                         ^~~~~~~~~~~~~~~
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:55:24: note: in definition of macro 'I2S_TDM_MSB_SLOT_DEFAULT_CONFIG'
   55 |     .data_bit_width = (bits_per_sample), \
[... cut ...]
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:151:3: note: candidate: 'constexpr i2s_tdm_slot_config_t& i2s_tdm_slot_config_t::operator=(i2s_tdm_slot_config_t&&)'
/Users/lt/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32c3/include/driver/i2s/include/driver/i2s_tdm.h:151:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'i2s_tdm_slot_config_t&&'

exit status 1

Compilation error: exit status 1
```

pschatzmann, 2024-10-30:
Ok, I just realized that the last correction has not been committed!
I just [did this](https://github.com/pschatzmann/arduino-audio-tools/commit/5873189ba0b597423360b4fe616567fd33d3d0f4)...

ltsui, 2024-10-30:
Thank you for the update! It works perfectly now! What a wonderful library and I cannot describe our appreciation to your kindness and effort!