pschatzmann/arduino-audio-tools #1768: assert failed: PureDataStream.h:109 (eff == sample_size)

dennisdebel, 2024-11-02:
### Problem Description

As soon as one includes the ADC~ (audio input) object into a pd patch, this error gets generated and my board reboots. All patches I tried, with only audio output (DAC~) work fine.

### Device Description

Ai Thinker ESP32 Audio Kit v2.2 A417 (using AudioKitEs8388V1)

### Sketch

```cpp
// adc~ only - puredata //

#define SAMPLE_RATE 44100

#include "AudioTools.h"
#include "AudioLibs/I2SCodecStream.h"
#include "Heavy_test.hpp"
#include "AudioLibs/PureDataStream.h"

AudioInfo info(SAMPLE_RATE, 4, 16);
Heavy_test pd_prog(SAMPLE_RATE);
PureDataStream pd(pd_prog);
I2SCodecStream out(AudioKitEs8388V1);
StreamCopy copier(out, pd);

void setup() {

  pd.begin();

  auto cfg = out.defaultConfig(); 
  cfg.sd_active = false;
  cfg.copyFrom(info);
  out.begin(cfg);

}

void loop() {
  
  copier.copy();
  
}
```

### What is your development environment

Arduino, Mac OSX

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

dennisdebel, 2024-11-05:
@xi-bot figured it out, see this issue [here](https://github.com/jlswbs/ESP32_PureData/issues/1#issuecomment-2456059008). They used AudioBoardStream instead of I2SCodecStream (In addition to actually dealing with the in and output stream ;) ).

closed 2024-11-05 by dennisdebel

pschatzmann, 2024-11-05:
The only difference of the AudioBoardStream compared to the I2SCodecStream is, that it also provides support for the buttons!