pschatzmann/arduino-audio-tools #1773: Issue with HttpRequest stream upload or API usage error?

djfm, 2024-11-04:
### Problem Description

Can't seem to post audio stream to server using i2s as input and AudioTool's `HttpRequest` for upload, while the other direction works fine (getting stream from server and playing it with on the ESP32 device using `URLStream`).

I have based my code on this example: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-communication/http-client/streams-http_post/streams-http_post.ino

and the above example doesn't seem to work for me either.

The server is made in node with express and is as basic as can be:

```
import express from 'express';

const app = express();

app.post('/listen', (req, res) => {
    console.log('[/listen]: Receiving audio data...');
    console.log('Headers:', req.headers);
    console.log('Body:', req.body);

    req.setTimeout(0);

    req.on('data', (chunk) => {
        console.log(`Received ${chunk.length} bytes of audio data.`);
    });

    res.on('close', () => {
        console.log('Connection closed.');
    });
});

const port = 6556;

app.listen(port, () => {
    console.log(`Server is running on port ${port}`);
});
```

The minimal arduino sketch is included in the dedicated form field, but I get the same results from the example quoted from this repository.

In both cases the Serial output of the ESP32 says:

```
[I] StreamCopy.h : 414 - try write  - 12 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 13 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 14 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 15 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 16 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 17 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 18 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 19 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 20 (open 1024 bytes) 
[I] StreamCopy.h : 414 - try write  - 21 (open 1024 bytes) 
[E] StreamCopy.h : 407 - write  to target has failed after 22 retries! (1024 bytes)
```
and the server displays:

```
Server is running on port 6556
[/listen]: Receiving audio data...
Headers: {
  host: '192.168.1.1',
  connection: 'keep-alive',
  'accept-encoding': 'identity',
  accept: '*/*',
  'content-type': 'audio/pcm'
}
Body: undefined
Connection closed.
```

While I am expecting it to emit 'data' events as it receives the stream.

For some reason that I do not understand the connection gets closed immediately.

Am I missing something obvious here? What is the server expected to reply?

### Device Description

esp32dev

### Sketch

```cpp
#include "AudioTools.h"
#include "AudioTools/AudioLibs/I2SCodecStream.h"

WiFiClient client;
HttpRequest http(client);
const AudioInfo info(44100, 1, 16);
Url target;
I2SCodecStream audioStream(AudioKitEs8388V2);
StreamCopy copier(client, audioStream);

void setup() {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Debug);

    WiFi.mode(WIFI_STA);
    WiFi.begin("***", "***");
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print('.');
        delay(1000);
    }
    Serial.println();
    Serial.println(WiFi.localIP());

    auto audioConfig = audioStream.defaultConfig(RX_MODE);
    audioConfig.copyFrom(info);
    audioStream.begin(audioConfig);

    // this doesn't seem to change anything
    // http.addRequestHeader("Transfer-Encoding", "chunked");

    Url url("http://192.168.1.1:6556/listen");
    if (!http.processBegin(POST, url, "audio/pcm")){
        Serial.println("post failed");
        stop();
    }
}

void loop() {
    if (copier.copy() == 0) {
        http.processEnd();
    }
}
```
```

### Other Steps to Reproduce

Also tried with polka instead of express as a server, no change.

Also tried to add "Transfer-Encoding: chunked" to the request headers, no change.

### What is your development environment

Development environment: PlatformIO under Ubuntu.
Platform: espressif32
Framework: arduino

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-11-04:
I think your server expects chunked data but you don't provide it in this format!

closed 2024-11-04 by pschatzmann

djfm, 2024-11-04:
As far as I know that's how node handles requests which do not have a
content-length header specified.

Do you have an example of server you're using for your tests? I couldn't
find one.

ffaerber, 2024-11-20:
@djfm i am working on the same problem.
i have a working client(nodejs) to server(nodejs) version working with streaming
now i try to do a client(arduino) to server(nodejs) version.

```
// server.js
const express = require('express');
const fs = require('fs');
const path = require('path');

const app = express();
const port = 3000;

app.post('/stream-audio', (req, res) => {
  console.log('Receiving audio stream...');
  console.log('Headers:', req.headers);
  console.log('Body:', req.body);
  const filePath = path.join(__dirname, 'uploads', 'audio.wav');
  const fileStream = fs.createWriteStream(filePath);

  req.pipe(fileStream);

  req.on('end', () => {
    console.log('Audio stream received and saved.', filePath);
    res.status(200).send('Stream received successfully');
  });

  req.on('error', (err) => {
    console.error('Error receiving stream:', err);
    res.status(500).send('Error processing stream');
  });
});

app.listen(port, () => {
  console.log(`Server listening at http://localhost:${port}`);
});
```

```
//client.js
const recorder = require('node-record-lpcm16');  // For recording audio
const request = require('request');  
const fs = require('fs');                        // For file system access
const path = require('path');                    // For path handling

const apiUrl = `http://localhost:3000/stream-audio`;  // Your API endpoint for receiving audio data

const recording = recorder.record({
  recorder: 'arecord'
});

function parseResult (err, resp, body) {
  if (err) console.error(err)
  console.log(body)
}

recording.stream().pipe(request.post({
    url: apiUrl,
    headers: {
      'Accept': 'application/json',
      'Content-Type': 'audio/wav'
    },
  }, parseResult))

 
setTimeout(() => {
    recording.stop()
}, 3000) // Stop after three seconds of recording
```

pschatzmann, 2024-11-20:
But isn't this just a simple chuned post ?