pschatzmann/arduino-audio-tools #1785: a2dp resets my DAC settings after pair

josef2600, 2024-11-12:
### Problem Description

since i want to convert 16bit to 24bit, i decided to do it in the code. very simple. the` void writeBytes2` doesn't matter. and it doesn't matter if i place `a2dp_sink` before or after the kit config. it is all the same. 
so i config the kit to be 24 bit `AudioInfo infoo(44100, 2, the_out_bitratee);` . it does it. but when the Bluetooth gets connected, everything gets reset to default. why? it doesn't happens if the bit rate is 16 bit. what in has to do with out? i did search inside the code, but couldn't get it. i know it is in the setting in the ESP32-A2DP , it checks the bit rate many times in many files to be 16 bit. 
but it is wrong. it simply wrong. it is the input source, why should it care about the output? this is a big error. 
no, convert doesn't work with 24 bit either, no matter what board version it is. for test, please play a music, not sine generator, it is useless. and i am not talking about convert. i am saying why input gets the setting from output? i need the i2s to be at whatever bit rate i need, and the input at whatever else. they should not be the same.
this is the output info:
```
Warning: Pin '23' not set up because of conflict
Warning: Pin '18' not set up because of conflict
[I] AudioTypes.h : 128 - in: sample_rate: 44100 / channels: 2 / bits_per_sample: 24
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 24
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 24
[I] I2SConfigESP32.h : 80 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Master
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 24
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: true
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 101 - pin_mck: 0
[I] I2SConfigESP32.h : 103 - pin_bck: 5
[I] I2SConfigESP32.h : 105 - pin_ws: 25
[I] I2SConfigESP32.h : 107 - pin_data: 26

```
then when paired, it resets the setting:
```
[I] I2SCodecStream.h : 95 - virtual void audio_tools::I2SCodecStream::setAudioInfo(audio_tools::AudioInfo)
[I] AudioTypes.h : 128 - in: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] I2SStream.h : 93 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioInfo)
[I] AudioTypes.h : 128 - in: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 - out: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] I2SStream.h : 102 - restarting i2s
[I] AudioTypes.h : 128 - I2SStream sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] AudioTypes.h : 128 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] I2SConfigESP32.h : 80 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Master
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 16
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: true
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 101 - pin_mck: 0
[I] I2SConfigESP32.h : 103 - pin_bck: 5
[I] I2SConfigESP32.h : 105 - pin_ws: 25
[I] I2SConfigESP32.h : 107 - pin_data: 26
```

### Device Description

AudioKitEs8388V2

### Sketch

```cpp
this is the code:

#define AUDIOKIT_BOARD  7

#include "AudioTools.h"
#include "AudioTools/AudioLibs/A2DPStream.h" // install https://github.com/pschatzmann/ESP32-A2DP
#include "AudioTools/AudioLibs/AudioBoardStream.h" // install https://github.com/pschatzmann/arduino-audio-driver
#include "BluetoothA2DPSink.h"

#define the_out_bitratee  24

AudioInfo infoo(44100, 2, the_out_bitratee);
AudioBoardStream kit(AudioKitEs8388V2);
BluetoothA2DPSink a2dp_sink(kit);

// Write data to AudioKit in callback
void writeBytes2(const uint8_t *dataaa, uint32_t size_bytes)
{
  const uint8_t* src_bytes = (const uint8_t*)dataaa;
  size_t num_samples = (size_bytes / 2); // Number of 16-bit samples
  // Allocate memory for 24-bit data (3 bytes per sample)
  //size_t new_size_bytes = num_samples * 3;// or *(the_out_bitratee/8)
  size_t new_size_bytes = num_samples * (the_out_bitratee/8);
  uint8_t* new_data = (uint8_t*)malloc(new_size_bytes);
  
  if (new_data == nullptr) return;
  
  for (size_t i = 0; i < num_samples; i++)
  {
    new_data[3 * i] = src_bytes[2 * i];  // High byte
    new_data[3 * i + 1] = src_bytes[(2 * i) + 1];   // Middle byte
    new_data[3 * i + 2] = 0;                     // Low byte
  }
  kit.write(new_data, new_size_bytes);
  if (new_data) free(new_data);
}

void setup()
{
  Serial.begin(115200);
  delay(500);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  //Info  Error  Warning
  delay(10);
  AudioDriverLogger.begin(Serial,AudioDriverLogLevel::Info);  //Info  Error  Warning
  delay(100);
  
  a2dp_sink.set_stream_reader(writeBytes2, false);
  delay(100);
  a2dp_sink.start("AudioKit");
  delay(100);
  auto cfg = kit.defaultConfig();
  cfg.copyFrom(infoo);
  cfg.sd_active = false;
  
  kit.begin(cfg);
  delay(100);
}

void loop()
{
  delay(500);
}
```
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-11-12:
The application does exactly what you tell it to do: you give in the constructor the information that is should output to kit, so it replicates the audio information provided by A2DP and this is 16 bits!

I don't understand why you are doing this, since you do the output via callback: just remove it in the constructor!

By the way, your 24 bit logic is wrong: it is jupposed to be 32 bit left aligned with the rightmost 8 bit beeing ignored.

closed 2024-11-12 by pschatzmann

josef2600, 2024-11-12:
sure, nobody knows anything and you are the only one who is right. you don't even understand the logic of how i2s works.

pschatzmann, 2024-11-12:
If you are questioning my 24 bits statement, I suggest to read the [Epressif i2s documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2s.html):

pschatzmann, 2024-11-12:
This is my understanding as well and that's the reason why I have implemented the AudioBoardStream which keeps the CODEC and the I2S definition consistent.

I guess the best would be to verify what's going on with an oszilloscope to make sure that a 24 bit setting is really showing up as expected if you feed it the documented 32 bit data. You best do this with a sine wave that let's you double check the data signal easily as well.

Taking into consideration that it depends on the ESP32 Core version makes me think that this is a problem in the Core and not my code.

pschatzmann, 2024-11-12:
ps. the writes in my API are blocking by default because this is the way most other I2S APIs on other platforms behave. However you can configure it to be non blocking.

josef2600, 2024-11-12:
you know what? i think you are correct. i did change all the codes of `arduino-audio-tools` and `ESP32-A2DP` . and for both versions of esp32 core. the old one is esp32 and new one is esp32V1. and i was directly sending the data and command to i2s driver of esp. but non of them worked. that pissed me off. something is fishy here! 
now the reason why i am doing this: 
first, your library becomes the best audio library.
second, i need it to become the best audio library for myself so i can make an audiophile device. or few of them.
i understand and read all of your codes related to a2dp and i2s, but the only part i didn't get was convert! i don't get how it works yet. although it kind of works with 16 to 32 bit, but the audio is kind of weird. it is not the quality it has to have. it gets jittery. i spend over 6 grand in the past 7 month only on my amps. they are excellent. but now my problem is the audio. 
since i don't want to use a 200$ DAC on default on the products, but they need a basic audio, i choose the esp32. 
if and when i finished my prototype, i will send you one so you can enjoy it. 
i guess maybe i am going the wrong way. the only reason i am using es8388 is the microphone input. otherwise there are excellent DAC's that are available that are not expensive. although, the highest end DACs do not have audio inputs either. 
probably i am going backward. do you think i should forget microphone? or not.

josef2600, 2024-11-12:
> ps. the writes in my API are blocking by default because this is the way most other I2S APIs on other platforms behave. However you can configure it to be non blocking.

the blocking way is bad. the reason being that it stops the flow of CPU works for doing other stuff. and it doesn't need it. the data goes to memory and the i2s reads it from there directly. if you try to do a blocking way, you are just wasting time on a waiting. that is why you have the `ticks_to_wait_write`

josef2600, 2024-11-12:
and can you please check this driver? i really want to know what do you think.
[es8388.zip](https://github.com/user-attachments/files/17717093/es8388.zip)

pschatzmann, 2024-11-12:
The ESP32 has FreeRTOS, so waiting is not really that bad: in fact calling wait() is just the indication for the other tasks that they can do their work.

Putting the Audio into a separate task makes the overall application also easier to maintain, if it getting more complicated then my usual demo examples.

Tomorrow, I will have an eye operation, so It might take quite some time to recover and give some feed-back.

josef2600, 2024-11-12:
hope you get well. i am sorry again if i got you upset. 
but i think this is going to give me some time too! shuffle more codes before you start changing all of it.