pschatzmann/arduino-audio-tools #1791: PWM MP3 Player on RP2040 got slow noise audio

cuu, 2024-11-14:
### Problem Description

I try to play mp3 files from SD card, on rp2040 ,output through PWM ,not I2S

the audio from headphone jack has a lot of noise
and I can tell the music is slowed down, like a player playing at 0.6x.

I also tried enlarge the buffer size and the CPU speed from 133 to 250Mhz

no any luck

and the reason I use MAD instead of Helix is Helix is just totally no sound at all

### Device Description

RP2040

### Sketch

```cpp
#define SD_FAT_TYPE 3

#define PWM_BUFFER_SIZE 4096
#define DEFAULT_BUFFER_SIZE 4096
#define PWM_AUDIO_FREQUENCY 44100

#define MAD_MAX_BUFFER_SIZE 4096
#define MAD_MAX_RESULT_BUFFER_SIZE 4096

#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3MAD.h"

const char *startFilePath="/";
const char* ext="mp3";
AudioSourceSDFAT source(startFilePath, ext,17,60);
PWMAudioOutput i2s;
MP3DecoderMAD decoder;
AudioPlayer player(source, i2s, decoder);

void printMetaData(MetaDataType type, const char* str, int len){
  Serial1.print("==> ");
  Serial1.print(toStr(type));
  Serial1.print(": ");
  Serial1.println(str);
}

void setup() {
  Serial1.begin(115200);
  AudioToolsLogger.begin(Serial1, AudioToolsLogLevel::Info);

  auto cfg = i2s.defaultConfig(TX_MODE);
  Pins pins = {26,27};
  //config.channels = 2;
  cfg.setPins(pins);

  i2s.begin(cfg);

  player.setMetadataCallback(printMetaData);
  player.setBufferSize(4096);
  player.begin();
}

void loop() {
  player.copy();
}
```

### What is your development environment

Arduino

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

cuu, 2024-11-15:
well.
really thanks for the reply

even If I comment out the metadata and Log 

No better

And what you mean resample? setup a AudioInfo with different sample rate , bits in sample ?
like 
AudioInfo(44100,2,16); ?

pschatzmann, 2024-11-15:
I suggest to google the topic about resampling in audio and/or read the corresponding Wiki