pschatzmann/arduino-audio-tools #1814: Sample "streams_i2s_webserver_wav.ino" compilation error: use assignment or value-initialization instead

honza540, 2024-11-28:
### Problem Description

Simply trying to compile the provided library sample "streams_i2s_webserver_wav.ino"

It fails with this error:
In file included from c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/AudioCodecs/AudioCodecs.h:22,
                 from c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools.h:93,
                 from C:\_projekty\_ESP32_Arduino\_projects\streams_i2s_webserver_wav\streams_i2s_webserver_wav.ino:10:
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/AudioCodecs/CodecWAV.h: In member function 'void audio_tools::WAVHeader::clear()':
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/AudioCodecs/CodecWAV.h:120:46: error: 'void* memset(void*, int, size_t)' clearing an object of type 'struct audio_tools::WAVAudioInfo' with no trivial copy-assignment; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&headerInfo,0,sizeof(WAVAudioInfo));
                                              ^
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/AudioCodecs/CodecWAV.h:18:8: note: 'struct audio_tools::WAVAudioInfo' declared here
 struct WAVAudioInfo : AudioInfo {
        ^~~~~~~~~~~~
In file included from c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/CoreAudio/AudioEffects/Synthesizer.h:6,
                 from c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools.h:110,
                 from C:\_projekty\_ESP32_Arduino\_projects\streams_i2s_webserver_wav\streams_i2s_webserver_wav.ino:10:
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/CoreAudio/AudioActions.h: In member function 'void audio_tools::AudioActions::clear()':
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/CoreAudio/AudioActions.h:230:17: error: deleting object of polymorphic class type 'audio_tools::AudioActions::Action' which has non-virtual destructor might cause undefined behavior [-Werror=delete-non-virtual-dtor]
       delete(act);
                 ^
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/CoreAudio/AudioActions.h: In member function 'void audio_tools::AudioActions::insertAction(audio_tools::AudioActions::Action&)':
c:\_projekty\_ESP32_Arduino\libraries\audio-tools\src/AudioTools/CoreAudio/AudioActions.h:246:28: error: deleting object of polymorphic class type 'audio_tools::AudioActions::Action' which has non-virtual destructor might cause undefined behavior [-Werror=delete-non-virtual-dtor]
         delete(actions[idx]);
                            ^
cc1plus.exe: some warnings being treated as errors

Using library audio-tools at version 0.9.9 in folder: C:\_projekty\_ESP32_Arduino\libraries\audio-tools 
Using library WiFi at version 2.0.0 in folder: C:\Users\j.sandera_gmc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\j.sandera_gmc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\WiFiClientSecure 
exit status 1

Compilation error: exit status 1

### Device Description

ESP32-WROOM-DA

### Sketch

```cpp
/**
 * @file streams-i2s-webserver_wav.ino
 *
 *  This sketch reads sound data from I2S. The result is provided as WAV stream which can be listened to in a Web Browser
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("ssid","password"); // the same a above

I2SStream i2sStream;    // Access I2S as stream
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

void setup(){
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 22050;
  config.channels = 2;
  config.bits_per_sample = 16;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.copy();  
}
```

### What is your development environment

Arduino IDE 2.3.3
Board manager: esp32 by Espressif system 2.0.14

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-11-28:
Please note that 2.0.14 is not supported any more. The actual version is 3.0.7!

I have the impression that you changed the compiler warnings and you decided to treat mere warnings as errors!
I could not reproduce your issue with an actual version, so I closing this issue.

Anyhow, I committed some corrections to prevent some of these warnings...

honza540, 2024-11-28:
Perfect, you are absolutely right, working fine with board manager 3.0.7.

Thank you very much Phil.
Best regards, Jan