pschatzmann/arduino-audio-tools #1819: Problems with PDM and the MP34DT05TR-A microphone There is no clock signal

jhohannfabian, 2024-12-02:
### Problem Description

When I enable the option cfg.signal_type = PDM; The clock signal does not work, but if I do not include it, the clock starts and measures noise or a non-coherent signal.

### Device Description

ESP32 S3 Wroom N16R8

### Sketch

```cpp
#define MIC_BCK 48
#define MIC_DATA 45
#define MIC_LR 47

#include "AudioTools.h"
AudioInfo info(44100, 1, 16);
I2SStream i2sStream; // Access I2S as stream
CsvOutput<int16_t> csvStream(Serial);
StreamCopy copier(csvStream, i2sStream); // copy i2sStream to csvStream
// Arduino Setup
void setup(void) {
    Serial.begin(115200);
    delay(1000);
    //pinMode(MIC_LR, OUTPUT);
   //  digitalWrite(MIC_LR, LOW); // Set to left channel
    AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);
    auto cfg = i2sStream.defaultConfig(RX_MODE);
    
    cfg.copyFrom(info);
    cfg.signal_type = PDM;
    //cfg.use_apll = false;  
    //cfg.auto_clear = false;
    cfg.pin_data = MIC_DATA;
    cfg.pin_ws = -1;//MIC_LR;
    cfg.pin_bck = MIC_BCK;

    i2sStream.begin(cfg);
    
    // make sure that we have the correct channels set up
    csvStream.begin(cfg);

}

// Arduino loop - copy data
void loop() {
    copier.copy();
    //copier.copyAll();
    delay(10);
}
```

### Other Steps to Reproduce

When I use cfg.signal_type = PDM; Only shows: 
 -30934
-30934
-30934
-30934
-30934
-30934
-30934
-30934
-30934

### What is your development environment

Arduino

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-12-03:
Double check the ESP32 core version: the actual version is 3.0.7 with 3.1 coming out soon.

jhohannfabian, 2024-12-03:
Ohh OK, I'm using 2.0. Thank you.

pschatzmann, 2024-12-03:
With 2.x the PDM clock signal is mapped to the ws pin

jhohannfabian, 2024-12-03:
Thanks for the tip!

Yes, that notices, on the oscilloscope, that when I assigned a pin to
ws, it oscillated. something like that I imagined. I'm going to review
it further. Thank you very much for such an excellent library. I'm
going to buy you a coffee. Thank you very much for this magnificent
work.

closed 2024-12-04 by pschatzmann