pschatzmann/arduino-audio-tools #1821: The audiokit example streams-audiokit-sd_wav does not work anymore

Heute13, 2024-12-04:
### Problem Description

Compilation error: 'AudioToolsLogger' was not declared in this scope

I want to read a wav file from a sd card and play it over the Earphones 3.5mm output

### Device Description

Ai Thinker ESP32 Audiokit v2.2 A417

### Sketch

```cpp
/**
 * @file streams-audiokit-sd_wav.ino
 * @author Phil Schatzmann
 * @brief Demonstrates how to use an encoder to write to a file
 * @version 0.1
 * @date 2023-05-04
 *
 * @copyright Copyright (c) 2023
 *
 */

#include <SD.h>
#include <SPI.h>

#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"

const char* file_name = "/song_1.wav";
AudioInfo info(16000, 1, 16);
AudioBoardStream in(AudioKitEs8388V1);
File file;  // final output stream
EncodedAudioStream out(&file, new WAVEncoder());
StreamCopy copier(out, in);  // copies data
uint64_t timeout;

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info); //<--Problem

  // setup input
  auto cfg = in.defaultConfig(RX_MODE);
  cfg.sd_active = true;
  cfg.copyFrom(info);
  cfg.input_device = ADC_INPUT_LINE2; // microphone
  in.begin(cfg);

  // Open SD drive
  if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {
    Serial.println("Initialization failed!");
    stop();
  }
  
  // Cleanup if necessary
  if (SD.exists(file_name))
    SD.remove(file_name);

  // open file
  file = SD.open(file_name, FILE_WRITE);
  if (!file){
    Serial.println("file failed!");
    stop();
  }

  // setup output
  out.begin(info);

  // set timeout - recoring for 5 seconds
  timeout = millis() + 5000;
}

void loop() {
  if (millis() < timeout) {
    // record to wav file
    copier.copy();
  } else {
    // close file when done
    if (file) {
      file.flush();
      Serial.print("File has ");
      Serial.print(file.size());
      Serial.println(" bytes");
      file.close();
    }
  }
}
```

### Other Steps to Reproduce

I also tried the example streams-sd_mp3-audiokit and the same issue accured.  So I tried an older version of the example code streams-sd_mp3-audiokit i dowloaded last Month and it works properly. I looked what changed and it was the AudioToolsLogger.

### What is your development environment

Arduino IDE

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

Heute13, 2024-12-04:
How can i find out which library to use? I am sorry I am a beginner

pschatzmann, 2024-12-04:
How did you install this library ?

Heute13, 2024-12-04:
i found them on GitHub downloaded the Zip file and added the zip library in arduino IDE

Heute13, 2024-12-04:
It works!!! Thank you so much

But I only get noise out of the earphones plugged into the EARPHONES output. Can you think of any mistake I made, that causes this?

pschatzmann, 2024-12-04:
Try to use the sine output test to figure out what driver you need to select

Heute13, 2024-12-04:
where do i find this example?

Heute13, 2024-12-04:
I got it working! Thank you, appreciate your help it very much