pschatzmann/arduino-audio-tools #1826: streams-i2s-serial.ino sketch won’t compile

biomurph, 2024-12-06:
### Problem Description

I have been using the v1.0.0 for a while, and just switched over to the v1.0.1.
I’m working on a project that streams I2C from a MEMs microphone, and so I wanted to use the streams-i2s-serial.ino example as a basic sanity check. However, the sketch won’t compile. I have pasted the errors below. Seems to be happening in the PDM section.

I’m using an ESP32C6, so I went into your AurioConfig.h file and commented out line 249. Now it compiles and the example works with my microphone.

In file included from /Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SStream.h:10,
                 from /Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools.h:81,
                 from /private/var/folders/d1/p8jfftcx5pn5dgh_l6zk614c0000gn/T/.arduinoIDE-unsaved2024116-71545-1iub3cn.8pb6/streams-i2s-serial/streams-i2s-serial.ino:11:
/Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:322:5: error: 'i2s_pdm_rx_slot_config_t' does not name a type; did you mean 'i2s_pdm_tx_slot_config_t'?
  322 |     i2s_pdm_rx_slot_config_t getRxSlotConfig(I2SConfigESP32V1 &cfg) {
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
      |     i2s_pdm_tx_slot_config_t
/Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:338:5: error: 'i2s_pdm_rx_clk_config_t' does not name a type; did you mean 'i2s_pdm_tx_clk_config_t'?
  338 |     i2s_pdm_rx_clk_config_t getRxClockConfig(I2SConfigESP32V1 &cfg) {
      |     ^~~~~~~~~~~~~~~~~~~~~~~
      |     i2s_pdm_tx_clk_config_t
/Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h: In member function 'bool audio_tools::I2SDriverESP32V1::DriverPDM::startRX(audio_tools::I2SConfigESP32V1&, i2s_channel_obj_t*&, int)':
/Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:369:7: error: 'i2s_pdm_rx_config_t' was not declared in this scope; did you mean 'i2s_pdm_tx_config_t'?
  369 |       i2s_pdm_rx_config_t pdm_rx_cfg = {
      |       ^~~~~~~~~~~~~~~~~~~
      |       i2s_pdm_tx_config_t
/Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:383:50: error: 'pdm_rx_cfg' was not declared in this scope
  383 |       if (i2s_channel_init_pdm_rx_mode(rx_chan, &pdm_rx_cfg) != ESP_OK) {
      |                                                  ^~~~~~~~~~
/Users/biomurph/Dropbox/Arduino/libraries/arduino-audio-tools-1.0.1/src/AudioTools/CoreAudio/AudioI2S/I2SESP32V1.h:383:11: error: 'i2s_channel_init_pdm_rx_mode' was not declared in this scope; did you mean 'i2s_channel_init_pdm_tx_mode'?
  383 |       if (i2s_channel_init_pdm_rx_mode(rx_chan, &pdm_rx_cfg) != ESP_OK) {
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |           i2s_channel_init_pdm_tx_mode

### Device Description

XIAO ESP32C6

### Sketch

```cpp
Using your example
```

### What is your development environment

Arduino v2.3.3
Arduino Audio Tools v1.0.1
Mac M2 

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-12-06:
I can't reproduce your issue: make sure that you install the actual main branch!

closed 2024-12-06 by pschatzmann

biomurph, 2024-12-06:
Oh, I downloaded the zip release v1.0.1
I will get the get the main branch and try it out.