pschatzmann/arduino-audio-tools #1834: usando: MemoryStream  para reproducir un archivo mp3 desde la PS_RAM

jhohannfabian, 2024-12-09:
### Problem Description

I was using Arduino ESP32 S3 core 2.0, and migrated to 3.X. I was loading an MP3 file from FATFs to PS_RAM, and then playing it. Now when I update, and update audiotools as well, the playback from PS_RAM no longer works. If I use the example that plays from Program Memory, it works fine, but if I load the file from PS_RAM, it doesn't play.

### Device Description

ESP32 S3. Arduino 3.X

### Sketch

```cpp
#include "FS.h"
#include "FFat.h"
#define SPIFFS FFat
File archivo;

#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"
#include "zero.h"

uint8_t* audioData = nullptr;
size_t audioDataLength = 0;

MemoryStream mp3(zero_mp3, zero_mp3_len);
I2SStream i2s;  // PWM output 
MP3DecoderHelix helix;
EncodedAudioStream out(&i2s, &helix); // output to decoder
StreamCopy copier(out, mp3);    // copy in to out
#define Pinbck 17//5
#define Pinws 16//6
#define Pindata 15//4 
#define PinMute 11

void sayWord() {
    Serial.println("Saying word...");
    mp3.begin(); // restart source 
    helix.begin(); // so that we can repeatedly call this method
    copier.copyAll();
    helix.end(); // flush output
}
void setup(){
  Serial.begin(115200);
  delay(300);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);  
  pinMode(PinMute, OUTPUT);
  digitalWrite(PinMute, LOW);
  Iniciar_Variables_Y_Archivos();
  Load_Audio_Desde_FFAT();
  digitalWrite(PinMute, HIGH);
  // begin processing
  auto cfg = i2s.defaultConfig();
  cfg.sample_rate = 24000;
  cfg.channels = 2;
  cfg.port_no = 1;
  cfg.pin_bck = Pinbck;
  cfg.pin_ws = Pinws;
  cfg.pin_data = Pindata;
  cfg.i2s_format = I2S_LSB_FORMAT;
  
  i2s.begin(cfg);
  out.begin();
  sayWord(); //OK
  Load_Audio_Desde_FFAT(Mapa_Sonidos.TonosCampana[9]);//LOAD audioData with mp3 data from FFAT, and size
  mp3.setValue(audioData, audioDataLength, PS_RAM);
  sayWord(); // NOT OK. ONLY SILENCE
  Serial.println("Fin");
  digitalWrite(PinMute, LOW);
}

void loop(){
}
```

### What is your development environment

Arduino ESP32 S3 core 3.X and last version audiotools.

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-12-09:
Did you create some test sketches to test if the psram is working properly with your board outside of my library ?
What error messages do you get ?

jhohannfabian, 2024-12-09:
Yes. I test PS_RAM, Its OK.
void Load_Audio_Desde_FFAT(String archivo)
{
    archivo = "/" + archivo;
    // Inicializar PSRAM
    if (!psramFound()) {
        Serial.println("PSRAM no encontrada");
        return;
    }
    File MP3file = SPIFFS.open(archivo);//, "r"); //Con la r, fallo! no se porque?
    if (!MP3file) {
        Serial.println("Error al abrir el archivo MP3");
        return;
    }
    audioDataLength = MP3file.size();
    Serial.printf("Tamanno del archivo MP3: %d bytes\n", audioDataLength);
    audioData = (uint8_t*)ps_malloc(audioDataLength);
    if (!audioData) {
        Serial.println("Error al reservar espacio en PSRAM");
        MP3file.close();
        return;
    }

   //I testing Here.
    MP3file.read(audioData, audioDataLength);
  for (int i = 0;i < audioDataLength;i++)
      Serial.println(audioData[i]);
   //Printing OK
    MP3file.close();
    Serial.print(audioDataLength);
    Serial.println(" Datos del archivo MP3 cargados en PSRAM");
}

jhohannfabian, 2024-12-09:
I checked again. Before I had created a constructor, for the class like this:
MemoryStream mp3;
Because I don't need parameters from the program memory.
But I use:
MemoryStream mp3(zero_mp3, zero_mp3_len);
According to the initial configuration of the class, it does play, but in the following way:
mp3.setValue(audioData, audioDataLength, PS_RAM);
 while (mp3.available() ) //SOUND OK
  {
      copier.copy();
  }
But I used
void sayWord() {
    Serial.println("Saying word...");
    mp3.begin(); // restart source 
    helix.begin(); // so that we can repeatedly call this method
    copier.copyAll();
    helix.end(); // flush output
}
Not Sound!

I put this construtor:
class MemoryStream : public AudioStream {
 public:
  /// Constructor for alloction in RAM
  MemoryStream() { //I put here
      info.clear(); // mark audio info as unknown
     }....
}
But, dont work.

jhohannfabian, 2024-12-09:
Its work again.
MemoryStream mp3(audioData, audioDataLength,PS_RAM);
The solution was easy, it was just to declare it, regardless of whether the data is loaded or not. Thank you very much.

pschatzmann, 2024-12-10:
I could not reproduce your isse (on my AudioKit). Here is the test sketch:
```C++
#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"
#include "SD.h"

AudioBoardStream i2s(AudioKitEs8388V1);
MemoryStream mp3;
MP3DecoderHelix helix;
EncodedAudioStream out(&i2s, &helix);  // output to decoder
StreamCopy copier(out, mp3);           // copy in to out
uint8_t* audioData = nullptr;
size_t audioDataLength = 0;

bool loadAudio(const char* archivo) {
  // Inicializar PSRAM
  if (!psramFound()) {
    Serial.println("PSRAM no encontrada");
    return false;
  }

  File MP3file = SD.open(archivo);  //, "r"); //Con la r, fallo! no se porque?
  if (!MP3file) {
    Serial.println("Error al abrir el archivo MP3");
    return false;
  }
  audioDataLength = MP3file.size();
  Serial.printf("Tamanno del archivo MP3: %d bytes\n", audioDataLength);
  audioData = (uint8_t*)ps_malloc(audioDataLength);
  if (!audioData) {
    Serial.println("Error al reservar espacio en PSRAM");
    MP3file.close();
    return false;
  }
  if (MP3file.read(audioData, audioDataLength) != audioDataLength) {
    Serial.println("Error: did not read complete file");
    return false;
  }

  return true;
}

void sayWord(const char* file) {
  if (!loadAudio(file)) {
    return;
  }
  mp3.setValue(audioData, audioDataLength);

  Serial.println("Saying word...");
  mp3.begin();    // restart source
  helix.begin();  // so that we can repeatedly call this method
  copier.copyAll();
  helix.end();  // flush output
}

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);

  auto cfg = i2s.defaultConfig();
  if (!i2s.begin(cfg)){
    Serial.println("i2s initialization failed!");
    stop();
  }

  if (!SD.begin(13)) {
    Serial.println("SD card initialization failed!");
    stop();
  }

  sayWord("/tts/one.mp3");
  Serial.println("End");
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

closed 2024-12-10 by pschatzmann

jhohannfabian, 2024-12-10:
It already plays, loading all the audio from PSRAM, but you don't want
to control the volume, it plays at the maximum level.

uint8_t* audioData = nullptr;
size_t audioDataLength = 0;

MemoryStream mp3(audioData, audioDataLength,PS_RAM);

I2SStream i2s;
MP3DecoderHelix helix;
VolumeStream Vol(i2s);
EncodedAudioStream out(&Vol, &helix); // output to decoder
StreamCopy copier(out, mp3);    // copy in to out
....

  digitalWrite(PinMute, HIGH);
  // begin processing
  auto cfg = i2s.defaultConfig(TX_MODE);
  cfg.sample_rate = 44100;
  cfg.bits_per_sample = 16;
  cfg.port_no = 1;
  cfg.pin_bck = Pinbck;
  cfg.pin_ws = Pinws;
  cfg.pin_data = Pindata;
  cfg.i2s_format = I2S_LSB_FORMAT;// I2S_LSB_FORMAT;
  cfg.channels = 2;
  i2s.begin(cfg);
  Vol.begin(cfg); // we need to provide the bits_per_sample and channels
  Vol.setVolume(0.5);
  out.begin();
  mp3.setValue(audioData, audioDataLength, PS_RAM);
  //mp3.begin(); // restart source
  helix.begin(); // so that we can repeatedly call this method
  while (mp3.available() )
  {
       copier.copy();
  }
  helix.end();
  Serial.println("Fin");
  free(audioData);
If I used mp3.begin(), Then No sound, But if I not used, Sound at Maximum power.

pschatzmann, 2024-12-10:
You forgot to call begin on the VolumeStream

jhohannfabian, 2024-12-11:
> this:  Vol.begin(cfg); // we need to
> I put it, or its another.
>
> El mar, 10 dic 2024 a la(s) 9:24 a.m., Phil Schatzmann (

I used this: cfg.i2s_format = I2S_MSB_FORMAT; and Work again. Thank you