pschatzmann/arduino-audio-tools #1841: WiFi Check, if PSW or SSID check

TvojMuc, 2024-12-20:
### Problem Description

Simple question. Is there an option to check if my credentials are ok, for login to WiFi? If I run ICYStream, the function just traying to connect for ever. Did I miss something?
...I really like this library and is all what I looking for...WiFi and Bloetooth on the same device ESP32-Wrover-IE.
BR Roman

### Device Description

ESP32-Wrover-IE

### Sketch

```cpp
#include "WiFi.h"
#include "Preferences.h"
#include "nvs_flash.h"
#include "NTPClient.h"
#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"
#include "BluetoothA2DPSink.h"
#include "Adafruit_GFX.h"
#include "Adafruit_GC9A01A.h"

Preferences pref;
#define MAX_CHARS_PER_LINE 18  // Maximum characters per line (with text size 2)
#define MAX_CHARS_PER_POST 13  // Maximum characters per line (with text size 2)
boolean CheckWiFi=false;

const char *urls[] = {
  "http://live.radio.si/Radio1",
  "http://live2.infonetmedia.si:8000/Radio2",
  "http://mp3.rtvslo.si/val202"
  };

const char *wifi = "TvojMucW";
const char *password = "aeiou123456";
String esid;
String epass = "";
esp_a2d_connection_state_t state;
esp_a2d_connection_state_t last_state;
//URLStream urlStream(wifi, password);
ICYStream urlStream(wifi, password);
AudioSourceURL source(urlStream, urls, "audio/mp3");
I2SStream i2s;
MP3DecoderHelix decoder;
AudioPlayer player(source, i2s, decoder);
BluetoothA2DPSink a2dp_sink(i2s);
```

### What is your development environment

Arduino, Java, html

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-12-20:
https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working

This has nothing to do with my project: try to make the wifi work first: Google is your friend!

closed 2024-12-20 by pschatzmann

TvojMuc, 2024-12-20:
Thank you for fast response
my Wifi and connection is working and application is also working without issue. 
But if I move this ESP to new location I need to enter new SSID and psw. I have code for that, but is triggered if I get WiFi.status() != 3. How to get response in your library that WiFi.status() is not CONNECTED. 
Is this possible?

pschatzmann, 2024-12-20:
Just google for arduino esp32 wifi event handling!

TvojMuc, 2024-12-25:
Thank you 
I did and also managed to get working like I wish. :)
I have one more question.
I have 2 links for streaming live radio from net. One is mp3 and one is aac. 
What would be the best way to detect type and how to switch decoder by selecting station? Or do you have already something in example folder?

BR Roman

pschatzmann, 2024-12-26:
I think the easiest is just to leverage from your knowledge and that you store which link is using what format.

Another way would be to ask urlStream for the content-type of the response.

You could also try to use the DecoderHelix which supports both mp3 and aac. It uses an automatic format recognition which should work for files, but I doubt that this will also work for streamed data.