pschatzmann/arduino-audio-tools #1843: PWMAudioSTM32.h Error

BODYPRINT, 2024-12-23:
### Problem Description

Sketch won't compile because 
"libraries\audio-tools\src/AudioTools/CoreAudio/AudioPWM/PWMAudioSTM32.h:120:24: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?"

### Device Description

STM32F303CC6T

### Sketch

```cpp
I dopn't think it's directly related to my sketch. Seems to be an internal library error.
```

### Other Steps to Reproduce

Try compile the example sketch streams-memory_mp3-PWM

### What is your development environment

Arduino IDE 2.3.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-12-23:
The STM32F303CC6T is not supported: Either extend the functionality yourself or deactivate the PWM support in the AudioConfig.h

closed 2024-12-23 by pschatzmann

BODYPRINT, 2024-12-23:
I commented out all the Timer 5 (TIM5) options in PWMAudioSTM32.h and AudioTimerSTM32.h and it works now. Cheers.