pschatzmann/arduino-audio-tools #1845: error build for esp-idf vscode extension. (esp-idf 5.3.2)

ngowangvinh, 2024-12-27:
### Problem Description

I add the lib to my project on [ESP-IDF VSCODE EXTENSION](https://github.com/espressif/vscode-esp-idf-extension) as a component, then add **idf_build_set_property(COMPILE_OPTIONS "-DESP32_CMAKE" APPEND)** in the root CMakeLists.txt file. Build and bellow error occurs:
..........................................................................................................
-DESP32_CMAKE -std=gnu++2b -fno-exceptions -fno-rtti -Wno-error -Wno-format -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32_DEV\" -DARDUINO_VARIANT=\"esp32\" -DESP32=ESP32 _-DESP32_CMAKE=1_ -fpermissive -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\main.cpp.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj -c D:/ESP_IDF/test3/test3/main/main.cpp
In file included from D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioConfig.h:60,   
                 from D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools.h:66,    
                 from D:/ESP_IDF/test3/test3/main/main.cpp:4:
D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools/CoreAudio/AudioRuntime.h:15:17 
 **error: ambiguating new declaration of 'uint64_t micros()'**
   15 | inline uint64_t micros() { return xTaskGetTickCount() * portTICK_PERIOD_MS * 1000;}       
      |                 ^~~~~~
In file included from D:/ESP_IDF/test3/test3/managed_components/espressif__arduino-esp32/cores/esp32/Arduino.h:36,
                 from D:/ESP_IDF/test3/test3/main/main.cpp:1:
D:/ESP_IDF/test3/test3/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal.h:138:15 
 note: old declaration 'long unsigned int micros()'
  138 | unsigned long micros();
      |               ^~~~~~
In file included from D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools/CoreAudio/AudioEffects/Synthesizer.h:6,
                 from D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools.h:110:   
D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools/CoreAudio/AudioActions.h: In member function 'void audio_tools::AudioActions::clear()':
D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools/CoreAudio/AudioActions.h:230:7 
 error: deleting object of polymorphic class type 'audio_tools::AudioActions::Action' which has non-virtual destructor might cause undefined behavior [-Werror=delete-non-virtual-dtor]
  230 |       **delete(act);**
      |       ^~~~~~~~~~~
D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools/CoreAudio/AudioActions.h: In member function 'void audio_tools::AudioActions::insertAction(Action&)':
D:/ESP_IDF/test3/test3/managed_components/AudioTool/src/AudioTools/CoreAudio/AudioActions.h:246:9 
 error: deleting object of polymorphic class type 'audio_tools::AudioActions::Action' which has non-virtual destructor might cause undefined behavior [-Werror=delete-non-virtual-dtor]
  246 |         **delete(actions[idx]);**
      |         ^~~~~~~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

### Device Description

build for ESP32 
arduino as component
esp-idf VScode extension on windows 10
ESP-IDF version 5.3.2

### Sketch

```cpp
#include "Arduino.h"//WORK WELL
#include "LovyanGFX.h"//WORK WELL
#include "ArduinoJson.h"//WORK WELL
#include "AudioTools.h"

extern "C" void app_main()
{
    initArduino();
    pinMode(4, OUTPUT);//worked
    digitalWrite(4, HIGH);   //worked
    
}
```

### What is your development environment

esp-idf VScode extension on windows 10
arduino as component, version 3.1.0
ESP-IDF version 5.3.2

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

ngowangvinh, 2024-12-27:
if I modify CMakeList.txt in arduino-audio-tools as bellow, the error relate to **delete(act)** and **delete(actions[idx]**) become warrning (no errror any more).
 target_compile_options(${COMPONENT_LIB} INTERFACE -DESP32_CMAKE=1 -Wno-error **-Wno-error=delete-non-virtual-dtor** -Wno-format -fpermissive)
But error relate to **ambiguating new declaration of 'uint64_t micros()** still there. I dont know how to resolve!!!
[image]

pschatzmann, 2024-12-27:
-DESP32_CMAKE is for the case when there is no Arduino environment: so if you have Arduino as component this is completley wrong!

ngowangvinh, 2024-12-28:
Thank @pschatzmann  for your fast response.

I remove **idf_build_set_property(COMPILE_OPTIONS "-DESP32_CMAKE" APPEND)** in the root CMakeLists.txt file. But can't resolve the error. See my project structure in the attached image.
[image]

pschatzmann, 2024-12-28:
You did not: Just look: it is still there!

ngowangvinh, 2024-12-28:
That strange!!! the source code of file AudioAction.h as bellow:
```

#pragma once
#include "AudioTools/CoreAudio/AudioBasic/Collections/Vector.h"
#include "AudioTools/CoreAudio/AudioLogger.h"

#ifndef TOUCH_LIMIT
#define TOUCH_LIMIT 20
#endif

#ifndef DEBOUNCE_DELAY
#define DEBOUNCE_DELAY 500
#endif

#if defined(IS_MIN_DESKTOP)
extern "C" void pinMode(int, int);
extern "C" int digitalRead(int);
#endif

namespace audio_tools {

// global reference to access from static callback methods
class AudioActions;
static AudioActions *selfAudioActions = nullptr;

/**
 * @brief A simple class to assign functions to gpio pins e.g. to implement a
 * simple navigation control or volume control with buttons
 * @ingroup tools
 */
class AudioActions {
public:
  enum ActiveLogic : uint8_t {
    ActiveLow,
    ActiveHigh,
    ActiveChange,
    ActiveTouch
  };

  struct Action {
    int16_t pin = -1;
    void (*actionOn)(bool pinStatus, int pin, void *ref) = nullptr;
    void (*actionOff)(bool pinStatus, int pin, void *ref) = nullptr;
    void *ref = nullptr;
    unsigned long debounceTimeout = 0;
    ActiveLogic activeLogic = ActiveHigh;
    bool lastState = true;
    bool enabled = true;

    /// determines the value for the action
    int debounceDelayValue = DEBOUNCE_DELAY;
    int touchLimit = TOUCH_LIMIT;

    virtual int id() {
      return pin;
    }

    virtual bool readValue() {
#ifdef USE_TOUCH_READ
      bool result;
      if (this->activeLogic == ActiveTouch) {
        int value = touchRead(this->pin);
        result = value <= touchLimit;
        if (result) {
          // retry to confirm reading
          value = touchRead(this->pin);
          result = value <= touchLimit;
          LOGI("touch pin: %d value %d (limit: %d) -> %s", this->pin, value,
               touchLimit, result ? "true" : "false");
[... 186 lines cut ...]
  static void audioActionsISR() { selfAudioActions->processAllActions(); }

  void setupPin(int pin, ActiveLogic logic) {
    // in the audio-driver library the pins are already set up
    if (use_pin_mode) {
      if (logic == ActiveLow) {
        pinMode(pin, INPUT_PULLUP);
        LOGI("pin %d -> INPUT_PULLUP", pin);
      } else {
        pinMode(pin, INPUT);
        LOGI("pin %d -> INPUT", pin);
      }
    }

#if !defined(IS_MIN_DESKTOP)
    if (use_pin_interrupt) {
      attachInterrupt(digitalPinToInterrupt(pin), audioActionsISR, CHANGE);
    }
#endif
  }
};

} // namespace audio_tools

```

pschatzmann, 2024-12-28:
Just compare it with the [main branch from github](https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioTools/CoreAudio/AudioActions.h)

ngowangvinh, 2024-12-28:
Ah! If I remove line **version: v1.0.1** in dependencies, the code will have virtual ~Action(){}.
```
AudioTools:
    git: https://github.com/pschatzmann/arduino-audio-tools.git
    # version: v1.0.1
```

This will resolve the error related to **delete(act);**
But can resolve: **error: ambiguating new declaration of 'uint64_t micros()'** 
[image]

pschatzmann, 2024-12-28:
I guess this is an issue with your build environemnt and the -DESP32_CMAKE has not been properly removed yet

ngowangvinh, 2024-12-28:
I created new project.  Added new dependencies. But can not resolve problem.

[image]

pschatzmann, 2024-12-28:
Check your print screen: you need to find out where the -DESP32_CMAKE=1 is coming from!

pschatzmann, 2024-12-28:
Try the following: in AudioConfig on line 62 comment out #include "AudioTools/CoreAudio/AudioRuntime.h"

pschatzmann, 2024-12-28:
Not sure if this is helping, but I [committed a correction](https://github.com/pschatzmann/arduino-audio-tools/commit/0ba5d3e615a6bf852ba2692cebc4479cf382e1ea).

ngowangvinh, 2024-12-28:
there is only one place I see the -DESP32_CMAKE=1 in build print:

-fmacro-prefix-map=C:/Users/Vinh/esp/v5.3.2/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu++2b -fno-exceptions -fno-rtti **-DESP32_CMAKE=1** -Wno-error -Wno-format -fpermissive -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32_DEV\" -DARDUINO_VARIANT=\"esp32\" -DESP32=ESP32 -MD -MT 

When I commend out **#include "AudioTools/CoreAudio/AudioRuntime.h"** there are many error: 

[image]

ngowangvinh, 2024-12-28:
> Not sure if this is helping, but I [committed a correction](https://github.com/pschatzmann/arduino-audio-tools/commit/0ba5d3e615a6bf852ba2692cebc4479cf382e1ea).

This commit resolve the error!
Thank you @pschatzmann  very much!! 👏👏👏👏👏👏👏👏👏👏👏👏

closed 2024-12-28 by pschatzmann